area515/Photonic3D #103: Using Projector Serial Control Results in "SerialPort already assigned to this job"

jmkao, 2016-01-11:
I have confirmed projector serial connectivity via /dev/ttyUSB1 with minicom from the command line and have been trying to configure CWH with it. I've tried manually configuring the USB device and using autodetection, as well as manually setting the `<ComPortSettings>`  in the machine configuration file to have the correct Speed, Databits, Parity, etc...

I'm not sure, however, that in these tests, the serial connection is actually getting initiated. It seems that regardless of the settings I've tried, I get the error message:

[image]

And the log.err shows the following exception when I try to start the printer with a projector serial port configuration in place:

```
org.area515.resinprinter.display.AlreadyAssignedException: SerialPort already assigned to this job:Titan1(SerialPort:org.area515.resinprinter.serial.JSSCCommPort@2248a4, Display::0.0)
        at org.area515.resinprinter.serial.SerialManager.detectResourcesAndAssignPort(SerialManager.java:137)
        at org.area515.resinprinter.serial.SerialManager.assignSerialPortToProjector(SerialManager.java:150)
        at org.area515.resinprinter.printer.PrinterManager.startPrinter(PrinterManager.java:120)
        at org.area515.resinprinter.services.PrinterService.startPrinter(PrinterService.java:190)
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:483)
        at org.jboss.resteasy.core.MethodInjectorImpl.invoke(MethodInjectorImpl.java:137)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invokeOnTarget(ResourceMethodInvoker.java:296)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:250)
        at org.jboss.resteasy.core.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:237)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:356)
        at org.jboss.resteasy.core.SynchronousDispatcher.invoke(SynchronousDispatcher.java:179)
        at org.jboss.resteasy.plugins.server.servlet.ServletContainerDispatcher.service(ServletContainerDispatcher.java:220)
        at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:56)
        at org.jboss.resteasy.plugins.server.servlet.HttpServletDispatcher.service(HttpServletDispatcher.java:51)
        at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:769)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1667)
        at org.eclipse.jetty.websocket.server.WebSocketUpgradeFilter.doFilter(WebSocketUpgradeFilter.java:172)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1650)
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:583)
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:221)
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1125)
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1059)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
        at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:52)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
        at org.eclipse.jetty.server.Server.handle(Server.java:497)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:311)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:248)
        at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:610)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:539)
        at java.lang.Thread.run(Thread.java:744)
```

WesGilster, 2016-01-11:
A couple of things to keep in mind. 

* If you haven't added your projector to your config.properties:
hexCodeBasedProjectors=[{"name":"Acer","onHex":"4F4B4F4B4F4B4F4B4F4B0D", "offHex":"2A2030204952203030320D", "detectionHex":"2A2030204952203030320D", "detectionResponseRegex":"Unknown"},{"name":"ViewSonic (pjd7820hd)","onHex":"0614000400341100005D", "offHex":"0614000400341101005E", "detectionHex":"071400050034000011005E", "detectionResponseRegex":"(?d:.*\\\\x14\\\\x00\\\\x00\\\\x00\\\\x14.*)"}]
Then the projector setting isn't going to do you much good.

* Once you put these settings into the config.properties, just use the "Autodetect projector" and you shouldn't have to do any kind of setup.

My assumption is that /dev/ttyUSB1 is being used for both projector and firmware? I'm not sure how that situation would have occurred though. If this bug easily reproducible, could you step through how you got this to occur and perhaps configure your email so that you can use the "Send Support Info" button. That diagnostic gives me quite a bit more information than just the stack trace above. Thanks for the help.

jmkao, 2016-01-12:
The hexCodeBasedProjectors property already appears to be present in config.properties with the same value as what you posted.

My printer's Marlin-based Arduino Mega is on /dev/ttyUSB0 and works properly, both when manually configured and when Autodetect 3D Printer Firmware is selected, so long as there is nothing configured for the Projector Serial Port.

I have an Acer projector that is on /dev/ttyUSB1.

Here is the LogBundle:
[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/86743/LogBundle.zip)

In it, I have done the following:

(1) Start the printer named "Titan1" with Motors Serial Port set to "Autodetect 3d printer firmware" and Projector Serial port unset. The printer starts fine.

(2) Stop the printer

(3) Change Projector Serial Port to "Autodetect Projector"

(4) Start the print - results in a "Internal error on server" message and exceptions written to the log.err.

Thanks!

WesGilster, 2016-01-13:
That line might be present in the config.properties, but your projector isn't yet supported. Do you have the hexcode specs for your projector? Notice that your projector is not in the list of supported hardware...

jmkao, 2016-01-13:
Yes, it's the same sequence. All Acers I've found docs for have the same sequences.

The docs specific to the H6510BD are:

[RS232_Control_Codes_H6510BD.pdf](https://github.com/area515/Creation-Workshop-Host/files/88303/RS232_Control_Codes_H6510BD.pdf)

Which show the same "OKOKOKOKOK" and "* 0 IR 002" for on and off respectively as the codes you already have, converted from ASCII to hex.

jmkao, 2016-01-13:
I also sent the ASCII-converted versions of the hex strings directly to /dev/ttyUSB1 using minicom at 9600,8,N,1 and verified that they do work.

WesGilster, 2016-01-13:
Oh Cool. Since I don't allow you to specify the projector, I'll need to add something like the following and we should be ready to go then. I might ask you to run the hardware testkit to determine if I've got these test codes correct...

detectionHex=* 0 Lamp ?
detectionResponseRegex=(?:Lamp 0)

WesGilster, 2016-01-13:
Notice how the current  "detectionResponseRegex" is currently "Unknown" for ACER.

WesGilster, 2016-01-13:
I added the projector autodetection hex codes for the Acer projector in development version cwh-0.191. If you could run the hardware testkit to determine if I've got them right, that would be cool.

jmkao, 2016-01-13:
Here is the terminal output of sending the off command while the projector is off, then the on command (which turns the projector on), and then the off command again (which turns the projector off):

```
* 0 IR 002
*001
OKOKOKOKOK
*000
* 0 IR 002
*000
```

It looks like the off command (`* 0 IR 002`) returns a different value depending on whether the projector is on or not. If the projector is off and the command did nothing, the response is `*001`, and if it actually did turn the projector off, the response is `*000`

The `* 0 Lamp ?` command  returns two lines in response:

```
* 0 Lamp ?
*000
Lamp 0
OKOKOKOKOK
*000
* 0 Lamp ?
*000
Lamp 1
```

The other query commands I tried (Lamp Hours, Model, Company) give a one line response when the projector is off:
```
* 0 IR 037
*001
```

But a two-line response when the projector is on:
```
OKOKOKOKOK
*000
* 0 IR 037
*000
Name Acer
```

I then ran testKitDev.sh, which was detected the firmware but not the projector:

```
Starting test kit
root@cwhost:/opt/cwh# JUnit version 4.12
.Projector detection test.
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.2 (CVS snapshot 2011.02.03, modified by CMU CREATE Lab, http://code.google.com/p/create-lab-commons/)
	native lib Version = RXTX-2.2pre2
Port:/dev/ttyUSB1
  JSSCCommPort projector detection:null
Port:/dev/ttyUSB0
  JSSCCommPort projector detection:null
E.Projector json parse test.
.Firmware detection test.
Port:/dev/ttyUSB1 Baud:9600
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB0 Baud:9600
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB1 Baud:115200
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB0 Baud:115200
  0. JSSCCommPort firmware detection:true
  1. JSSCCommPort firmware detection:true
  2. JSSCCommPort firmware detection:true
  3. JSSCCommPort firmware detection:true
  4. JSSCCommPort firmware detection:true
  5. JSSCCommPort firmware detection:true
  6. JSSCCommPort firmware detection:true
  7. JSSCCommPort firmware detection:true
  8. JSSCCommPort firmware detection:true
  9. JSSCCommPort firmware detection:true
```

I notice, though, that org.area515.resinprinter.projector.DetectProjector test sets the port speed to 115200, although I'm not sure if the speed may be reset to 9600 elsewhere since I'm still getting up to speed on the codebase.

WesGilster, 2016-01-14:
It seems I wasn't using 9600 on the projector test for some reason. I've fixed to make sure all of these tests use a couple of common speeds (9600, 115200). I also fixed the regex to span multiple lines of output as you showed above. Fixed in development cwh-0.193.

WesGilster, 2016-01-14:
Keep in mind that although this test uses speed detection, CWH doesn't use it yet. The point is that you'll have to change the speed manually through the basic printer setup. Let me know if this and your other bug is working and I'll promote this to prod.

jmkao, 2016-01-14:
The output of testKitDev.sh is successfully detecting a projector now, although for some reason it is matching the ViewSonic rather than Acer config, and turns on my projector during the run (which surprised me):

```
Port:/dev/ttyUSB1 Baud:9600
  JSSCCommPort projector detection:ViewSonic (pjd7820hd)
Port:/dev/ttyUSB0 Baud:9600
  JSSCCommPort projector detection:null
Port:/dev/ttyUSB1 Baud:115200
  JSSCCommPort projector detection:null
Port:/dev/ttyUSB0 Baud:115200
  JSSCCommPort projector detection:null
.Projector json parse test.
.Firmware detection test.
Port:/dev/ttyUSB1 Baud:9600
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB0 Baud:9600
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB1 Baud:115200
  0. JSSCCommPort firmware detection:false
Port:/dev/ttyUSB0 Baud:115200
  0. JSSCCommPort firmware detection:true
  1. JSSCCommPort firmware detection:true
  2. JSSCCommPort firmware detection:true
  3. JSSCCommPort firmware detection:true
```

However, I am having problems combining firmware detection with projector detection. Here's what I did:

(1) Delete all printers to start from scratch. Stop.sh, then startdev.sh

(2) Create a new printer with only the firmware configured to autodetect. 

[image]

Printer starts and controls work fine.

(3) Stop the printer, Edit to turn on automatic projector detection like:

[image]

(4) Save and start the printer. Printer fails to start with the error:

[image]

The stack trace in the error log, however, appears to come from the projector detection:

```
org.area515.resinprinter.display.InappropriateDeviceException: No serial ports f
ound for:Autodetect 3d printer firmware
        at org.area515.resinprinter.serial.SerialManager.detectResourcesAndAssig
nPort(SerialManager.java:131)
        at org.area515.resinprinter.serial.SerialManager.assignSerialPortToProje
ctor(SerialManager.java:150)
        at org.area515.resinprinter.printer.PrinterManager.startPrinter(PrinterM
anager.java:120)
        at org.area515.resinprinter.services.PrinterService.startPrinter(Printer
Service.java:189)
```

The autodetection in the actual start printer routine did not turn on my projector.

The LogBundle is attached:

[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/89756/LogBundle.zip)

jmkao, 2016-01-14:
I did some additional testing today and startdev.sh pulled a new version of CWH (0.195).

When the projector's serial port is manually set to /dev/ttyUSB1, the __printer starts and the projector is fully controllable__. I am able to successfully turn it on and off and log.out shows:

```
Printer opened
Printer received:* 0 Lamp ?
Printer received:^G^T^@^E^@4^@^@^Q^@^
lineRead:ok
```

When the projector serial port is set to Autodetect Projector, the printer starts, but clicking the projector on/off controls results in the error message "Couldn't write gcode to /dev/ttyUSB1".

LogBundle for this is attached.
[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/90832/LogBundle.zip)

WesGilster, 2016-01-14:
Awesome. Thanks for the aggressive testing! I didn't publicize that version because I've still got some work to do. I've also built a projector test command line app that will do hex code testing for new projectors. Eventually we could put this into the application itself for adding support for new projectors.

What this last release tells me is that I've got projector detection and the hex codes perfect, but I still have a couple of procedural printer startup issues to resolve.

Is it possible that you attempted to use the projector controls when the printer wasn't started? I know I shouldn't allow that, but I suspect this is the case because it says the port wasn't open:
Couldn't write gcode to /dev/ttyUSB1 Caused by: jssc.SerialPortException: Port name - /dev/ttyUSB1; Method name - writeBytes(); Exception type - Port not opened.

I still have some further debugging I need to do on my own, so I can check that out myself as well.

jmkao, 2016-01-16:
I double checked this again, and the printer was definitely started. I used the motor controls to move the Z-axis a couple of times before then trying to turn on the projector, with the same error result.

WesGilster, 2016-01-16:
Yep I reproduced it this morning. I'll have a fix ready tonight, as I'm at a wrestling meet for most of today.

WesGilster, 2016-01-17:
Fixed in cwh-0.200.

jmkao, 2016-01-17:
Looks good. Connections and autodetect are working well. There's only one test case with manually setting the projector serial port that didn't work.

The testKitDev.sh results show the projector being detected as an Acer.

In CWH itself, I tested the following scenarios:

* **Auto Firmware + Auto Projector** - This worked fine, both projector and firmware were detected
* **/dev/ttyUSB0 Firmware + Auto Projector** - This also worked fine, both were detected and the printer started almost immediately (there's a 5-10 second delay for firmware detection, but not for projector detection)
* **/dev/ttyUSB0 Firmware + /dev/ttyUSB1 Projector** - Couldn't start printer,  error message: `
Couldn't determine model of projector on port:org.area515.resinprinter.serial.JSSCCommPort@8c4f8b`

After the error case, I also went back to /dev/ttyUSB0 firmware + Auto Projector, and the printer again started. So all the serial ports seem to be cleaned up well, even after errors.

LogBundle.zip is below.

[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/93575/LogBundle.zip)

WesGilster, 2016-01-18:
Reproduced and Fixed in cwh-0.202, but I didn't confirm it on my Raspberry Pi. If you are interested, in using the projector hex code tester, you have to run it under the testing location (/opt/cwhTestKit)

java -Djava.library.path=/usr/lib/jni:os/Linux/${cpu} -cp lib/*:. org.area515.resinprinter.projector.ProjectorOutput

jmkao, 2016-01-18:
I tried it on the RPi, and everything looks good. Both manual USB port and automatic detection are working.

Thanks!

closed 2016-01-18 by jmkao

WesGilster, 2016-01-18:
Awesome. I'll push to production.