area515/Photonic3D #105: Files are gone after a reboot

kloknibor, 2016-01-14:
Hi,

After I reboot my respberry pi all uploaded files are gone. Would it be possible to safe these somewhere and decide on your own if you want delete them? (So an delete button for every printable)

Thanks

kloknibor, 2016-01-14:
I notice that for some reason on windows it doesn't delete the files... (atleast there's an image file still present on windows) But the raspberry pi did

WesGilster, 2016-01-14:
That's very weird. We don't delete printables from disk. PrintJobs are only in memory, so they get wiped out on restart. Are you sure you are losing printables?

kloknibor, 2016-01-14:
Yes I do alough the printables have become printjobs... I will try to make soem proof along with the logs for you in a couple of days ;)!

WesGilster, 2016-03-11:
I've never been able to reproduce this issue if this is related to us saving to tmp we can fix this by adding this to your config.properties:
printdir=somedir
uploaddir=someotherdir

This will keep those directories out of tmp storage.

kloknibor, 2016-03-11:
Hmm sorry I promised to give you logs a while back... I'll do a clean install from stable version, upload files, reboot and send you the logs tonight ;)! Should I record to prove I'm not crazy? Haha ;) just kidding. I'll move try the different upload dir too ;)

WGAndrew, 2016-09-16:
If it helps, I've found that printables are volatile too. Whenever I restart a pi, they do just disappear until they're re-uploaded.

WesGilster, 2016-09-16:
This has been fixed long ago in two different ways:
1. In production with the above uploaddir
2. In my current dev version without the above uploaddir fix.

WesGilster, 2016-09-18:
Closed in cwh-1.0.0-beta3.

closed 2016-09-18 by WesGilster