area515/Photonic3D #107: Uploading from thingiverse

kloknibor, 2016-01-14:
Hi,

It tried to upload an stl file from thingiverse. Since the download links aren't ending in .stl CWH won't upload them I guess... I remember that you talked about that thingiverse upload was supported too... How do we do this than? Do I need another link? Or get an link with the .stl path from thingiverse?

Thanks!

kloknibor, 2016-01-14:
( I use a link like : https://www.thingiverse.com/download:87320)

WesGilster, 2016-01-14:
By "doesn't work" I assume you meant the "upload url" button isn't enabled. If thats the case, it's because you didn't type a filename. That filename gives cwh the idea of which print processor to use when a print is started.

kloknibor, 2016-01-14:
Sorry Wes :$ I still have to learn to be more specific... I can understand that you can't understand the trouble with the sortage of information... Sorry! 

But no I did fill in an filename and I was able to press the upload url button but after clicking that I get an "Problem communicating with host printer." error. That's when I try test.stl as filename.

When I try to only use test as filename it says uploading after clicking upload url but it never appears in the printables menu... 

here you can find the log files : https://www.dropbox.com/s/g5soy1hbs61jhq1/LogBundle%20%281%29.zip?dl=0

WesGilster, 2016-01-14:
No Problem. By the way, you can drop the log bundles right in these github text boxes without using dropbox.

A couple of things:
1. Unfortunately the log.err and log.out weren't being included into these log bundle zips for Windows users. 
2. I successfully downloaded the URL you provided and tried to print it on the Raspberry Pi. When I tried to print, it failed with a heap space issue. The fix is easy, but I started questioning whether there is really almost 1 billion(980,448,372) triangles in that model? Given that the file size is only 140104, I'm pretty sure CWH's stl load is misinterpreting the triangle count.

Regardless of the second issue, I fixed the problem in version cwh-0.197, but I'm going to need another log bundle for this. Sorry about that...

kloknibor, 2016-01-15:
ah oke. I tried that with the complete .zip file but github didn't want to do that :p Will unzip it next time ;)! The other logbundle you need should be on my raspberry pi I guess than? Or is windows fine too now? The raspberry pi is at home and my laptop is always with me ;)! That why I try to reproduce it on windows most of the time :)

About the triangles I don't know... To be fair I just picked the first file I could find on thingiverse just to test the working :$

I will try again later on with my respberry pi :)

WesGilster, 2016-01-15:
That's ok you don't need to use the Raspberry Pi, now.  That's what I 
fixed, however you need to use the latest dev version.

kloknibor, 2016-03-07:
Still doesn't work on windows... Log here : 

[LogBundle.pdf](https://github.com/area515/Creation-Workshop-Host/files/161080/LogBundle.pdf)

WesGilster, 2016-03-07:
I confirmed, you are correct. The on-click is totally gone... Maybe that's what the other guy was having trouble with? Upload URL wasn't working for him? I'm fixing the file upload utilities to be more generic. Soon they'll work with all sorts of uploads, not just printables.

kloknibor, 2016-03-07:
I guess that wasn't the problem of the other guy if you meant the one from today? His file is uploading and saved to the pi but than it returns an error as far as I understood :P And why support uploading files which are unsupported...? what are the pros?

WesGilster, 2016-03-07:
Upload printable will stay the same. But I wanted to be able to upload new fonts and plugins, that we require the upload logic to become generic and not locked into printables only.

kloknibor, 2016-03-07:
ah allright :)! thanks for explaining!

WesGilster, 2016-03-11:
Testing this in version cwh-0.259.zip

closed 2016-03-18 by WesGilster