area515/Photonic3D #110: Add bulb hour querying into variable calculations

WesGilster, 2016-01-17:
Once the projector autodetection is working well enough, I could easily add hex codes to query the current bulb hours from the projector(I know the ViewSonic supports that) and allow the user to graph exposure time as a function of bulb life. For those manufacturers that support bulb hour querying, I'm assuming that they would also provide bulb brightness function over the course of the bulbs life. This could also easily factor into the alpha channel on the projector mask function as well.

WesGilster, 2016-01-18:
This document shows the curve of brightness over the course of an unknown bulb's life. This is a pretty dramatic curve and if it applies to UV output at a similar scale, then it's impact is significant enough to compensate for. I'll implement the functionality and you can determine how to apply it to the exposure time. Anyone have any experience running printers with old bulbs vs new bulbs? It looks like exposure times could vary by at least 50%.

https://www.christiedigital.com/TechPapers/Christie-Lamp-Life-Explained.pdf

jmkao, 2016-01-19:
Interesting, and that explains a lot. My bulb is at 289 hours, and I've found that my exposure times are have increased by around 30% even though I'm expecting the bulb to last about 2000 hours total.

I had naively presumed that the light dropoff would be approximately linear, but looking at the chart, there is a steep dropoff during the first 600 hours that explains the dramatic change.

WesGilster, 2016-02-17:
This is implemented in cwh-0.229.

WesGilster, 2016-03-11:
Tests built in cwh-0.259.zip

closed 2016-03-11 by WesGilster