area515/Photonic3D #114: Fing more users

kloknibor, 2016-01-20:
Hi!

Do you guys think creation workshop is ready for some more users so they can test and maybe help develop this software even more :)?

If yes I might suggest to start an website for CWH? (Or maybe even a name change since in my opinion it doesn't have to do much with CW anymore... Except it can run those files too). I can offer some webspace for free if intrested! (On a SSD server in a server park in the Netherlands) This way potential users can find us much more easily! We can post what CWH is capable off, updates on the project and installation method ;)! Maybe an raspbian with CWH included would be nice to offer for download!

Also some social media like twitter and maybe facebook to give some updates from time to time might be nice!

What are your opinions on this?

jmkao, 2016-01-20:
My opinion is that we need more and better documentation, not only on the core functionality of CWH, but also how to setup and use CWH for specific printers and the workflow customary to the users of those printers, like what Muve3D did with NanoDLP:

 * http://www.muve3d.net/press/muve-1-documentation/
 * https://drive.google.com/file/d/0B8xCOdl2soS-UDQ1ckE3SEZvVnM/view?usp=sharing

To that extent, what I think we need is more exposure to and communication with printer manufacturers, especially those who might pay Wes to do some customization for them.

The docs that I've found necessary to work on (and I have a big list) have ended up gravitating towards being more useful to printer integrators rather than end users. Looking at what I wrote for slicing, for instance, it ended up far too technical for an end user to actually comprehend and more at the level for a printer manufacturer or integrator looking to figure out the right settings. Getting the right functions and constants for lifting isn't something that every user should need to do. Once those are dialed in correctly for a given type of printer, they should be set for all users of that printer, and support for those functions and constants really need to come from someone dedicated to that printer.

When I started working on that slicing doc, I had the grand idea that I would get CWH set up for my printer and document my experience along the way over a 3-day weekend. I got a lot done that weekend and found that CWH has fantastic architecture, but also found that there is a long way to go between what CWH provides as a framework, and what is necessary to actually get a specific printer working. FOr example, while I wrote up a lot of stuff about pondering the math of lifting, I have not yet actually printed anything with my settings, and I know that once I start printing, I'll probably need to print quite a lot to dial those settings in. That kind of R&D could be done by users, but would ideally involve the manufacturer.

Before doing a full blown website and marketing release for CWH, what do you guys think about just having a forum somewhere that we can use for discussions and for inviting select users that would be a bit more structured and less daunting to newbies than opening up github issues for everything?

ergobot, 2016-01-20:
This sounds like a good idea.  The software started as a way to run sliced
files on a resin printer with common/inexpensive hardware.  The current
version is able to slice, provide stats of printing, control multiple
printers, provide diagnostics, and more.

From what I'm hearing, to support a bigger development effort of both
software and community, we'll need the following:

A distinct name to reflect the direction
Forum - forum.somehost.org - Community discussions
Frontpage - www.somehost.org - Presentation to others
Wiki - wiki.somehost.org - Community documentation
Repo - github org/ repo for development effort

I have hosting with easy one click solutions for forum, WordPress, and
wiki.  This weekend, I can build this setup with a test domain to show as
an example, if there's interest.

If this is the direction we want to go, first thing todo is a distinct name.

Thoughts, suggestions, opinions?

WesGilster, 2016-01-20:
I agree with almost everything that jmkao said except for a couple of things:
1. I really feel like end users have no business touching the printers section of CWH. I even have an 'end-user' configuration setting that removes it completely.
2. The documentation that muve-1 has is great, but it's pointless and too complicated for an end user. This MUST be touch and go. My last commit to the dev branch should reinforce this theory. If the CWH client can't find a printer, it finds devices(Raspberry Pis, Linux boxes) that it could make into a printer. After that they should choose the printer they have, and they are done. Really done. No command lines, no finding ip addressses, no putty and no command line for extra setup. I should be able to setup my printer with my phone.
3. CWH has so much architecture built around serial autodetection, that there is no reason that printer profiles can't be shared for end user setup. This was never possible in Creation Workshop due to a ton of reasons. These printer profiles could easily be matured over time, which also would be transparent to the end user as our automated upgrade process.
4. I don't see why we can't start on a web page, but keep it closed until a few printer profiles are ready for use. We don't have to support every printer and projector perfectly, just a few before we release.

That being said, we all have different printers at home:
Little RP
mUVE 1
Draken
Titan 1
I think we all need to start printing with CWH and support those printers first... 

Good conversation guys...

kloknibor, 2016-01-20:
I agree with all of you :p I think that ergobot is right but maybe like Wes says it might be to early to make it that public ;)!

Altough I do feel there should be a better way to communicate with each other ;)!

By the way I will have an LCD printer next week :D (next to my Draken and FDM) the LC 10" (from photocentric3D) I will try if it works with creation workshop ;)! They are currently searching for an CW alternative and I suggest to take a look at CWH. Once I get it to work with CWH I will try to convince them how great CWH is :) Maybe they want to work together with CWH :)!

jmkao, 2016-01-20:
Actually, I agree with all of Wes' points. I think that end-users of printers cannot be exposed to all of the nuances to creating a correct machine profile. I think that where we are now, the persona of a person downloading and using CWH is more a manufacturer or integrator, and until we have a library of profiles (either written by us or by manufacturers we can onboard) we won't be end-user ready.

But I do think we have reached sufficient size (at least among the 4 people on this issue) that we are ready for some kind of expanded community.

WesGilster, 2016-01-21:
Great points, there is a ton of stuff to talk about here. As Ergobot suggested, we need a set of next steps:
1. Non public end user web site (Let's hear hosting propositions from Ergo/kloknibor)
2. Public community web site for talking about new ideas and upcoming milestones (Let's hear hosting propositions from Ergo/kloknibor)
3. We all put together a working printer profile for each of our machines and we'll setup a github directory for the data. Remember this doesn't need to be a perfect profile, just a proven stable working profile. I want printout pictures from everyone. :)
4. Add "Select your printer" to cwh-client. (I can take this) Eventually, cwh-client WILL need a rewrite and when it happens we need to ensure the code is sharable with Android. We just abstract out the GUI pieces...
5. A better name. I'm not sure if we are going to change the Github repo name though.
6. What other stuff? Do we need any further printer functionality before we release?

jmkao, 2016-01-21:
I think there's only one outstanding item that I've run into that is probably not 100% necessary but might be worth thinking about. There's the question of how best to tie machine specific UI functions to the machine profile without having the fork the entire project. Rather than belabor the point in here, I'll open a new issue describing it.

jmkao, 2016-01-21:
Oh, and if ergo or kloknibor is looking at hosting, I would encourage the use of Discourse as the forum :).

jmkao, 2016-01-22:
https://www.youtube.com/watch?v=7_Kgx1-G2x4

Need a bit of better logic for the exposure time function, but for a first try, it was pretty good.

Not depicted in the video is that I have been having intermittent issues with the RPi losing USB connectivity to the firmware after a few hundred layers. It isn't clearly reproducable and I didn't have this problem when the projector and firmware were hooked up to a hub and then to the Windows tablet. Since `dmesg` is showing a device disconnect, it's probably more of some kind of RPi issue than CWH. But as we get closer to having actual user-level users who won't go SSHing into the pi to dig through kernel logs, this might be something else we'd have to think about dealing with.

ps-bond, 2016-01-22:
Looks good; is it a Pi2, and what PSU are you using?  There have been issues in the past with power & USB; not entirely clear if this happens when the projector & fw are hooked to a powered hub with the Pi or if you're driving direct from the Pi's ports.  If it happens with the powered hub + Pi then it isn't a power issue, obviously.

kloknibor, 2016-01-22:
About the hosting, I can offer an own webhosting account (I'm a reseller) with CPanel on an SSD server in Haarlem (Netherlands in an big data centre). So it will be a seperate account and I could grant more users acces to the backend ;) The space I could offer is like 2-3GB for now but I'm planning on expanding in some months ;)! (To an Cloud server instead of en dedicated server). Since I got Cpanel with softalicious installing of all free apps (Like forums) is just a push at the button :)! I don't have any experience with Discourse but I think it would be fine by me ;)!

kloknibor, 2016-01-22:
(Discourse looks a bit like an reddit look a like ;))

ergobot, 2016-01-22:
For demo purposes, I've used a test domain.
wordpress site - http://www.crowdedapp.org/w/  (it should be without the w in prod)
wiki site - http://wiki.crowdedapp.org/
Will get the forum site later in weekend, unless kloknibor has a one click setup.  

For discourse, I would probably host on an amazon ec2 instance, and point the subdomain forum.crowdedapp.org to the amazon instance. http://blog.discourse.org/2014/04/install-discourse-in-under-30-minutes/

Doesn't matter to me who hosts, getting examples up there. Throwing out ideas.

kloknibor, 2016-01-22:
@ergobot I will set up some of the examples from my 1 click host :) Than we can choose

kloknibor, 2016-01-22:
I posted it on a leftover domain from me : 

wiki's : 

www.dlptool.nl/wiki1
www.dlptool.nl/wiki2
www.dlptool.nl/wiki3
www.dlptool.nl/wiki4

Forums : 

www.dlptool.nl/forum1
www.dlptool.nl/forum2
www.dlptool.nl/forum3
www.dlptool.nl/forum4
www.dlptool.nl/forum5
www.dlptool.nl/forum6
www.dlptool.nl/forum7
www.dlptool.nl/forum8
www.dlptool.nl/forum9
www.dlptool.nl/forum10
www.dlptool.nl/forum11
www.dlptool.nl/forum12

kloknibor, 2016-01-22:
Let me know what you guys like :)!

ergobot, 2016-01-22:
Plenty of options to go over, very cool. Thanks

WesGilster, 2016-01-23:
Wow, again, lots of info on here.

jmkao,
Cool video with CWH. I'll try to print something this weekend.

I don't spend much time on forums, so I'm not sure I'm going to be much input I can give. I'd like to see a new look and feel, but other than that, I'm good.

kloknibor, 2016-02-16:
Sorry I haven't been active lately, just very busy but I'm getting some more time this week :-) I won't be able to code very much... I started learning some C# but I found out you are coding in java :/ So that won't help... But anyway if you want I can start building a website/private community for us ;)! What are the preferences from you guys? Which one did you guys love most :)!?

jmkao, 2016-02-16:
I'm still partial to Discourse because of it's extremely sane behavior on mobile, especially in the flow of taking a picture on a phone and then posting an entry about it. I do a lot of forum support on the Kudo3D forums and find that being able to rapidly iterate on diagnostics with supporting attachments is extremely helpful.

For the other forums, I guess my main request would be to enable integration with Tapatalk, which smooths over the mobile flows. From the desktop layout perspective, they all seem fairly similar.

WesGilster, 2016-02-17:
@kloknibor 
I'd like to to take you up on your offer to help build the web site. I think Ergobot, would also be able to help in this area as I've seen him create some sharp looking guis. I'll purchase the domain names this week and can setup the DNS to point to your ip, but I'd like to know if Discourse is going to be a possibility (I like the idea of github SSO)

forums.[sitename].com -> discourse forum
www.[sitename].com -> showcase web site (WordPress or something like that)
wiki.[sitename].com -> one of your 4 wiki apps

I also need to know if you have a dedicated ip for your hosting...

kloknibor, 2016-02-17:
Hi Wes,

Let me know what it costs when you buy it. Since I'm an reseller I can get some discount too and if it's only 1 domain name I don't care to bare the costs ;)! That's easier to implement but I also understand if  you want it in own management. By the way, since I'm a reseller I will make a complete separate hosting account, complete with Cpanel. This way people who like to contribute (like ergbot) to the website can get acces. So also with the website we can team up together ;)! but I will share the details by e-mail for who is intrested and I will make automatic backups :)!

the subdomain thing won't be a problem either ;)! 

I'm currently an a shared server, so no I don't have a private IP. But you can just change the nameservers so it points to the host ;)!

WesGilster, 2016-02-19:
It's going to cost $25 for the DNS name. What are the nameservers for your hosting?

WesGilster, 2016-02-20:
Let me know what nameserver(s) to use so I can get this pointed to your hosting.

kloknibor, 2016-02-22:
25$ is way to much for 1 domain name... first year is 9,00 euros and after that 12,50 euros (for a .com adress) 

tyhe nameservers are : 

een.dnssrv.nl
twee.dnssrv.nl
drie.dnssrv.nl

jmkao, 2016-02-22:
What is the domain name you guys were thinking? Now that Creation Workshop is a totally commercial product, it may be good to have a name that doesn't infringe upon any trademarks they may have

WesGilster, 2016-02-22:
You are right, it's much too expensive. If you have an alternative route that I can purchase these, let me know. I haven't decided on a name, but more of a vision. :) 

I've updated the name servers for the following domain:
3DPRINTANYFILE.COM

When we decide on our name offline, maybe we could find a better route for me to purchase these?

kloknibor, 2016-02-22:
@WesGilster you can purchase them from me but I'm willing to donate them to you for the first year ;)! No problem, you work hard enough already :)

kloknibor, 2016-02-22:
So just let me know what you want and I'll purchase them for you and connect them to a cpanel account you get acces too ;)!

kloknibor, 2016-02-23:
@WesGilster Hi discourse will be a problem to install on shared hosting since it runs an completly own server... I'm trying to install something that looks like it : http://flarum.org/ is this fine with you guys?

kloknibor, 2016-02-23:
http://forums.3dprintanyfile.com/ you can find de discourse alternative here, it's installed already :)!

kloknibor, 2016-02-23:
wiki.3dprintanyfile.com/ works too ;)! and main page.

So we got : Flarum, wikimedia and wordpress now ;)!

WesGilster, 2016-02-23:
Very nice. That was quick. Let's keep everything as quiet as we can until we are ready to open it to the world. There is still a few enhancements I'd like to address first.

jmkao, 2016-02-24:
Flarum looks good, and the mobile posting workflow is fairly clean. The only concern I have, mainly for your effort, is that they have not reached their stable release milestone yet:

https://github.com/flarum/core/issues/74

and have a big warning on their README.md:

> Flarum is currently in beta and should not be used in production. It is being developed openly on GitHub. Check out the Roadmap to follow along with our progress.

In particular, they warn that updates may not work:

> Don’t use it in production. We can’t support you if things go awry. And upgrading to subsequent versions might involve getting your hands dirty.

That means that whoever runs this will likely be signing up for a bunch of maintenance work keeping the forum software up to date (since there are bugs, and some of them may be security bugs) and perhaps require extra effort when those updates don't work exactly right.

Are you okay with that?

kloknibor, 2016-02-24:
Yes I know... But it is closes what I found to discourse ;) And I like the software very much!

at first i had some trouble getting all dependencies ready by SSH but I managed to do that :)! Know installing is easy ;)! (Just 1 ssh command and voila) so it won't be a problem to update as long as we keep the database safe and well :)!

So yes I'm fine with it ! + we aren't really in business at this point so flarum will improve before than I hope :)! I alreadu had to do some bug fixing but these were all permision related and solved in seconds ;)! (When you know where to look :P)

kloknibor, 2016-02-24:
Also we can make it look a lot nicer :) So their own forum : https://discuss.flarum.org/

WesGilster, 2016-02-24:
The "don't use it in production" warning is a bit concerning. I'm pretty unfamiliar with most admin forum software management. In the event something "goes very bad", what are our migration options for moving the data out and into another piece of software?

kloknibor, 2016-02-24:
the database is safe but to be honest I'm unfamiliar with converting the database data manually to another database for another forum... But the database is quite simple and we can back-up it so I don't feel really alarmed... But feel free to suggest another software piece :)! I've got SSH/CGI/mod rewrite/Full php acces etc (So we have much options in comparison to most hosters) so most software is install-able but deploying a new server within my server (What discourse does) Isn't possible for me...

WesGilster, 2016-02-24:
I see, so running Docker is the real sticking point here right? I'm fine with the choice, but without migration options, this feels like a pretty permanent choice. I'd like to get other opinions as well.

kloknibor, 2016-02-24:
Sorry I don't quite understand your first sentence about docker... :$

WesGilster, 2016-02-24:
No problem. I assume that the core problem is that you aren't able to deploy Docker containers from within your shared hosting.

kloknibor, 2016-02-24:
Hmm I'm gonna switch to cloud hosting in 1-2 months or an VPS it will be possible to do that there I guess ;)!

kloknibor, 2016-02-28:
Would "LCDLP Host" be a new name? Because we support LCD and DLP printers or something? Maybe you guys have better suggestions :)!

WesGilster, 2016-02-28:
I think we may want to stay away from DLP. I thought it was trademarked by TI.

jmkao, 2016-02-29:
My preference is for single word names, I think "Host" is unnecessary since we can slice too and one day will be able to lay out prints as well. Some ideas:

- LumiPrint (can have .io, although there is a small retail operation in the UK with this name)
- LumiPrint3D (can have .com on this one, zero google hits)
- PiProjector (can have .com on this one, but we can probably support more than RPi)
- ProjectPi3D (can have .com on this one, but I don't love the noun/verb ambiguity with "project")
- Radiant3D (can have .io, small service operation in India with this name)
- Radiate3D (can have .com on this one)

WesGilster, 2016-09-18:
We picked Photonic long ago.

closed 2016-09-18 by WesGilster