area515/Photonic3D #117: Configure LittleRP

ergobot, 2016-01-23:
Starting a new thread for the littlerp conversation.  

Goal is to test CWH setup on littlerp and get the configuration common to all littlerp's.  

Setup steps
Using a pi2, raspian wheezy 2015-05.  Using the install instructions on the main page, have performed a fresh install.  
In the gui, I created a new printer using the configurations in the attached image.  Started the printer.  Went to the controls for the printer.  Clicked movement buttons for z axis.  Z axis on littlerp moved.  No problems experienced.

Printer Name: LittleRP
Display: :0.0
Motors Serial Port: /dev/ttyUSB0
Speed: 9600 (default was different)
Databits: 8
Parity: N
Stopbits: One
[image]

Can anyone else with a littlerp or using grbl on their machine reproduce?
Will move onto the configs next.

WesGilster, 2016-01-23:
That's great news.
How about autodetect firmware?
Do we support your projector; if so have you tried autodetect projector?

ergobot, 2016-01-23:
Problem 1:
The autodetect feature was unable to detect the littlerp (grbl).  

Solution 2 (server side):
In SerialManager:70 is3dFirmware(...), server was only attempting to connect to any ports at only 115200 baud speed.  Attempting to connect using each possible baud rate (x8 total speeds) will successfully connect to grbl.

Problem 2:
In SerialManager:70 is3dFirmware(...), cwh determines that a successful connection has been made by sending "G91\r\n" to grbl and looking for "ok" String response that matches "[Oo][Kk].*".  Grbl is returning "ok\r" String and this is not matching "[Oo][Kk].*".

Solution 2 (server side):
Match the "G91\r\n" String response to "[Oo][Kk].*" or "[Oo][Kk][\r].*".  

Client side:

The server side will take approximately 16-17 seconds to complete the operation (8 possible speeds x 2 second timeout for serial connection attempt).

On the client side, when starting the printer the client will timeout in cwh/printers.js:27
this.executeActionAndRefreshPrinters = function executeActionAndRefreshPrinters(..

I have written fixes for the server side but not the client.  How do we want to handle the long autodetection time on the client?

WesGilster, 2016-01-24:
Great testing!

The serial autodetect isn't detecting attempting detection at different speeds. You must have added an enhancement for speed autodetection?

I figured that since most printers of the same brand will be communicating at the same speed, we could put that as a constant in the printer configuration for that printer. Autodetection was mostly designed to test the difference between the printer firmware and a projector or some other device hooked up to the serial port. I didn't really have a speed detection solution for projectors though so it would be interesting to see where you put the speed detection.

Although, I don't mind if we add speed detection into the mix, if we can keep the timeouts low enough.

I've got a HUGE logging rewrite coming so you may want to hold with your changes so they don't get lost in the mix. I think we want to test your changes independently of mine just to make sure I didn't hork anything up.

WesGilster, 2016-01-27:
I'm thinking about an enhancement that adds the serial port configuration parameters(excluding port name) to each projector model. This seems like a pretty reasonable change since these parameters won't change within a single projector model. This would allow us to increase the number of supported projectors without adversely affecting our auto-detection performance. 

Ergo mentioned that it might be a good idea to autodetect the projector model once, and not have to perform that process on every printer start. Good idea for the future when we start supporting more projector models.

kloknibor, 2016-03-07:
Did this already work like it should? because when it does this can be closed and we can get some more overview on the project ;)!

WesGilster, 2016-03-07:
It's currently in development.

WesGilster, 2016-03-07:
Meaning, Yes, it has been released in the latest development version. Seems to work great for me.

kloknibor, 2016-06-21:
time to close it than :)!