area515/Photonic3D #118: Install to available devices in network

kloknibor, 2016-01-27:
Hi,

For some reasons me CWH is behaving very weird... It comes online for like some seconds... And is reachable but after that it is gone... So I can't get logs and I wanted to try the new install method. Is it done already? Than I can test it!

Thanks!

Robin

WesGilster, 2016-01-27:
1. Does that mean you aren't able to SSH into the box to get the logs?
2. You aren't able to browse to CWH in the browser?

I assume you mean that you can't browse to CWH in a browser. The only 
time that CWH stops listening on 9091 is when it is restarted or you 
configure a different port for it. The only time that CWH issues a 
restart is after a reconfiguration of WIFI. Did you happen to do that?

Otherwise the only other time that I've seen intermittent network 
outages is when I'm on WIFI, and the Raspberry Pi/Wifi dongle are 
intermittently dropping the net interface. I can tell that it's not CWH 
causing the wifi outage because SSH terminals are out during that time 
as well.

If you see this happen again, attempt to SSH into the Raspberry Pi 
through a terminal. If you can't, then your Raspberry Pi is probably 
dropping it's network interface. Although, to be honest, I've never seen 
this issue with Ethernet.

If you can get in, you need to determine if CWH is running by doing a:
ps -ef |grep java

If it's not running, you'll need to send the cwh.log file manually.

If CWH is running you need to run something like this on the command line:
sudo wget http://localhost:9091/services/machine/executeDiagnostic

Which is the same as pressing the "Send diagnostic button" inside of a 
browser accept it rules out your LAN as a potential problem.  It would 
be interesting to determine if this command is responsive...

Thanks,
Wes G.

kloknibor, 2016-01-27:
Hi Wes... 

I think I found out what the problem was... The electronics of the printer where draining to much power from the raspberry pi and it crashed as soon the wifi started working... Sorry for your trouble!

WesGilster, 2016-01-27:
No trouble at all. I'm actually interested. I was under the impression that the Raspberry Pi guys resolved these power issues in the latest Raspberry Pi2 rev. However, both you and I are experiencing similar issues with wifi. I'm a bit skeptical, and may have to reread their claims of "better power support."

Although, I'm wondering. Are you powering the electronics direct from USB, or are you attempting to power the electronics from an external source?

I have a hard switch and can't really remember what state the switch has been in when I've seen my outages occur. I'll need to pay attention to that in my next tests...

kloknibor, 2016-01-27:
Hi Wes,

My printer should have it's own external power supply ;)! This was disabled at the time this happend... When the power is on with my printer it works like a charm! 

And when the power of the printer is off I can normally power the electronic board from an USB altough... There are blinking some LEDS and the PC recognized it...! But this must be to heavy for the raspberry pi ;)! I've got similair issues with other wifi adapters and an 1A power supply... It would reboot the raspberry pi when plugged in or it would just make the pi unable to power up...! But that isn;t the case with my current set up ;)!

WesGilster, 2016-01-28:
It seems like maybe you are describing something similar. I've experienced these situations.
1. external power supply on + USB projector + camera + ethernet = No problems
2. external power supply off + USB projector + camera + ethernet = Can communicate with firmware but can't control steppers.
3. external power supply ? + USB projector + camera + wifi = Intermittent network outages (maybe due to Raspberry Pi power outages?)

I need more troubleshooting with wifi to determine the cause of the network outage.

jmkao, 2016-03-11:
Closing this, handled in #159

closed 2016-03-11 by jmkao