area515/Photonic3D #121: Fail print and navigate to printer controls screen if printer has never been calibrated

WesGilster, 2016-01-31:
Only perform these actions if the printer has not been calibrated or initialized or setup or whatever you want to call it.
1. Fail print and redirect to the printer control screen if the printer has never been calibrated.
2. Direct attention to the calibrate button with a "click me" bubble. Put directions in the bubble that describe how to measure the space between horizontal/vertical calibration lines.
3. After they have clicked the calibrate button, allow them to enter their measurements directly into the CWH.
4. Allow live editing of their started printer "just this once". Generally this is a big NO-NO, because printer starting, stopping, printing are all atomic actions designed to allow expected outcomes in the event of concurrent multi-user manipulation. Maybe we'll allow non-calibrated printers to save even after they have been started?

WesGilster, 2016-02-28:
I still have to start testing everything out, but if you attempt to print when you haven't calibrated your printer, the print will fail with this message. I can take this back out if it's too annoying. The calibration is currently located under the printer controls. Starting it will show you the following calibration screen:
[image]

For those that get confused on the x/y orientation, I've provided a thick dotted line(x) and a thin line(y) and that is the direction you'll hold your ruler(or measuring device) on the build platform. You can calibrate with any number(larger numbers will be more accurate), just move the lines until they line up to the number of mm you've chosen on your screen.

kloknibor, 2016-02-28:
Hmm... Can it be disabled easy? I personally still preslice in CW since we can't create support for now so I got it calibrated in CW. So for that case it useless for me ...

Also does it disable when you choose a default printer profile?I'm currently the owner of an LCD printer at the moment. this way calibration is as easy as give the resolution and build size :)!

jmkao, 2016-02-29:
I like the way this works. In one could laser etch or just print a template sheet with known dimensions and then hit the +/- buttons until they lined up.

I tried this today, and one problem is that given a 1920x1080 resolution display, it takes a lot of presses of the + button to get out to something that takes up a reasonably large amount of the display.

Like the distance jog buttons, I think I would need+/- 1, 10, and 100 pixel buttons to make this a more reasonable number of clicks.

jmkao, 2016-02-29:
I just did a full run and compared it against my previous, pencil based measurements, and the methodology seems quite good, although either the math is off or there's something about the way this is supposed to work that I don't understand.

I set the "X in mm" and "Y in mm" values to 100 each. I put a metric ruler grid onto my printer and then lined up the center crosshairs with lines of the ruler grid to ensure that the ruler is square. Then I hit the + buttons until the distance between the X and Y pairs of outermost lines were 100mm each. Then I hit calibrate.

When I look at the DotsPermmX and DotsPermmY values saved to the profile, they ended up as 5.56 and 5.58mm for X and Y respectively.

The reference values from what I have been printing with is 11.14mm for X and Y.

The numbers seem to be off by a factor of 2, like the code may be overcompensating somewhere for the fact that the pattern is being drawn and moved symmetrically, two pixels at a time.

However, taking that factor of 2 into account, the result is very very good. Measuring the reference values was pretty time consuming and getting an equivalent value this way took only a fraction of the time.

WesGilster, 2016-02-29:
The force calibrate on first use is now a variable that can be disabled in the config.properties. I've disabled it by default. I'm not so sure I liked the idea anyway. 

I'll add some "higher increment" buttons for JMKao.

Weird on that factor of two bit, I'll take a look.

WesGilster, 2016-03-01:
I checked out that "x2 bug" last night and I'm not sure I see the issue yet. I've added logging to determine the exact number that were used. That should tell us what is going on. Checked in to version cwh0.256.

jmkao, 2016-03-02:
Your code is correct, the problem is user error. I had cut off the numbers in the plastic ruler grid and thought that it was 1cm/square, but upon checking with a regular ruler, it turns out that it's 0.5cm/square.

Calibrating to the correct distance resulted in the correct result.

WesGilster, 2016-03-02:
Awesome. Thanks again for testing.

closed 2016-03-11 by WesGilster