area515/Photonic3D #123: remote install not working

kloknibor, 2016-02-03:
Hi,

I can't get the remote isntall to work. It gives an error after reboot and doesn't get started... A youtube clip of it can be found here : http://youtu.be/m-zmTvY1v9Y

hope it is enough to show you the problem :)

WesGilster, 2016-02-03:
Great, the video tells me everything. I've noticed that my Raspberry Pi reboots in about 27 seconds, and CWHClient will timeout looking for it after 30.  It seems maybe I'm pushing the limit on this timeout because your's wasn't found after 30 seconds... I'll increase this further and release a version for you tonight.

WesGilster, 2016-02-05:
I doubled the time allowed to find the Raspberry Pi. This should work fine in cwh-0.219.

kloknibor, 2016-02-05:
Nearly perfect! : http://youtu.be/MZs1sxORWR8 the not found install pop up comes up again with an not yet compatible mote, when I close it He finds my printer and it works like a charm!

WesGilster, 2016-02-06:
I fixed the current issue in cwhClient-0.220. Although I've left one issue unsolved, where it will continue to run forever if cwhClient is never found. I'll fix this tomorrow. 

Ultimately, all of these issues stem from the fact that cwhClient is a mess. Lot's of ideas were just kindof thrown together. Once we start have a small set of community maintained printer profiles setup, we'll need to tackle the cwhClient rewrite. It's actually a pretty small task, but not worth taking until we have all the functionality we think we'll want in it...

WesGilster, 2016-02-07:
All issues fixed in cwhClient-0.222. Let me know if everything looks good and I'll push to prod.

kloknibor, 2016-02-22:
1 year	2 year	3 year	price Period¹
.com	 € 8,99	€ 9,99	€ 8,99	1 jaar

WesGilster, 2016-02-28:
This should be retested now that you can choose to install a printer on CWH installation.

kloknibor, 2016-02-28:
Worked like a charm last time I used it :)!

kloknibor, 2016-02-28:
Sorry doesn't work like a charm anymore... Just tested latest version... browseprinter seems te be broken in the client version ;)! I only get -cp and the area515.resinprinter.somemore and than it closes immediately...!

WesGilster, 2016-02-28:
There were some dependency problems that needed to be resolved. Should be fixed now in cwh0.254.

kloknibor, 2016-03-07:
worked for me like a charm can be closed

closed 2016-03-07 by kloknibor