area515/Photonic3D #124: Wifi connection lost after a while but printer keeps printing

kloknibor, 2016-02-05:
Hi!

My CWH install seems te lose all internet connectiviy after an while but it keeps printing... But I'm unable to get the logs for you since it's unreachable... How can I help to fix this problem?

Robin

WesGilster, 2016-02-06:
CWH no longer wipes out it's internal logs after each startup, but it does roll logs every 24 hours. What this means is that the next time you are able to get in contact with the device(before a file rollover occurs), you can perform the "Send Support Diagnostic". 

I'll warn you though, it's very unlikely that CWH is causing the loss of WIFI connectivity. I've been seeing the same WIFI issues with(or without) CWH installed on the Raspberry Pi. I will say that the outages I've experienced do seem to be related to high network IO.

I first started noticing this as I attempted to install CWH via WEP over an old router. Not only was a simple file transfer failing after about 10meg, it was also killing my entire access point and forcing me to reboot the router! This was 100% reproducible with a simple SFTP file transfer. Next I switched to WPA and WPA2 on the same router(802.11G) which "seemed" to be slightly better. I decided to start using a completely new modem/router(802.11N) and things got significantly better, but I still get outages much too often.

I have not been comforted by what I've found on the Raspberry Pi forums. I've seen quite a few scripts that people are throwing around that bounce the network interface when things get ugly. I could do the same I suppose, but my opinion is that those solutions put this hardware back in the realm of experimental rather than production ready.

WesGilster, 2016-02-07:
Duplicate of #118

WesGilster, 2016-09-18:
Closed long ago...

closed 2016-09-18 by WesGilster