area515/Photonic3D #126: Allow user to choose printer profile during installation

WesGilster, 2016-02-07:
Get the printer profiles from Github and ask the user to choose one during the CWH installation.

WesGilster, 2016-03-04:
Allow profiles to be installed inside of CWH, not just when the printer is installed.

WesGilster, 2016-03-14:
If you don't have any printers, you'll be able to install a community printer profile now. The app is hardcoded to my repo since I'm the only one that has uploaded a printer, but this should probably be an option in config.properties.

[image]

kloknibor, 2016-03-14:
Since this looks like a .json what files do you want for adding an printer profile? And do we still need te make for every height and resin another printer?

kloknibor, 2016-03-14:
I'm geussing this one? : https://github.com/WesGilster/Creation-Workshop-Host/blob/master/host/printers/mUVe%201.json will make it straight away :)!

WesGilster, 2016-03-14:
For now, it's just the printer profile install. #148 will address machine configs, slicing profiles and ink configurations.

closed 2016-03-18 by WesGilster