area515/Photonic3D #127: Better error handling when uploading

kloknibor, 2016-02-07:
Hi!

When you slice an .cws file in CW the slices get exported to the .cws but I accedentily uploaded the files without the slices included. This just does nothing... Since there isn't a progress bar or something you just keep waiting... But it never shows up ;)! Maybe make an error for it? After I included the slices and uploaded the file again there were no problems at all ;)!

Robin

WesGilster, 2016-02-07:
I still haven't decided what I'm going to do for a multiuser environment. It would be easy to simply show a progressbar for the upload, but technically there could be six different uploads being performed at any given time. Where are those going to be shown? I'm still playing around with whether I should allow the upload in the event of a failure and mark the "printable" as "broken" or continue to fail the upload as I'm doing now but still notify users that the upload failed. I'm leaning more towards the first option

kloknibor, 2016-06-21:
Hi Wes,

the upload progressbar can be made client side, so this way you don't have to perform it anywhere on the server side. I think that's easiest to implement. we could use HTML5 / with JS, maybe this helps : https://www.sitepoint.com/html5-javascript-file-upload-progress-bar/

And I'm also leaning towards the first option!

WesGilster, 2016-06-21:
Yeah that looks good to me. A progress bar will always be client side, but a single client could potentially be uploading multiple files. My previous post seemed to imply that you could view the progress of other clients uploads, I didn't mean to imply that.