area515/Photonic3D #129: Bug(ish) : hangs when filming

kloknibor, 2016-02-07:
Hi,

First sorry for the prefixes like "bug" in the title but I'm not quite sure how to add labels...

Second I was filming at 700x700 yesterday for some hours when I was printing. But than everything started to go very slow... I assume the file had taken up all space on the memory card, is this possible? I was also unable to make an mp4 from it ;)! Even after waiting very long.... If my assumption is true we might want to give an error of some kind?

(And the mp4 is only playable with VLC player and not windows video or quicktime)

WesGilster, 2016-02-07:
Wow, that's a long time to record. Most likely you will filled the Raspberry Pi. I will definitely need a "send diagnostic" to be sure, what errors occurred. I'll also see if I can reproduce.

WesGilster, 2016-02-07:
To your other comment. I've been able to play the videos with VLC player and Windows Media Player. I haven't tried quicktime though.

kloknibor, 2016-02-07:
Weird... Windows media player said something about an unsopperted/wrong codec... But well since VLC player can play it, it should be fine in some way :/ and I'm affraid I lost the logs... It has been over 24 hours and I just threw everything out because the print failed... Will try so simulate it ;)

closed 2016-11-17 by kloknibor

WesGilster, 2016-11-17:
Your not crazy. I remember seeing the same issue from Windows media player the first time I attempted to view the video. I've never been able to reproduce it after that first failure though.