area515/Photonic3D #130: Feature request : timelapse

kloknibor, 2016-02-07:
Hi!

Maybe it would be nice to make an time lapse onstead of a real movie, this would use significant less space on the memory card and still produces reasonable quality timelapses :)! The timing would be a variable the user could set. What do you think?

WesGilster, 2016-02-13:
With the correct parameters, I'm betting I already support this, but you need to modify your config.properties file a bit.

Try this:
["raspivid", "-td", "-cycle", "500,5000", "-ih", "-w", "{0}", "-h", "{1}", "-n", "-t", "0", "-o", "-"]

The above statement attempts to put the camera in time delay mode with this parameter:
-td

While the following parameter says to cycle between 500ms record 5000ms pause:
-cycle 500,5000

If this works for you, I can make those parameters definable in the GUI as you mentioned.

WesGilster, 2016-03-11:
Everyone good to move this to done?

kloknibor, 2016-03-11:
I didn't got time to test it, but as far as I can remember it could be improved... Gonna play with it tonight ;)! (Haven't used the camera's in a while)

closed 2016-03-18 by WesGilster