area515/Photonic3D #131: Allow user to choose printer profile during installation

WesGilster, 2016-02-07:
Get the printer profiles from Github and ask the user to choose one during the CWH installation.

WesGilster, 2016-02-08:
[image]
[image]

jmkao, 2016-02-20:
What mechanism are thinking of for selecting templates from GitHub? Or do you think we should take templates for different printers into the master repo itself and ship them with the software?

WesGilster, 2016-02-20:
There are so many ways of doing it, I'm open to suggestion. I figured the simplest way would be to throw them into a directory on github(or somewhere else). Then prompt the user to choose which they want to install.

For those that wanted to share their profiles, they can run this in their browser to get the contents of the whichever printer profile:
http://[raspberryPi]:9091/services/printers/get/[printer name]

In my case it would be:
http://192.168.0.18:9091/services/printers/get/mUVe 1

WesGilster, 2016-02-26:
This feature is complete in version cwh0.249 but you can't test with the WesGilster repo without the changes in cwh0.250.

[image]

WesGilster, 2016-02-28:
There were some dependency problems that needed to be resolved. Should be fixed now in cwh0.254.

WesGilster, 2016-03-11:
Duplicate of #126

closed 2016-03-11 by WesGilster