area515/Photonic3D #134: Use LensFun database for projector lens correction

WesGilster, 2016-02-07:
If the projector that was autodetected by CWH has a LensFun configuration in the json profile, it will automatically download lensfun and use perform the proper lens correction. 
Two different ideas:
1. Somehow I should make users aware that this process is happening... 
2. Should I perform the download in the background regardless of if their projector requires lens correction?

I'll probably implement this along with object translate functions I mentioned in #133.

kloknibor, 2016-02-07:
This sounds awesome! Altough I don't see any projectors in the list but I assume the projectors and the camera's might use the same lenses? The one thing I was disappointed with is the lens dissortion and this might solve it correct? Is Acer H6510BD supported?

WesGilster, 2016-02-07:
Yep, that's the idea. I figured they use the same lenses as well, but that's just a pure guess. If it's not supported, it's just xml, so it shouldn't be a big deal to add your projector to their database. There are like 4-5 variables that control the algorithm. I figured they have a program that aid's in it's creation.

kloknibor, 2016-02-07:
I tried to find out which lense my beamer uses but it's unfindable :( i saw the software to create the xml but it worked with creating an image... So that would mean we should take an image 100% perpendicular to the VAT, use lensfun to correct the image, crop it and use the software to configure the xml, would this still be reliable enough? I reallt do hope so :)

WesGilster, 2016-02-08:
Hmmm. If you can't find the lenses in the database, their process seems a little painful... I suppose we could integrate the process into CWH itself in order to shortcut that workflow. Then you could manipulate the variables on the fly. 

Maybe you could put a transparency sheet (with a printed grid) then show the calibration screen and line up the lines?

kloknibor, 2016-02-08:
Wouldn't an simple white A4 paper do? The projector would shine trough, just attach the pi camera to you build platform :)! And voila correct it :)! Or am I saying something stupid :p?

WesGilster, 2016-02-09:
No nothing stupid at all, I'm just trying to understand the workflow... After you take the picture, you'd import the picture into their software and correct it? Maybe that's not so bad, I just felt it might be a little painful. On the other hand, the part that I like is that I wouldn't have to write anything... :)

jmkao, 2016-06-22:
I have been researching this more, in context of the work on AffineTransform and discussion with some DLP users with different projectors.

There are several different types of "distortion" for a projector, many of which cannot be corrected in software. The most severe, as seen in the Acer 6510BD, which I also have, is the sharp dropoff in focus and brightness in the corners. Most projectors have this kind of unevenness in some regard, but the 6510BD's is particularly extreme.

The reason why many people use the 6510BD is because it is the only projector on the market that can focus close, from 37-200 micron resolution, without any modifications to the lens. Every other projector needs some change to the optical path to allow it to focus close enough for printing.

There are two ways to change the optical path:

1. Modify the projector to increase the distance between the light source and the lens, generally by shimming the lens array with some kind of gasket or some washers by 1-4mm depending on the projector and the desired final resolution.

2. Add close-focusing lenses (essentially rated magnifying glasses) to the front of the array.

Modifying the lens array is mechanically complex, warranty voiding, and fairly permanent. Printers like the Ilios Ray and B9Creator go this route, supplying factory-modified projectors.

Adding a close-focusing lens is easier, but still requires a custom part to attach the lens to the projector and introduces barrel distortion into the output image. The lenses themselves are pretty cheap, like $12 for a set of different magnifications. Printers like the Reify Solus go this route, supplying an adapter and lenses in the kit.

For software image correction, we:
 * Can correct for brightness variation using image masks
 * Cannot correct for focus variation, that's theoretically impossible.
 * Can correct for barrel distortion

The LensFun database is for SLR lenses can can correct for barrel distortion and a couple of different related effects that mostly don't affect us at the pixel counts that we print at compared to pro DSLR cameras. That's why the lens database contains only SLR lenses, not projectors.

However, the corrections it does are probably overkill for us as well. I think that all we need is correction for barrel distortion, and we can probably implement it as an interactive calibration routine similar to how XY resolution calibration works.

The algorithm for correcting barrel distortion is relatively simple, and described at:

http://stackoverflow.com/questions/12620025/barrel-distortion-correction-algorithm-to-correct-fisheye-lens-failing-to-impl
http://mipav.cit.nih.gov/pubwiki/index.php/Barrel_Distortion_Correction

WesGilster, 2016-06-23:
Perfect, I was going to attach the exact same reference. When I implemented it I was going to pull the variable references from lens fun. Then I figured we could submit our projector setups to Lens Fun. Yes, your correct... a bit overkill...

It would be nice if we could implement this in some form of pluggable Filter interface, but I haven't seen a standard Java interface that fits this setup very well.