area515/Photonic3D #136: autodesk print studio image error

nerginer, 2016-02-13:
Hi,
I export png images with autodesk print studio and upload to CWH. I am getting below error. Can you please support print studio images

Regards,
Nuri

Status: Failed Slice image is not in a 3 or 4 byte BGR/ABGR format.Please open an issue about this and let us you know have an image of type: 10

jmkao, 2016-02-13:
Cool, I've been looking to integrate more closely with Print Studio but haven't had time, this will accelerate things. Can you upload and attach to this issue, the zip of PNG images you are trying to print?

nerginer, 2016-02-13:
Great to hear this. Here are the files.
[http://wikisend.com/download/925408/sil3.zip](url)

jmkao, 2016-02-13:
Ughh.... those file names are not zero-padded so they won't sort properly. That will require more complicated work than adding support for image type 10, 1-byte greyscale, which I just finished the fix for.

jmkao, 2016-02-13:
Fixed in my dev branch and a PR is open for Wes' dev branch in https://github.com/WesGilster/Creation-Workshop-Host/pull/8.

If you wish to try this build now, you can download my dev build directly at:

https://github.com/jmkao/Creation-Workshop-Host/raw/dev/host/cwh-0.223.zip

And then unzip this over an existing install directory.

If this works for you, it would be great if you could add a wiki page to this repo describing how you use Print Studio together with CWH. If it doesn't work, let me know what's broken.

nerginer, 2016-02-13:
WOW very quick solution. I will test it and let you know. I can also try to add a wiki about using Print Studio with CWH.

The Print Studio outputs tar file can CWH can open and handle it?

Thanks
Nuri

nerginer, 2016-02-13:
Great! it opened the image files. Please find the original tar file.

[http://wikisend.com/download/547226/sil3.tar.gz](url)

jmkao, 2016-02-13:
Hmm... Chrome really doesn't like wikisend, in the future, you can attach files directly to issues by drag-and-dropping them directly into compose pane.

jmkao, 2016-02-13:
Also, what kind of printer do you have?

nerginer, 2016-02-14:
Hi,
Sorry for the attachments. I will drag and drop next time.
I have an old B9 Creator and I am workıng on my own DLP printer

nerginer, 2016-02-14:
sorry but where are the xml configuration files like machine config etc.

WesGilster, 2016-02-14:
They are stored in three separate directories (a vestige from Creation Workshop). You find them in the user directory for whatever operating system you are on:
For example, my printers are in:
C:\Users\wgilster\3dPrinters

The profiles and machines are in these directories:
C:\Users\wgilster\Profiles
C:\Users\wgilster\Machines

WesGilster, 2016-02-14:
I've pulled these changes and put them into the dev branch. There are quite a few new enhancements into this push as well, so let me know if I've broken anything and if I don't see any new bugs opened, I'll push them to prod. I won't list them here and instead will try to put them into each of the open bugs.

WesGilster, 2016-02-16:
I pushed this all to prod yesterday. I was experimenting a bit using Eclipse instead of the github command line and so things don't quite look right in the Github network graph. I still believe that I got your changes pushed though. Let me know if I missed anything.

jmkao, 2016-02-16:
Yes, it looks like the commit was rebased into the 'cwh0.226' commit (1b121e2e81efebab16e291f7d4cab716d8a0d0bf) and is there. I think this issue is fixed and can be closed, I'll work on the .tar.gz support separately.

WesGilster, 2016-02-16:
Awesome. I'll close...

closed 2016-02-16 by WesGilster