area515/Photonic3D #140: Cannot start print job when no projector is connected

jmkao, 2016-02-19:
In build 229, if I leave the projector disabled in the printer config, I am unable to start new print jobs with the following stack trace:

```
2016-02-19 04:57:10,744 ERROR o.a.r.j.PrintJobManager$JobCloser [PrintJobProcessorThread-1] Job Failed:PrintJobProcessorThread-1
java.util.concurrent.ExecutionException: java.lang.NullPointerException
	at java.util.concurrent.FutureTask.report(FutureTask.java:122) ~[?:1.8.0]
	at java.util.concurrent.FutureTask.get(FutureTask.java:192) ~[?:1.8.0]
	at org.area515.resinprinter.job.PrintJobManager$JobCloser.run(PrintJobManager.java:42) [cwh/:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:1.8.0]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:1.8.0]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:1.8.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [?:1.8.0]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [?:1.8.0]
	at java.lang.Thread.run(Thread.java:744) [?:1.8.0]
Caused by: java.lang.NullPointerException
	at org.area515.resinprinter.printer.Printer.getBulbHours(Printer.java:328) ~[cwh/:?]
	at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:91) ~[cwh/:?]
	at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150) ~[cwh/:?]
	at org.area515.resinprinter.job.AbstractPrintFileProcessor.performHeader(AbstractPrintFileProcessor.java:89) ~[cwh/:?]
	at org.area515.resinprinter.job.STLFileProcessor.processFile(STLFileProcessor.java:79) ~[cwh/:?]
	at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:38) ~[cwh/:?]
	at org.area515.resinprinter.job.PrintJobProcessingThread.call(PrintJobProcessingThread.java:1) ~[cwh/:?]
	... 6 more
2016-02-19 04:57:10,750 DEBUG o.a.r.p.PrinterManager [PrintJobProcessorThread-1] Attempting to dissassociate job:Letter Press Frame.stl assigned to printer:Titan 1(printerFirmwareSerialPort:/dev/ttyUSB1, projectorSerialPort:null Display::0.0) from printer
```

LogBundle attached.

[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/137517/LogBundle.zip)

WesGilster, 2016-02-20:
Sorry about that, I must have broken this with my bulb hours implementation. I fixed this NPE, but it looks like maybe your projector model might not be detected now. I'll test this on my printer as well to be sure it's working again.

WesGilster, 2016-02-20:
Oh wait I didn't see that you said "when **no** projector is connected"...

WesGilster, 2016-02-26:
This has been fixed in version cwh0.249.

jmkao, 2016-02-26:
Confirmed.

closed 2016-02-26 by jmkao