area515/Photonic3D #142: Cannot print any .cws file

djambo, 2016-02-20:
I have CWH installed on a Raspberry PI2 using LittleRP printer.

Whenever I upload a .cws file and try to print it. The print never starts and I get this error:

[image]

Does anyone experienced this issue before?

kloknibor, 2016-02-20:
Hi djambo! 

Welcome here ;) 

I myself haven't seen this error before and it runs stable on my machine... From what branch did you get the software? Area515? That should be the stable one...

Furthermore it seems you got the .cws uploaded so that file should be fine (with the slices etc). I think Wes or James might be able to help you farther ;)

jmkao, 2016-02-20:
I think we're going to need the support files. Either via the logs from directly out of /opt/cwh or by setting up your email server/address under the Settings tab and then clicking the "Send Support Info" link that's at the bottom of every page.

jmkao, 2016-02-20:
Also, attaching the .cws file to the issue would also help in case we need to try to reproduce it.

djambo, 2016-02-20:
Hello everyone!

I am using the Area515 branch. And never experienced this issue before. I took a look at the log file but I can't understand much, anyways

The log file you can find it here:
https://drive.google.com/file/d/0B4wxEaUcAEvpLVdldHNlQXJvZjA/view?usp=sharing

and here is the calibration_cube.cws, sliced with CW 1.0.0.36
https://drive.google.com/file/d/0B4wxEaUcAEvpLVdldHNlQXJvZjA/view?usp=sharing

jmkao, 2016-02-20:
Ahh, the error message is truncated in the UI, according to the log, the full message is:

**java.util.concurrent.ExecutionException: javax.script.ScriptException: TypeError: [JavaClass java.lang.Integer] is not a function in projector gradient script at line number 15**

Did you change your projector gradient function recently? Can you check it for any obvious errors or paste it here for us to look at?

You also seem to be getting an authentication failed exception connecting to your mail server, so you might check that you have your SMTP server username and password set correctly in settings for generating the full support file (which would have included the gradient settings along with the log).

djambo, 2016-02-20:
Here is my projector gradient function, I have never touched it that i know.

function getFractions(count, start, end) {
	var incrementAmount = (end - start) / count;
	var fractions = [];
	for (t = 0; t < count; t++) {
		fractions[t] = start + incrementAmount * t;
	}
	//return new float[]{0, 1};
	return fractions;
}
function getColors(fractions, start, stop) {
	var colors = [];
	var colorRange = stop - start;
	var atanDivergencePoint = Math.PI / 2;
	for (t = 0; t < fractions.length; t++) {
		colors[t] = new Packages.java.awt.Color(0, 0, 0, (java.lang.Integer)(Math.atan(fractions[t] * atanDivergencePoint) * colorRange + start));
	}
	//return new Packages.java.awt.Color[]{new Packages.java.awt.Color(0, 0, 0, (java.lang.Integer)(opacityLevelModel.getValue()/(float)opacityLevelModel.getMaximum())), new Packages.java.awt.Color(0, 0, 0, 0)};
	return colors;
}
var bulbCenter = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var bulbFocus = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var totalSizeOfGradient = $buildPlatformXPixels > $buildPlatformYPixels?$buildPlatformXPixels:$buildPlatformYPixels;
var fractions = getFractions(totalSizeOfGradient, 0, 1);
var colors = getColors(fractions, 0.2, 0);//Let's start with 20% opaque in the center of the projector bulb
new Packages.java.awt.RadialGradientPaint(
	bulbCenter,
	totalSizeOfGradient,
	bulbFocus,
	fractions,
	colors,
	java.awt.MultipleGradientPaint.CycleMethod.NO_CYCLE)

djambo, 2016-02-20:
About the SMPT settings, what address should I put where it says: yourprintermanufacturer@gmail.com?

jmkao, 2016-02-20:
Hmm... that seems significantly more complex than what I get when I create a new printer, although I'm not familiar with the history of the code; maybe @WesGilster would recognize what you pasted.

If you're not using a gradient, you can try replacing it with:

```
var bulbCenter = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var bulbFocus = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var totalSizeOfGradient = $buildPlatformXPixels > $buildPlatformYPixels?$buildPlatformXPixels:$buildPlatformYPixels;
var fractions = [0.0, 1.0];
//Let's start with 20% opaque in the center of the projector bulb
var colors = [new Packages.java.awt.Color(0.0, 0.0, 0.0, 0.2), new Packages.java.awt.Color(0.0, 0.0, 0.0, 0.0)];
new Packages.java.awt.RadialGradientPaint(
	bulbCenter,
	totalSizeOfGradient,
	bulbFocus,
	fractions,
	colors,
	java.awt.MultipleGradientPaint.CycleMethod.NO_CYCLE)
```

This seems to be the current default value for the field 

For the e-mail address, use your own email address, and then attach the zip file, once you receive it over e-mail, to the issue by drag-and-drop or by clicking the "selecting them" link in github's comment editor. I use my own gmail address as the Email Address, Support Email Address, and Username.

djambo, 2016-02-20:
Guys nevermind!
I just deleted my old printer profile and recreated a new one. Everything seems to work now!

I should have done that before bothering here. Thanks again!

WesGilster, 2016-02-20:
If you are extremely attached to your old printer profiles, we can get them converted over, but we'd have to work on it a bit. Printer masks have become much simpler as you can see...

yourprintermanufacturer@gmail.com is designed to be the email address that you want your diagnostic logs sent to. In this specific case that would have been either your email address or straight to jmkao since he was helping you out at the time.

WesGilster, 2016-02-20:
I'll close this since your original question is fixed. Feel free to open another issue if you have another problem.

closed 2016-02-20 by WesGilster