area515/Photonic3D #143: Truncate binary history and/or handle binary builds outside of the repo

jmkao, 2016-02-21:
Cloning the repo now approaches nearly 1GB, most of it seems to be history on deleted binaries (e.g. all the build zips). With more active development, we'll probably grow this even faster.

The process to prune this is not super hard for a single repo. I tried out the instructions at:

http://blog.ostermiller.org/git-remove-from-history

on a clone of area515 using Case A for the 400 biggest files in the history. I edited the output file list to remove the latest build zips and prune out any non-zips that came up (e.g. big jar files and images), and then did the rest of the steps. It took less than 10 minutes and dropped the size of the repo from 900 to 280MB.

What's a little scary is that I'm not sure how forks should pick up a cleanup like this and how it will affect our chains of forks, since right now our dev process is fork/PR heavy rather than branch heavy and involves pushing PRs across chains of forks rather than branch merges. If the process is annoying, then we may want to consider handing the binary builds a different way in the future.

WesGilster, 2016-02-21:
It's great to see someone else a little concerned with this problem as well. I've been waiting for a nasty message from git hub telling us we've got a problem... I've looked at some continuous integration options hooked into hithub:
https://github.com/integrations

I've even tried one out, but these generally involve utilizing a preexisting cloud option for building/testing/storage and I'm afraid we'd be incurring some steep costs as often as we produce builds. It's also not an option to stop producing builds on check in. That's just too important in my opinion.

Another option is to see if we can utilize their large file storage without increasing the size of our repo:
https://github.com/blog/1986-announcing-git-large-file-storage-lfs

I'm actually not that concerned with the propagation of those deletions, in fact it seems as though those deletions should transfer rather quickly(as opposed to the actual zip files themselves). 

I'd like to wait to perform our cleanup until we have a continuous build solution in place, but 1GB is really pushing the limit...

jmkao, 2016-02-21:
Interesting, we're thinking along the same lines then. I wrote an ant build.xml today for command-line building that I was planning to try out with Travis-CI, which seems to have a free tier for open source projects and a fairly extensive set of build options. I'll probably have some kind of POC done later tonight. I'll report back on how it goes.

https://github.com/integrations

https://github.com/blog/1986-announcing-git-large-file-storage-lfs

!DSPAM:56c91a9b124773864439082!

WesGilster, 2016-02-21:
Awesome, I'm looking forward to it. This will be a pretty nice step forward to have these issues behind us.

jmkao, 2016-02-21:
The integration works pretty well, although the way that it works may mean we have to formalize the way we flag versions and releases.

@WesGilster I have added you as a contributor to my repo. If you push any changes (the most interesting change is one to build.number), you will trigger a travis-ci build. I suggest that you `git checkout dev` to switch to my dev branch first before trying it. If you do, I suggest you do your `git clone` with `--depth 3` so as to not pull down the full gig of history.

You can see the travis-ci console showing any current build and the past builds at: https://travis-ci.org/jmkao/Creation-Workshop-Host

If a build succeeds, it will deploy the zip files to bintray under the "Dev" version of a Creation-Workshop-Host package. The version's release information is at:

https://bintray.com/jmkao/CWH/Creation-Workshop-Host/Dev/view#files

And the simplified distribution download directory for the package is at https://dl.bintray.com/jmkao/CWH/. It would be this directory that automated downloaders (like start.sh) would be pointed to.

That leads to the question of how we want to do versioning. Travis-CI's notion of versions for deployment are fairly static. In the case of the way that it would push binaries back to Github, it's based on the git tag that the build was based on. In the case of bintray, it's baked into a config file, that would also be presumably tied to the tag (kind of like a lightweight branch) that the commit was for. The implied workflow here would be that deployments only really occur in release and release candidate tags, rather than feature tags.

Anyway, take a look at these artifacts, try out the commit-push-CI-deploy flow, and see if that sparks any ideas about how you'd like the flow to work.

WesGilster, 2016-02-22:
I really like where you are taking this. I kicked off a build through a check in to your dev branch and saw the versioning questions you are talking about... 

Here's what it looks like we have left:

1. Run non-hardware based tests in build: org.area515.resinprinter.test.FullTestSuite
2. Tag on every build. (I'd really like to stop manually adding the cwh version comments in my commits)
3. Integrate tags into CI versioning...
4. I assume we then keep using the ant version(build.number) for the auto upgrading? If not, we need to think about how the start.sh should know when to upgrade.

   - Offline zip installs
   - Online (build.number) install checks

jmkao, 2016-02-22:
(1) is pretty easy, I can add that.

(2) Is somewhat the reverse of how the CI system works. We can make Travis submit a build whenever it sees a tag on the commit, which is how github releases work (binaries outside the repository can be attached to tags to represent releases in github). Doing the reverse, creating a tag on a build, would require that the automated script itself push commits to the repository, which may introduce race conditions (spurious failures) if commits get pushed during a build.

I think the more "github-ish" way of doing things would be to scrap the auto-incrementing build number, and instead have the release zips with a consistent name. When an admin decides that a version is release worthy, they tag the commit. The tag will trigger Travis to create a Github Release for that tag.

The auto-update script can determine the latest release by hitting the latest release URL for the repo (e.g. https://github.com/bosun-monitor/bosun/releases/latest) which then does a 302 redirect to the latest release URL (e.g. https://github.com/bosun-monitor/bosun/releases/tag/0.4.0). Based on the value of that tag in the URL, the script can determine whether it is at the same level or not and then download the release file(s).

WesGilster, 2016-02-22:
I'm certainly not married to the ant version file... I like standardizing the way we do versioning with the rest of the world. I also understand how the latest url works. There are a couple of parts you'll have to help me with:

1. Before this point I had assumed that each build would be the "release" in "major.minor.release". 
2. How do we test our dev builds if not through a new tag/version.

WesGilster, 2016-02-23:
Just wanted to clarify my question a bit.

I misused the word "release" in the above post...
To rephrase: I had assumed that each checkin build would be "revision" in "major.minor.revision" 
So the version (1.2.260) would be:
1-Major
2-Minor
260-Revision/build

jmkao, 2016-02-23:
I spent awhile thinking about this today. The github way of doing things treats releases as major events, with a notion of "draft" releases in a repo, which are less major, but only visible to contributors. This doesn't map to the way that I think we want to have more casual and fast moving releases, both for dev and stable.

Option 1 is that since tags are just strings, for us to have DEV-##.## and STABLE-##.## tags. Then the auto-updater script would use more complex logic to list tags and filter for the newest with a given prefix.

Option 2 is that since we already use entirely different repos for dev and stable, where we mostly feed contributions into WesGilster/CWH and then you push those to stable via PRs to area515/CWH, that we keep the model and have separate CI flows for each repo. We stick to some kind of numbered tag versioning with all tags triggering a build and release in the repo where the commit occurs, and the repo itself is how we determine whether it's a dev or stable release.

Both options would involve changes to the way we interact with git for development.

In a more traditional git organization, branches represent in-progress features and local repos represent people, with only one origin repo. In a simple small project with just 2 or 3 people, there's a "dev" branch that represents all in-progress features, and everyone commits to the dev branch and pushes to the origin repo. When we near a stable release, people stop committing unstable stuff to dev, and a build master begins merging dev into master.

As the team grows, it expands to have separate feature branches. When we near a stable release, the build master merges in the feature branches that are stable, and leaves the unstable branches out. When multiple feature branches depend on each other, they make an integration branch to collaborate on the dependency, and it's the integration branch that gets merged into master.

In this model, there's only 1 origin repo. CI may run on all branches, but only master's CI flow generates builds.

Our current model is a bit different. We have lots of origins and very few branches. Since we have different areas of expertise and don't have multiple people working on the exact same feature usually, we have a model where feature == person, with area515 as a virtual person who represents the stable release. Our model is more federated, where we're each more like free agents with people each owning repos, rather than having one Github Organization that owns a repo we all share.

And that, I think is the crux of the difference between options 1 and 2. Do we like the current federated "no organization" model, where area515 is a virtual person who represents stable? Since we have no organization, we each have our own CI and each have our own releases. Area515's releases are "special" because "he" or "she" is the arbiter of stability, but otherwise each of our repos work identically and independently in terms of CI and release mechanics.

Or, do we want to be an organization, and move away from the notion that we all have our own remote origin repos, and use the same common remote repo with different branches, and where we do more merging than PR processing.

Both have ramifications on our individual developer workflow. Option 1 is, of course, a radical change that involves much more extensive use of branching and merging.

Option 2, however, is not completely branch free. Since each of our repos have independent CI, they each have to have their own and different `.travis.yml` files. In most cases where we have "local" configs like this, we add them to .gitignore. However, we **must** check in and push .travis.yml, otherwise there's no way for Travis to get the config. That forces us to at least have one separate branch that we do development in which contains out own repo's `.travis.yml` file. We also need to have master branch that is synced to wherever we want to make PRs to. We're ready to make a PR to another repo, we have to merge all our changes **except** .travis.yml from our personal dev branch to our personal master branch, and then initiate the PR from the personal master so as not to include our personal .travis.yml file. This is because PRs are always branch to branch, you pick and choose what to include in the PR by selectively populating a branch, you don't cherry pick in the PR itself.

Thoughts? There are certainly more nuances and more options than these two, but I think they represent the most philosophically different approaches.

WesGilster, 2016-02-24:
There is quite a bit to process here. Basically I feel like this boils down to future project workflow scalability. While I think Option 2 is going to serve us well in the short term and keep us on a fast paced release cycle, if the project becomes fairly successful, there will be a crossover point. That point would occur when there is a sufficient amount of developers working on similar features and option 2 suddenly would degenerate into a bit of a mess. On the other hand, the more traditional approach(Option 1) seems to have higher repo management costs, but would scale to many developers.

I know I'm oversimplifying the situation here, but does that sound accurate? I am leaning towards Option 2 because I genuinely feel that part of our success criteria is a rapid release cycle. As a rule, I try not to choose short term solutions, but it's hard to see what the state of the project will be like in a year. Of course the irony is that my belief in the rapid release cycle as a success criteria could actually hasten the need for Option 1 as popularity grows. I also see Option 2 as a practical choice because we aren't a large project yet. In the event we change theories at some point in the future, what do you see as our biggest threat? Do don't see the project artifacts dramatically changing? I suppose the issue would be a complete reeducation of the developers involved in the project with the new workflow?

jmkao, 2016-02-24:
The decision we make here isn't set in stone, we can change it later without much penalty. I intended to, but somehow forgot by the end of the post, to state my opinion. I also prefer option 2, it's a less drastic change, and lets us each get more comfortable with a slightly more formalized branch/merge before PRs without a major process change at the same time.

I do think that maybe a few small things, like making a Github Organization and putting area515's repo in that might be helpful, if only that allows a few more of us to manage both the triaging/tagging of issues and the third party dependencies, like Travis CI. This also does not need to be done at the same time as anything else, although a good time for something like that might be when/if we decide to rename the project.

jmkao, 2016-02-24:
Note for option 2:

https://medium.com/@porteneuve/how-to-make-git-preserve-specific-files-while-merging-18c92343826b#.chty9978j

WesGilster, 2016-02-24:
area515 is already an organization(https://github.com/area515). I'm only loosely associated with it, but I'm guessing it wouldn't be a big deal to add more members. I'll see what I can do to get owner privileges.

That URL post is exactly what we need for CI management...

kloknibor, 2016-02-24:
Hi I haven't followed this really much and to be honest didn't read it all... But I understood there will be a limit in space that we can acces on github. You could store some of the files on the own server and if it only need to be downloadables than I can get some file dumping cloud storage ;)! (I maybe can get us 1TB free cloud space from where people might download really fast!)

WesGilster, 2016-02-24:
At the moment we are using a free tool chain for CI that also uses a CDN fo fast download. Eventually we may want to bring it all internal into the cloud, but I don't see a need to increase your hosting costs just yet.

kloknibor, 2016-02-24:
@WesGilster the 1TB is a free service ;)! No costs at all... :)

kloknibor, 2016-02-24:
I can give you as we speal 50-100GB for free (I can share some space from my personal free 1TB space) but I applied for an free 1TB account for us already :)! It hasn't to do anything with the hosting provider ;)!

WesGilster, 2016-02-24:
That storage might be a great place to archive our past builds, but our CI processes also require some custom apps to be installed in as well. Many times(not always) these apps are deployed as Docker containers. I kindof got the impression from the other thread that your shared hosting doesn't yet support Docker.

kloknibor, 2016-02-24:
The storage space with 1TB is uncapable of installing services on it ;)! that server is running an modified owncloud version, so see it as an dropbox alternative ;)! 

And indeed currently I guess I don't understand it... I'm gonna ask about it ;)! But the cloud or VPS I will get in 1-2 months will be capable :)!

jmkao, 2016-02-24:
Storage is not really a limiting factor here, places like bintray are free for open-source projects. The main issue is the automation flow and the reduction in manual steps so that releases can flow rapidly without burdening developers with large downloads of files they don't need.

jmkao, 2016-02-26:
@WesGilster If you want to start with option 2, I can deploy the integration into area515 if you add me to the org with with push+pull or push+pull+admin privileges. We can start with a mechanism that can run alongside the existing build mechanism (one that builds releases to tags using consistent naming in travis, but maintaining the current build.number based system for manual builds. And then proceed from there.

The only one remaining question I have is about the host/testbin directory. I notice that while srcbin is left for individuals to build, you have the compiled src from host/tests checked into the repo. Is there a reason you might want to run tests out of the checked-in class files versus having CI build the test code out of the host/tests directory?

WesGilster, 2016-02-27:
@ergobot Could you add us(jmkao and I) as owners to area515 organization? I'm not sure if you have this ability within Area515. If the Area515 group isn't hip to the idea, we may want to think about another organizational structure that gives that membership freedom to this group.

> Is there a reason you might want to run tests out of the checked-in class files versus having CI build the test code out of the host/tests directory?

No, it was a total mistake. I just forgot to add host/testbin to the .gitignore. No binaries should be checked into the repo. Sorry about that...

Yeah let's go with option 2.  Now that you are our new build master I have a couple of requests... :)

WesGilster, 2016-02-27:
Sorry, I posted a little early before I wrote my build requests, and now that I've taken a second look at your build, you are already doing all the stuff I was looking for(except for running tests on build) This isn't a huge priority anyway because I have some computer vision tests that I still need to fix and running those would break our build. Let me know when you feel this is a priority and I can fix the vision algorithm to tighten the tolerances to where I have the tests written.

jmkao, 2016-03-06:
CI is now integrated into area515. All commits into the repo will trigger travis-ci to run, and the output is visible at https://travis-ci.org/area515/Creation-Workshop-Host.

The easiest way to uptake this, for organization members, is simply to begin working in a branch rather than in a fork of the repo. All commits to branches will also trigger CI.

The next step is to automatically build and run the tests as well.

jmkao, 2016-03-06:
Commit 189da8b77718226a54abee3708423b7257c9caf0 has some test cleanup and the beginnings of an ant script to run tests.

@WesGilster need some help or info from you on which tests are supposed to be runnable, since right now there are a lot of failures and errors. We may need a way to identify/group tests by some naming conventions so that the ant target can pick out the right ones to run by pattern.

WesGilster, 2016-03-06:
There are two suites. I suggest we only run the first until we have all of the hardware dependencies for the second suite figured out.

org.area515.resinprinter.test.FullTestSuite
This suite is mocked and is designed to run on any hardware: (Some vision tests don't run)

org.area515.resinprinter.test.HardwareCompatibilityTestSuite
This depends on several hardware features being available (raspicam, wifi, email, any projector, etc...)

jmkao, 2016-03-06:
Ok, pretty close now. There are a number of test breakages, but only two fundamental problems

1. Test failure in `org.area515.resinprinter.inkdetection.visual.TestVisualPrintMaterialDetector.testAgainstKnownPercentages`
1. Errors in tests that use `org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)` and seem to be passing a null value somewhere that reaches the templating engine.

Output details:
```
    [junit] Testcase: testAgainstKnownPercentages took 0.153 sec
    [junit]     FAILED
    [junit] Algorithm outcome:0.6101882 for image:CircleLine10-14.png not within tolerance:0.09 of predicted value:0.71428573
    [junit] junit.framework.AssertionFailedError: Algorithm outcome:0.6101882 for image:CircleLine10-14.png not within tolerance:0.09 of predicted value:0.71428573
    [junit]     at org.area515.resinprinter.inkdetection.visual.TestVisualPrintMaterialDetector.testAgainstKnownPercentages(Unknown Source)
    [junit] 
    [junit] Testcase: determineLiquidLevelInAToughSituation took 0.465 sec
    [junit] Testcase: generateHoughSpace took 0.37 sec
    [junit] Testcase: testLines took 0.048 sec
    [junit] Testcase: generateHoughSpace took 0.36 sec
    [junit] Testcase: getNetworks took 0.359 sec
    [junit] Testcase: syntaxErrorInTemplate took 0.526 sec
    [junit]     Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit]     at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:89)
    [junit]     at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)
    [junit]     at org.area515.resinprinter.job.AbstractPrintFileProcessorTest.syntaxErrorInTemplate(Unknown Source)
    [junit] 
    [junit] Testcase: usingUnsupportedBuildAreaWithLiftDistance took 0.011 sec
    [junit]     Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit]     at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:89)
    [junit]     at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)
    [junit]     at org.area515.resinprinter.job.AbstractPrintFileProcessorTest.usingUnsupportedBuildAreaWithLiftDistance(Unknown Source)
    [junit] 
    [junit] Testcase: properGCodeCreated took 0.01 sec
    [junit]     Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit]     at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:89)
    [junit]     at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)
    [junit]     at org.area515.resinprinter.job.AbstractPrintFileProcessorTest.properGCodeCreated(Unknown Source)
    [junit] 
    [junit] Testcase: EnsureMethodsThrowExceptionIfNotInitialized took 0.001 sec
    [junit] Testcase: noNullPointerWhenWeReturnNull took 0.081 sec
    [junit]     Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit]     at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:89)
    [junit]     at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)
    [junit]     at org.area515.resinprinter.job.AbstractPrintFileProcessorTest.noNullPointerWhenWeReturnNull(Unknown Source)
    [junit] 
    [junit] Testcase: getExceptionWhenWeReturnGarbageForLiftDistanceCalculator took 0.056 sec
    [junit] Testcase: unsupportedBuildAreaDoesntBreakLiftDistanceCalculator took 0.014 sec
    [junit]     Caused an ERROR
    [junit] null
    [junit] java.lang.NullPointerException
    [junit]     at org.area515.util.TemplateEngine.buildData(TemplateEngine.java:89)
    [junit]     at org.area515.resinprinter.gcode.GCodeControl.executeGCodeWithTemplating(GCodeControl.java:150)
    [junit]     at org.area515.resinprinter.job.AbstractPrintFileProcessorTest.unsupportedBuildAreaDoesntBreakLiftDistanceCalculator(Unknown Source)
    [junit] 
```

WesGilster, 2016-03-06:
So here comes the importance of why version number/build tag sync come into play. I know it seems like a waste of time to have me commit  "cwh-0.2XX" as the version number on every commit, but it's the only way I can line up numbers with builds. For example, I'm looking at the line number you have above, and it doesn't match with the correct line:
Line 89: 		root.put("now", new Date());
I know the null pointer didn't happen here because root was initialized on the statement immediately before this. In this case, it's really not that big of a deal because I can just run the test myself. On the other hand, when I need to start looking at stacktraces coming in from many different users and versions, I have to easily determine what the precise codebase they are using. Right now, I can use "compare to commit" and since "cwh-0.2XX" is the first line of the commit, I have their code base locked tight. However, it would be much nicer to "Compare to tag" where tag is the build number...  What's the plan for how this is all going to latch together?

I'll start looking at the broken tests.

jmkao, 2016-03-06:
Good point.

For production, we can log the tag based on a property file that github will set so that we can always identify the tag that a given logged file comes from.

For development, this is trickier since we have different repos, although for the main area515/CWH, I can probably do the same thing and have the log also indicate the branch name and commit date or commit ID.

That will at least ensure that logs clearly indicate a point in code where they originate.

For test breakages like this, travis-ci would handle things automatically, since a build is always done in the context of a commit. So once we have tests running in builds, the output would always be associated with a specific commit.

jmkao, 2016-03-12:
I've got a gradle script that allow one to generate releases into their own fork/repo without having to deal with setting up Travis CI and the complications of having to set those aside to not mess up the upstream repo.

With the gradle wrapper in the repo, this also means that installing ant or gradle is unnecessary. Gradle will install itself when run.

A usual clean/build/run/test would be run with:
 - `.\gradlew clean`
 - `.\gradlew run`
 - `.\gradlew build`

etc...

The "personal fork" release process would be:

 - Run locally, test thoroughly
 - `git commit` changes, pushing them too would be good
 - `.\gradlew clean dist`
  - this will generate the distribution zips, currently still using build.number
 - `.\gradlew githubRelease`
  - this will prompt you for a tag, then tag the commit, push the tag, and then upload a release to github on that tag.

The resulting release will be visible in your github release page, like https://github.com/jmkao/Creation-Workshop-Host/releases

Next step after this would be to update start.sh to pull the latest release from area515 by default, or from a fork based on the command line parameter.

The 'latest' alias will take you do the latest release (e.g. https://github.com/jmkao/Creation-Workshop-Host/releases/latest)

Then use this method to grep out the file you want and compare it to the current build number:
http://stackoverflow.com/questions/24085978/github-url-for-latest-release-of-the-download-file

If different, download and install.

jmkao, 2016-03-13:
There is  now a `newstart.sh` that checks and downloads from Github Releases instead of from inside the repo. It should work for both installation and incremental updates.

jmkao, 2016-04-29:
With newstart.sh in as the default, I am cleaning the repo history now and then will place a backwards compatibility zip in

jmkao, 2016-04-29:
@WesGilster @kloknibor @ergobot 

The repo cleanup is done. Rather than download 500MB and expanding to 1.6G on disk, a full clone now downloads 35MB, expanding to 120MB on disk.

However, because this rewrites history, there are **important changes that all existing clones or forks need to do**!

For clones:

Your best bet is to simply delete your local repo and clone it again, like:

- `rm -r Creation-Workshop-Host`
- `git clone https://github.com/area515/Creation-Workshop-Host`

However, if you have unpushed changes, you'll have to save those changed files to the side and then re-apply any changes manually.

For forks:

This is a bit more complex. The easiest thing to do would be to erase your fork from the GitHub UI an then fork again. But then people who have starred or are watching your repo might need to re-star and re-watch your repo. Alternatively, you can force repopulate your repo, with the following:

- Clone the clean area515: `git clone https://github.com/area515/Creation-Workshop-Host`
- Change the upstream URL to your fork: `git remote set-url origin https://github.com/jmkao/Creation-Workshop-Host`
  - Change the `jmkao` to be your username
- Force push the repo to your fork, this will overwrite your repo with the area515 contents: `git push --force`
- Erase and re-clone any local copies of your old repo

jmkao, 2016-04-29:
Also, one note for those taking the cleanup into their forks, the clone of the fork won't actually get smaller until some mysterious amount of time passes and some kind of github-side periodic gc/cleanup occurs. However, it should be immediately compatible with the history of area515/CWH.

kloknibor, 2016-04-29:
wow that is awesome james! That will speed up the download very much! I'm gonna update my forks tonight ;)

WesGilster, 2016-04-29:
Awesome. This was a pretty painful process, but we are set in a much better position now. Thank you James!

jmkao, 2016-04-30:
I have all the steps described at: https://github.com/area515/Creation-Workshop-Host/wiki/Fix-Fork-After-April-2016-Cleanup

In particular, more steps than I posted earlier are required if you want to get your own repo down to 30MB.

WesGilster, 2016-04-30:
So I've been trying to figure out how this tool has rewritten the history and it's been quite an effort today. I'm seeing a couple of nasty side effects here. Any push or PR to area515 from a repo that hasn't itself been cleaned or forked post April 29 carries a significant risk of bringing back the entire history. Is that a correct statement?
1. Best case: shouldn't these historical purges have been performed by the tool as commits such that they can be propagated through git's normal push mechanism?
2. Worst case: do we not have the capability of performing some sort of rebase?

jmkao, 2016-04-30:
Yeah, the way that binaries are removed out of the commit history, whether by BFG or by git filter, is by going back through the commit history and altering commits of binaries so that they exclude them. This alteration of history changes the hashes of those past commits so that previous clones see themselves as ~520 commits ahead and ~520 commits behind.

That's the only way to purge them out of the repo history, it's not possible to remove the binaries as commits, since new commits can only add to history.

This is the equivalent of rebasing back across all history and does create problems for people who don't share that history, who will try to rewrite history to match their repos if they try to submit PRs without first re-cloning or re-forking per what I wrote at: https://github.com/area515/Creation-Workshop-Host/wiki/Fix-Fork-After-April-2016-Cleanup

WesGilster, 2016-05-01:
Ok. We'll just have to wary of large commits or else everything will come back into the master repo.

closed 2016-05-14 by jmkao