area515/Photonic3D #144: lcd character display i2c

nerginer, 2016-02-23:
Did you think about a character LCD on the machine. It can handle the below functions

- On start up when rasp-pi on, ready an connected it can display a "Printer ready" message with its IP address
- After file upload a "Upload done" message
- While printing something like "Slice 235 of 555"
- After Job compilation "Job completed" message
- Other error messages
- some debug messages for developers

What do you thing about implementing this functionality? I can provide you the circuits and the connection diagram also the hardware itself.

kloknibor, 2016-02-23:
I like it :)!

WesGilster, 2016-02-23:
Sounds great. Are these connected directly to ramps and controlled through gcode? If so, I know I can do that. Otherwise, I'll need to understand a bit more about how you plan on interfacing with the LCD before we get started.

WesGilster, 2016-02-23:
Sorry, I didn't see that you said i2c in your subject line... We've got an API for that. I'd like to see the specs next.

nerginer, 2016-02-23:
My plan is to connect an I2C LCD to raspi GPIO I2C bus. But I am open to any other or better way.
How can I reach to the i2c API?

In the other way how to control the LCD through gcode?

WesGilster, 2016-02-23:
That's why I was asking you about how you were planning on wiring this.

Here are two ways it could be done...
1. Firmware must support it AND motor driving hardware must support it
Apparently some 3D Firmware are already designed to do this: Checkout GCode M117 on: https://github.com/MarlinFirmware/Marlin/wiki/G-Code-in-Marlin
Then, all you would do is add the gcode in our GUI and you are completely done. 
Very simple...

2. Direct from Raspberry Pi GPIO I2C bus in CWH.
You include these in your build:
```
pi4j-core.jar
pi4j-device.jar
pi4j-gpio-extension.jar
```
You write an implementation for what events you want to take place:  
`org.area515.resinprinter.notification.Notifier`

You enable your notifier in the config.properties like this:
`notify.org.area515.resinprinter.notification.LCDMessageNotifier=true`

After your workspace and dependencies are setup, I would say you could get a proof of concept done in a matter of minutes if you are familiar with I2C.

WesGilster, 2016-02-24:
Also, I didn't mean to give the impression I didn't want to write this. I would very much like to write it and have some pretty cool ideas about the message templating we could use with FreeMarker.

Let me see the documentation and I'll let you know if I can take a crack. Most likely it's not going to be a big deal, but I don't want to make promises about hardware I don't know anything about.

nerginer, 2016-02-25:
Sounds great. This will be a great enhancement. I take a look at pi4j. We need to find an I2C LCD sample code. There are some libs but can not be sure which one to use. I will build up a test hardware setup and test the libs and find the best one. Than I will write to you to implement the FreeMaker templates. People can modify the templates to use for multi language supports and  for custom messages. I will be very happy to help you on the hardware side.

WesGilster, 2016-02-25:
Sounds great, just send me a PR when you are ready for me to integrate FreeMarker.