area515/Photonic3D #145: Customizing CWH to a specific DLP printer

nerginer, 2016-02-23:
Suppose that  you have a specific DLP printer. For example your machine has 2 axis. One for tilt the other is Z axis. You know all the gcode for the movements the speed and the distances. What is the best way to customize the CHW ? what are the configuration files that can be hard coded? For the UI is it enough to modify (make invisible) the printer settings?

jmkao, 2016-02-23:
You'll need to modify more settings than what is available in the UI. Off the top of my head you'll need to:

1. Create a printer in the UI
1. Create the G-code and slicing calculators (which I documented in gory detail at https://github.com/area515/Creation-Workshop-Host/wiki/Slicing-With-Creation-Workshop-Host)
1. Modify the machine definition and slicing profiles on the RPi in /root/Machines and /root/Profiles to set all of the correct dimensional variables (X/Y resolutions, sizes, and dots per mm) for your build area.

If it's the same machine, then you can distribute those files. If you want to rename them from the default names, edit the printer configuration from /root/3dPrinters on the RPi.

If you set the machine definition to use Autodect for firmware and projector, the only thing that a user would need to change manually would be to disable the projector control altogether if they don't have their projector hooked up.

WesGilster, 2016-02-23:
At the moment, I'm seeing two places where the GUI isn't enough:
1. The printer still needs to be calibrated which I'll be implementing with enhancement: #121 (That's the dots per mm that jmkao mentioned)
2. The X/Y resolutions/sizes do also need to be properly set, but the sizes will be defaulted to sizes of the last available detected display at the time your printer was created.

In retrospect, I wish I wouldn't have carried over all of the variables from CW, because they really aren't that important. I've even thought about the X/Y resolution being autodetected from the hardware on every print instead of recorded in the printer profile as I do now. That would make our printer profiles compatible on any hardware as they change it. If there is a setting that you think you need in the GUI(and I don't already have a bug opened for it) just let me know.

CWH is built upon a restful interface so you can see all of the options "behind the scenes" with this command:
http://[RaspberryPiIP]:9091/services/printers/get/[NameOfYourPrinter]

Then you'd want to format it with a json formatter or something...

jmkao, 2016-06-22:
Closing this since i think the question is answered, if specific features are necessary, feel free to open a feature request.

closed 2016-06-22 by jmkao