area515/Photonic3D #146: Windows version broken

kloknibor, 2016-02-24:
Hi!

I tried to use the windows version yesterday but it isn't working... When starting up I get the following error :

C:\Users\xxxx\Desktop\CWH>java -Dlog4j.configurationFile=log4j2.properties -Dj
ava.library.path=os/win64 -cp lib/*;. org.area515.resinprinter.server.Main  1>lo
g.out 2>log.err

this is what is in the log file when I try it again : 

2016-02-24 12:37:16.917:INFO::main: Logging initialized @3554ms
2016-02-24 12:37:17.865:INFO:oejs.Server:main: jetty-9.2.3.v20140905
2016-02-24 12:37:17.909:INFO:oejsh.ContextHandler:main: Started o.e.j.s.ServletContextHandler@3972a855{/video,null,AVAILABLE}
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Deploying javax.ws.rs.core.Application: class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding provider singleton org.area515.util.ExceptionMarshaller from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.SettingsService from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.MachineService from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.PrintJobService from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.PrinterService from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.MediaService from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding provider singleton org.jboss.resteasy.plugins.providers.jackson.ResteasyJackson2Provider from Application class org.area515.resinprinter.server.ApplicationConfig
feb 24, 2016 12:37:19 PM org.jboss.resteasy.spi.ResteasyDeployment 
INFO: Adding singleton resource org.area515.resinprinter.services.PrintableService from Application class org.area515.resinprinter.server.ApplicationConfig

It's from the stable version from are515 and older versions of cwh still work ;)!

Robin

WesGilster, 2016-02-24:
Is this the full log? I don't seem to see any errors in here. It would be much better if I got a log bundle. Also, what feature were you having trouble with that it isn't working under Windows? I do my testing under Windows and haven't seen any issues.

kloknibor, 2016-02-24:
Hmm sorry I reported this error wrong... CWH starts fine you just cleaned up all the text in the cmd :)! Great :)! When I try localhost:9091 it works fine but the browseprinter.bat gives an error and closes soon after opening and doesn't open CWH... (Running start.bat and browseprinter.bar from the same machine) 

the error is as follows : 

[image]

Is browseprinter also logged somewhere? Than I'll send that log too ;)!

WesGilster, 2016-02-24:
Java is probably ok, but I see the problem. You are trying to run the browseprinter.sh from a cwh.zip instead of cwhClient.zip. That won't work because cwhClient requires SSH tools that aren't available in cwh.zip itself.

I should fix this so that browseprinter.sh won't be available in situations where it can't be used.

kloknibor, 2016-02-24:
allright! Thanks for the quick response :)!

WesGilster, 2016-02-26:
This was fixed in version cwh0.249. I no longer deploy browseprinter.sh with cwh-0.xxx versions.

kloknibor, 2016-02-28:
this can be closed ;)!

closed 2016-02-28 by WesGilster