area515/Photonic3D #147: Feature request : Add wifi password from SD card

kloknibor, 2016-02-25:
Hi,

So I figured that some people won't have easy acces to their ethernet cable to set up wifi. Why not make it possible to just put something like a .txt file on the SD card which can be seen in the explorer when they insert their card. Than people can just add their wifi password and SSID to that .txt and once booted the wifi network setting will be retrieved and it will connect automatically... That can make life much easier for some I guess :)!

Robin

WesGilster, 2016-02-25:
Yeah, I can add this as a plugin feature.

kloknibor, 2016-02-25:
Was thinking about this more... To make this happen we need to provide custom roms or raspbian needs to support it... We will need wifi to install CWH... sorry :$ So this might be more helpfull for later ;)! But a plugin would be nice too :) We can make custom roms with easy with win32 but we will have problems as soon as the used card is slightly smaller than the one we used... Must be a better way to make this happen ;)!

WesGilster, 2016-02-25:
Not necessarily. There is a Windows mountable partition that also mounts under linux that serves as a sort of bios for Raspbian. That's partially what raspiconfig attempts to setup. Try to make a file on that partition, if you can, this is doable. Otherwise we'll need to table this feature for a bit.

kloknibor, 2016-02-25:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=75762 maybe this helps ! And yes you can create files there !

From: Wes G. 
Sent: Thursday, February 25, 2016 2:51 PM
To: area515/Creation-Workshop-Host 
Cc: kloknibor 
Subject: Re: [Creation-Workshop-Host] Feature request : Add wifi password from SD card (#147)

Not necessarily. There is a Windows mountable partition that also mounts under linux that serves as a sort of bios for Raspbian. That's partially what raspiconfig attempts to setup. Try to make a file on that partition, if you can, this is doable. Otherwise we'll need to table this feature for a bit.

WesGilster, 2016-02-26:
Wow, I didn't actually expect that, for some reason I thought the 
partition was read only under Windows. I'll still hold off on 
implementing this feature right away since there is already a wifi 
solution in place. I'd like to attack the last feature(printer 
calibration) for my "user friendly" release. Then I want to get back to 
work on some of the cool 3d features like hardening up the stl model 
slicing, stl model error feedback, model preview and fixing the stl 
loading bug. Ergobot showed me some 3d model smoothing algorithms that 
should allow me make the previews pretty fast. That was really the 
limiting factor behind 3d model previews and I'm excited to see if I can 
get them implemented.

Thanks,
Wes G.

jmkao, 2016-02-26:
Personally, the best mechanism I've experience so far for setting up a Raspberry Pi's wifi network is what Astroprint (https://github.com/AstroPrint/AstroBox) does.

Their print server has some code that will detect if the system has network connectivity, and if not, it will make the server act as its own AP (https://github.com/AstroPrint/AstroBox/blob/master/src/astroprint/network/debian.py).

Then they have has a part of their UI dedicated to selecting a client wifi network and configuring it for the RPi, similar to what we already have under the settings menu.

We could do something similar, or abstract out the "Create AP" logic to be in a standalone startup script that just does this at boot. This wouldn't be quite as dynamic as what Astroprint has, but it would mean that if your network required reconfiguration, then reboot the RPi to get it to broadcast a CWH AP, then log into the CWH to configure the network.

WesGilster, 2016-02-26:
Yeah, when I originally made the wifi manager, I found out it wasn't going to be much more setup to build an AP. I'm assuming they are just working with hostapd and then setup a captive portal for forwarding to their server? Very doable...

kloknibor, 2016-02-26:
Hmm I like that also better! Didn't think of that... But we do need to have an image than right? (Atleast for new users...) Shall I try to find out how this is possible and make it as small as possible so it fits on every SD card or is it just peanuts for you just need time to find a way to automate it :$?

WesGilster, 2016-02-26:
You wouldn't have to have an image, but that would be nice for first time users. I've never done it before, but as a part of our CI process it could just call dd to build the image, but I'm not sure if our CI platform allows those low level calls.

WesGilster, 2016-02-26:
While we are on this subject, what other features do we need before we release this to users as a beta? 
I know we need printer calibration. In what other ways do you guys have to modify the underlying slicing/machine configuration files?

jmkao, 2016-02-26:
I think XY sizes are the only thing, but I'm also using CWH to do the slice/exposure calculation using the calculators and not running CWS files. Not sure if what I've had to do also covers printing from CWS files sufficiently.

WesGilster, 2016-02-26:
So this is what I was thinking about:
1. Create an image as Kloknibor suggested that has the following software configuration preinstalled.
2. Raspberry Pi preconfigured for an open WIFI AP.
3. CWH installed.
4. A light DNS server that resolves all name queries to the Raspberry Pi IP.
5. A light http proxy server(perhaps the same software in step 4) that redirects all web requests to:
http://[raspberrypi]:9091/settingsPage

That solves the captive portal, but how does one get Windows(or any other OS) to open the browser when they click on the "Connect" button of our AP? I don't care where the user is directed(because every request will go to my settingsPage), but I just want the browser to open when the user clicks on the "Connect" button in Windows. Does Windows open the browser automatically on all open networks? I know this is possible because Belkin routers create the captive portal towards:
http://go.microsoft.com/fwlink/guestlogin.htm

Obviously the domain name doesn't matter in the above URL, only the path...

jmkao, 2016-02-26:
Hmm... building all of that from scratch seems like a big expenditure in effort since others have built it already and have stable code. Maybe, in a first phase, we do it based on a config file on the vfat /boot partition like @kloknibor suggested. Doing it that way, we could directly use the entire OS image for OctoPi:

https://github.com/guysoft/OctoPi

And just modify the launcher to launch CWH instead of Octoprint in our fork.

Then as another phase (as this will take some investigation), I can take a look at the license for Astroprint's OS level scripts (in particular their wifi_access_point script that does all of the work for setting up hostAP with a captive portal DNS server that will redirect to their UI). I backed them on Kickstarter and have one, and they have said in the forums that all their code is open source, but the OS level stuff isn't in their git repo.

kloknibor, 2016-02-26:
I think there have to become a better way of choosing printer with an according profile to just switch the resin profile but I haven't got a good suggestion... Will think about it... But it's kind off annoying and unneeded to start/stop the printer so much.... The printer connection will stay the same while changing resin ;)! 

I really enjoy it that everyone is throwing in their opinion about the wifi thing! Great guys! I think the access point solutions would be the most simple way for the users. I personally like to put the raspberry pi somewhere hidden in the printer and I move the printer around a lot... (Also between different locations). It would be nice if I didn't have to take the raspberry pi out every time ;)! But well we might have more problems... 

First : We need an wifi dongle which is able to create an AP right? Or is it possible to create an adhoc network with all wifi dongles?

Second : If we use normal adapters to create an ad hoc network we don't want it to leave in AP mode all the time because we want internet to update CWH. So how are we gonna put it in AP mode? Special button or are there other ways? One solution would be to only support wifi dongles which can create an acces point and connect to the internet at the same time (there are some but not much). But than we might want to put something like a password on the adhoc network for configuring CWH....

From what I read on the forums it's pretty easy (Well there are many different online examples) to input the wifi password by SD card. If you guys would like it I can choose one that works for me and make a wiki with it and do an pull request (Yes I started coding with github too :)!). But it isn't part of CWH so maybe it doesn't belong on github... Would be good to make some instructions on this too on our future website ;)! Maybe just leave it in the wiki and use external sources for now ;)! I think that we might just want to support as much as possible... Some people want to use an old desktop which doesn't have wifi and haven't easy acces to an LAN cable, than the SD card method might work... But I think this won't happen very much.... Tell me your opinions ;)!

jmkao, 2016-02-26:
All of the recommended Raspberry Pi dongles (like the official one or the EDIMAX ones) are based on the BCM4XXXX chipsets which support HostAP. If you're using something else, then it may not, but the EDIMAX dongles are pretty cheap...

kloknibor, 2016-02-26:
I'm using lb-link and TP-link at the moment but I wouldn't mind to switch to something else as long as it has an antenna... But I know one that supports hosting an acces point too :)! So no real problem for me ;)! I'll just buy something else ;)!

jmkao, 2016-02-26:
Actually, the EDIMAX dongles use RTL8188 based chipsets, sames as the TP-Link and do support HostAP.

kloknibor, 2016-02-26:
The one I use most has an RTL8192cu chipset and I see it supports it too :$ Thanks for pointing this out :)!

kloknibor, 2016-02-26:
Hmm I'm doubting if it does by the way... It says it Supports Ad-Hoc, Infrastructure WLAN network, Wireless roaming, Data rate auto fall-back under noisy environment or longer range distance, Site Survey with Profile function... But that's only ad-hoc and not AP right?

WesGilster, 2016-02-27:
> I think there have to become a better way of choosing printer with an according profile to just switch the resin profile but I haven't got a good suggestion...

I agree and this is pretty straight forward, I just need to have a drop down on the "basic printer setup" for Machine Configuration and Slicing Profile instead of it being a simple label. The same should be done for ink configuration. I opened a feature for this.

> But it's kind off annoying and unneeded to start/stop the printer so much.... 

I agree, I've been a bit annoyed by this, unfortunately it's the only thing that keeps a sane user experience as you scale the application. For example, imagine how insane it would be if you were hosting two live printers and 3 concurrent users are making modifications to a printer profile while it is in the middle of a print. We are annoyed because we are only thinking in a single user environment.

> Then as another phase (as this will take some investigation), I can take a look at the license for Astroprint's OS level scripts (in particular their wifi_access_point script that does all of the work for setting up hostAP with a captive portal DNS server that will redirect to their UI)

That would be great for you to point me to this and I could hook it up under the NetworkManager.

> If we use normal adapters to create an ad hoc network we don't want it to leave in AP mode all the time because we want internet to update CWH. So how are we gonna put it in AP mode? Special button or are there other ways?

Yes, that's correct, AP mode is designed for two situations:
1. People who aren't able to start with Ethernet first
2. When the Raspberry Pi determines that there isn't a network available(we've already tried bouncing the interface)

It's extremely easy to determine when a network isn't available, Java doesn't even need OS scripts for that. Once we determine that a network is unavailable, we'd ask the NetworkManager to go back into captive portal mode.

> Hmm... building all of that from scratch seems like a big expenditure in effort since others have built it already and have stable code. Maybe, in a first phase, we do it based on a config file on [...]

I suppose your right, even though I'd really like to build the captive portal, we should prioritize this. It would be really nice to hook our image creation into CI...

jmkao, 2016-02-27:
There's a lot we can do with CI, image creation is one thing. I also did some research on how to simulate serial ports, and we can probably incorporate testing of autodetection, with simulated projectors and firmware, into the CI as well. Doing it based on OctoPi should be relatively easy since their build system supports the idea of building "variants" with different stuff in it. Once we have that setup, it should be possible to later incorporate the wifi stuff into that.

WesGilster, 2016-02-27:
I noticed that as well, I didn't realize that these CI tools have gotten that advanced. In fact, have two test suites(mocked and hardware test suites) because I didn't think we would have this ability.

jmkao, 2016-03-06:
WTF... this is a lot harder than I thought it would be.

Turns out, making hostap work for RTL8188 devices is insanely hard given that the image is based on Jessie. Apparently the way everyone's been doing it for wheezy depends on one specific binary compiled patched version of hostapd as a magic step (e.g. https://learn.adafruit.com/setting-up-a-raspberry-pi-as-a-wifi-access-point/compiling-hostapd).

Since jessie uses linux kernel 4.2.x, the "magic binary" doesn't work and stuff would need to be compiled from scratch. The process is somewhat complicated and involves integrating a couple of different patches from a few different github repos and then building a kernel compilation environment on an RPi itself and compiling a bunch of stuff. I'm not even certain that the current state of the various patch projects is at the point where things could work. It seems that nobody has yet tried this on Jessie.

I've done this sort of thing before for x86 Linux to support weird devices but doing it on an RPi is painfully slow, especially given that I don't know that there is success waiting at the end without having to manually hack the driver code...

I think the best plan right now is to park this and as more people update Jessie, we'll probably see this work get done elsewhere and can incorporate it later.

WesGilster, 2016-03-06:
Shoot that's disappointing, but I think you are making the right choice to put this on hold. There are so many other things you can do. Are you planning on putting this together with a Raspbian build or pausing the OS image all together?

I realize that Java doesn't help this case given that we'd have to rely on hostapd anyway, this reinforces why I use Java, I really HATE dependency management

jmkao, 2016-03-06:
Only putting this part on hold. The Raspbian build image in #157 works well, and TBH, it's less steps overall to enter the wifi info in the text file than to connect to an  AP and do it. The advantage of the AP is that it's more error-resistant.

The AP path will also require additional security so that random people driving by your house can't connect to your printer and make it self-destruct...

WesGilster, 2016-03-06:
I do have basic auth with certificate generation setup so that should help for the authentication. However, there isn't any velocity checking in place, and you can't yet set the username password from the fat mounted disk, so it's not a real AP solution yet.

kloknibor, 2016-03-06:
Another pro for the AP is that integration in a product is much easier. I have the raspberry pi inside the metal casing of the printer with the wifiantenna sticking out. When I move it (it's very portable when there isn;t a laptop needed so I take it to other places as well) I don't have to get the whole printer apart to get to the microSD card.

kloknibor, 2016-03-06:
But however... Having the SD card wifi setup is already very awesome! Nice work! I'm testing your image now. Had some laptop problems... (accidently formatted my second drive and over wrote it with win32 with an raspbian image... So trying to save as much as possible...!)

kloknibor, 2016-03-11:
Does this help? : https://www.raspberrypi.org/forums/viewtopic.php?t=137932&p=920070#p919980

jmkao, 2016-03-11:
That's for the BCM43438 chipset, which is much less common than the RTL4x that is in the "hostapd that's been floating around on the Internet" that's mentioned.

What we need, is a version of that for Jessie.