area515/Photonic3D #149: Clicking the printer "Edit" button does nothing

jmkao, 2016-02-27:
Clicking the Edit button with a printer selected results in nothing happening. No errors or activity in the JavaScript console and no output in the cwh.log on the server.

The Create button opens the save printer dialog correctly, so I'm not sure why Edit doesn't work...

WesGilster, 2016-02-27:
This must be from my dev branch. I started integrating bootstrap.ui and broke some stuff. Sorry about that. I've fixed this and will check it back in tonight with all of my calibration code.

jmkao, 2016-02-28:
No prob, saving changes to slicing profiles also doesn't work. I'll do the stuff I was intending to test out of stable then. Thanks!

WesGilster, 2016-02-28:
Thanks again for testing this, I'm not sure how many times I would have broken prod... I checked in the saving fix last night, but I'm still testing. I'm slowly moving away from using functions on controllers towards using scope. The subtleties of this move is why all of this functionality broke.

WesGilster, 2016-02-28:
Fixed in cwh0.252.

jmkao, 2016-03-05:
@WesGilster Hmm... in cwh0.256, the Edit button still doesn't work. Create works properly (copying the existing config and is usable as a workaround, creating a new printer rather than editing the existing one), but clicking on Edit still results in nothing happening, either in the browser JS console or in the server log.

WesGilster, 2016-03-06:
Ouch. I'm sorry. It's about time I get protractor going on Angular, but there is always just one more feature I want to push in...

Fixed in cwh0.257. Let me know when I should modify my building habits.

jmkao, 2016-03-06:
Yay! This works now.

closed 2016-03-06 by jmkao

WesGilster, 2016-03-06:
I did it!  LOL. :) By "modify my building habits" I should have said when the build process is ready for tag and build increment.