area515/Photonic3D #150: Add 'limits and variables' card to printer setup screen

WesGilster, 2016-02-28:
I've never really understood the point of the build platform limits in CW until I'm ready to start putting together the 3d previews. Creating the 3d preview bounding boxes are only possible if you know the total size in millimeters of the build area. This will also serve as an alternate(actually superior) means of calibration, than what I just recently created. No real wasted time though since that code should be adopted to be more reusable anyway...

The variables part of the title will have some of the tilt variables/misc things that I don't really believe should exist, but will reintroduce for fear that someone will say such a feature "isn't supported". I'll also strikeout those variables to show they aren't needed and should use an alternate mechanism...

WesGilster, 2016-04-30:
I decided to call this "Advanced" printer setup card, but I haven't added the "tilt" and other variables to the UI card yet.

jmkao, 2016-04-30:
I think why some slicers want both pixels/mm and the physical bounding area is that some printers have apertures that only expose part of the printable projected image, perhaps to only print the brightest or most in-focus part of the projection region. Thus we may need both the purely pixel based calibration that you have today (which can be adjusted to fit within the aperture) and a measured maximum dimension that takes any physical limiters into account.

WesGilster, 2016-04-30:
Sounds good.

WesGilster, 2017-06-28:
This bug has long been completed. Both total pixels and pixels per mm are now supported both in calibration and manually setting them.

closed 2017-06-28 by WesGilster