area515/Photonic3D #152: Feature request: Support for Raspbian jessie

kloknibor, 2016-02-28:
I'm unaware of the workload needed to get this done but Jessie Light starts really quick on an raspberry pi 2! And it takes less space on the SD card :)! Would it be possible to understand raspbian jessie light too?

Kloknibor

WesGilster, 2016-02-28:
My guess is that there isn't any work involved in doing this. Our installs no longer even require raspi-config to be installed even when you haven't used the full SD card and want to use the raspberry pi camera.

I've worked darn hard to keep OS dependencies to the absolute minimum. There are probably a couple of other commands that are required but this is a pretty good list:
1. xserver/bash/ps (Linux only)
2. wget (Linux online install only)
3. sshd (Linux remote install only)
4. parted/fdisk/resize2fs (If installing on a Linux SD card and the partitions aren't fully used)
5. cut/grep/tail/readlink/sed/tar (Linux online/offline install only)
6. Java 1.8

If your Linux distribution doesn't have those, it's very unlikely you'd be wanting to run CWH on it.

Optional:
1. wpa_cli (If you want wifi on linux)
2. raspivid (or some other raw h246 capture app for video and timelapse)
3. raspistill (or some other image capture app for imaging and live streaming)

On a side note, are you keeping track of all of the stuff you've tested this on?  Because someday we might want to list all of our tested platforms...

kloknibor, 2016-02-28:
I just tried jessie light and it doesn't install... I installed with the old wget method from area515 the last error I found was something about an old java install? But while installing I saw you got some files from (external)links with (raspbian and) wheezy in it. This way I thought that maybe they would need to be changed ;)!

WesGilster, 2016-02-28:
It sounds like they don't have Java installed by default. Try java -version on the commandline and see what it says.

WesGilster, 2016-02-28:
Java wasn't installing because the URLs were wrong and the new RaspberryPi uses an arm7 instead of an arm6. This is all fixed in version cwh0.255, but keeping track of these Java install URLs are short lived. It's probably time we start looking into the CI image creation process.

jmkao, 2016-02-28:
I can take on CI image creation.

WesGilster, 2016-02-29:
Thanks a ton, that really helps out.

kloknibor, 2016-02-29:
may this be of any help? : http://www.aoakley.com/articles/2015-10-09-resizing-sd-images.php

kloknibor, 2016-02-29:
And I'll test raspbian jessie tonight ;)!

kloknibor, 2016-02-29:
still no luck... Log : 

pi@raspberrypi:~ $ sudo wget https://github.com/WesGilster/Creation-Workshop-Hos                                                                                                             t/raw/master/host/bin/start.sh
--2016-02-29 16:51:51--  https://github.com/WesGilster/Creation-Workshop-Host/ra                                                                                                             w/master/host/bin/start.sh
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/ma                                                                                                             ster/host/bin/start.sh [following]
--2016-02-29 16:51:52--  https://raw.githubusercontent.com/WesGilster/Creation-W                                                                                                             orkshop-Host/master/host/bin/start.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.31.19.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.31.19.13                                                                                                             3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5943 (5.8K) [text/plain]
Saving to: ‘start.sh’

start.sh            100%[=====================>]   5.80K  --.-KB/s   in 0s

2016-02-29 16:51:52 (24.9 MB/s) - ‘start.sh’ saved [5943/5943]

pi@raspberrypi:~ $ sudo chmod 777 start.sh
pi@raspberrypi:~ $ sudo ./start.sh WesGilster
Installing RxTx
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  librxtx-java
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 167 kB of archives.
After this operation, 848 kB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ jessie/main librxtx-java armh                                                                                                             f 2.2pre2-13 [167 kB]
Fetched 167 kB in 0s (204 kB/s)
Selecting previously unselected package librxtx-java.
(Reading database ... 30238 files and directories currently installed.)
Preparing to unpack .../librxtx-java_2.2pre2-13_armhf.deb ...
Unpacking librxtx-java (2.2pre2-13) ...
Setting up librxtx-java (2.2pre2-13) ...
Processing triggers for libc-bin (2.19-18+deb8u2) ...
No X server running, starting and configuring one
./start.sh: line 38: startx: command not found
./start.sh: line 39: xhost: command not found
mv: cannot stat ‘cwh-0.*.zip’: No such file or directory
Either Java is not installed, or an incorrect version of Java is installed. Inst                                                                                                             alling from this URL: http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8                                                                                                             u73-linux-arm32-vfp-hflt.tar.gz
rm: cannot remove ‘jdk-8u73-linux-arm32-vfp-hflt.tar.gz’: No such file or direct                                                                                                             ory
--2016-02-29 16:52:07--  http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jd                                                                                                             k-8u73-linux-arm32-vfp-hflt.tar.gz
Resolving download.oracle.com (download.oracle.com)... 95.100.96.210, 95.100.96.                                                                                                             177
Connecting to download.oracle.com (download.oracle.com)|95.100.96.210|:80... con                                                                                                             nected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: https://edelivery.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-                                                                                                             arm32-vfp-hflt.tar.gz [following]
--2016-02-29 16:52:07--  https://edelivery.oracle.com/otn-pub/java/jdk/8u73-b02/                                                                                                             jdk-8u73-linux-arm32-vfp-hflt.tar.gz
Resolving edelivery.oracle.com (edelivery.oracle.com)... 23.206.105.201
Connecting to edelivery.oracle.com (edelivery.oracle.com)|23.206.105.201|:443...                                                                                                              connected.
HTTP request sent, awaiting response... 302 Moved Temporarily
Location: http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jdk-8u73-linux-ar                                                                                                             m32-vfp-hflt.tar.gz?AuthParam=1456764848_00e05739e47e86c291a07bc4bf4d1ae7 [follo                                                                                                             wing]
--2016-02-29 16:52:08--  http://download.oracle.com/otn-pub/java/jdk/8u73-b02/jd                                                                                                             k-8u73-linux-arm32-vfp-hflt.tar.gz?AuthParam=1456764848_00e0 [... cut ...]
./start.sh: line 128: xset: command not found
./start.sh: line 129: xset: command not found
./start.sh: line 130: xset: command not found
Installing CWH as a service
cp: cannot stat ‘/opt/cwh/cwhservice’: No such file or directory
chmod: cannot access ‘/etc/init.d/cwhservice’: No such file or directory
update-rc.d: error: initscript does not exist: /etc/init.d/cwhservice
Determinging if one time install has occurred
grep: /root/3dPrinters/config.properties: No such file or directory
Starting printer host server

kloknibor, 2016-02-29:
Fixed it, Wesgilster isn't the appropriate command anymore for the repo... Other than that it's fine

kloknibor, 2016-02-29:
Pleasure replace WesGilster on the front page too : WesGilster/Creation-Workshop-Host in the installing from dev repo command ;)! 

It still gives me these errors I'm kind off worry about : 

Turning off screen saver and power saving
./start.sh: line 128: xset: command not found
./start.sh: line 129: xset: command not found
./start.sh: line 130: xset: command not found
 

other than that it works ;)!

kloknibor, 2016-02-29:
Grmpf sorry for spamming here... I get an error when using jessie light it is:

Server Error
Graphics environment not supported?

[LogBundle.pdf](https://github.com/area515/Creation-Workshop-Host/files/151387/LogBundle.pdf) (please rename the .pdf to .zip ... .zip won't work it gives me a error....)

jmkao, 2016-02-29:
Missing xset is weird since that's basic to the X server graphics, so I did a little more research on this. The TL;DR; is that Jessie Lite is a dead end.

The difference between Jessie and Jessie Lite is just that Jessie Lite does not include an X server for graphics. However, without an X server, we cannot draw images to project.  Thus to get Jessie Lite to work, you would need to install the Xorg packages and associated libraries. But that would turn Jessie Lite into a regular Jessie, and it would be easier just to start there.

It is true, however, that we do not need a window manager in order project images, just the bare X server itself. But that's mainly a startup script change, we don't get much savings from actually uninstalling all the window managers. I actually run this configuration on my Raspberry Pi, where I disabled the window manager, and this works fine.

I'll look at slimming down the system service startup as an optimization phase as I work on the adapting OctoPi to generate a CWH OS image.

WesGilster, 2016-02-29:
Yeah, xserver is number 1 on the dependency list, I suppose I could make that mechanism portal, but I'm not sure it's really a high priority.

Awesome, at first I was a bit concerned with starting with the OctoPi image which seems to have quite a few more dependencies than we actually need, but it seems like you are looking where you could slim it down.

kloknibor, 2016-03-02:
I tried running this on regular jessie but got also problems with the graphics... Will investigate further ;)

jmkao, 2016-03-05:
The OctoPi generated image in #157 is based on Jessie Lite with a small number of additional packages:

`apt-get -y --force-yes install xinit x11-xserver-utils oracle-java8-jdk`

Closing this issue, further work to be done on #157

closed 2016-03-05 by jmkao