area515/Photonic3D #154: Editing theme

kloknibor, 2016-02-29:
Hi,

I tried to make an custom theme. I added an copied version of resourcesnew under a new name to the root (where resources and resourcesnew are also located) and in config.properties I put the name of this folder under HostGUI this gives en not found page from jetty... What else do I need to change?

Furthermore I want to disable the DLPcontrols bootcard. What is the best way of doing this? Also I want to remove the X-axis and Y-axis control. This is unnecessary, I tried to just remove the list items but cwh started to behave weird... Can it be because I removed this list things? I think it can't be possible but not quite sure...

Thanks in advance!

Robin

WesGilster, 2016-03-01:
Awesome, that's cool. You've done everything you need to do to make a new GUI... I noticed you have a capital H on HostGUI and that isn't going to work. Keep in mind, details like that are important, I have to assume that you copied that straight from the config.properties file. Also, you'll need to restart CWH after you make the copy and change to config.properties. Make sure you've spelled everything case sensitive and that is all you need to do.

> Furthermore I want to disable the DLPcontrols bootcard.

Add this to the PrinterControlsController in printerControls.js
```
        $http.get('/services/settings/visibleCards').success(function(data) {
        	$scope.visibleCards = data;
        });
```
Add this:
`ng-show='visibleCards.indexOf("printers") > -1'`
as an attribute to line 90:
`	<div class="col-sm-4 bootcards-list" id="list">`
of printerControls.html and it will be hooked up to the config.properties file.

> 
What is the best way of doing this? Also I want to remove the X-axis and Y-axis control.

What you did will work fine, I just tried it. Make sure you are careful about what you are removing. My guess is that you left an unclosed div tag which basically breaks alot of stuff...

kloknibor, 2016-03-01:
Thanks for the quick reply! I'll try to work on it tonight ;)! will report back :)!

The H from HostGui was a small letter (just changed the text after it) So for some reason I made it capitol in here :P And I'm recompiling it, upload to my fork and restart the pi so that should be fine ;)!

kloknibor, 2016-03-01:
I posted the virst thing in printerControls.js here : 

(function() {
	var cwhApp = angular.module('cwhApp');
	cwhApp.controller("PrinterControlsController", ['$scope', '$http', '$location', '$routeParams', 'cwhWebSocket', function ($scope, $http, $location, $routeParams, cwhWebSocket) {
		controller = this;
		
		function attachToPrinter(printerName) {
			this.printerSocket = cwhWebSocket.connect("services/printerNotification/" + encodeURIComponent(printerName), $scope).onJsonContent(
				function(printerEvent) {
					controller.currentPrinter = printerEvent.printer;
				}
			);
			if (printerSocket == null) {
				$scope.$emit("MachineResponse",  {machineResponse: {command:"Browser Too Old", message:"You will need to use a modern browser to run this application."}});
			}
		}
		
		this.currentPrintJob = null;
		this.gcodeProcessing = "";
		this.gCodeToSend = "";
		this.squarePixelSize = 10;
		var printerName = $location.search().printerName;
		$http.get("/services/printers/get/" + printerName).success(
	        		function (data) {
	        			controller.currentPrinter = data;
	        		})
		$http.get("/services/printJobs/getByPrinterName/" + printerName).success(
        		function (data) {
        			controller.currentPrintJob = data;
        		})
        		
        	 $http.get('/services/settings/visibleCards').success(function(data) {
        		 $scope.visibleCards = data;
		 	});

but I'm doubting it's the right place... Is it? I can't compile right now to test since I'm on work :(

WesGilster, 2016-03-01:
that'll do fine.

kloknibor, 2016-03-01:
Removing the x y controls worked fine this time. So I may have broken something more back than ;)! But disabling the dlp card doesn't work... I did recompile it and uploaded it with eclipse but it still shows. (Yes after reboot ;)) I added it like I said above here and I added it to the div tag like you said as attribute... Any suggestions?

Maybe it's nice to make a "light" version of the new GUI. Disable the mail support, and printer settings. Even better would be an file in which all different bootcard can be enabled and disabled with ease, if more people agree I'll make a new request ;)!

kloknibor, 2016-03-07:
So we decided we don't want to remove DLP  controls all together since calibration frid and blank screen are still usefull... I just renamed it and deleted projector on|off and shutter control does work fine now!

closed 2016-03-07 by kloknibor