area515/Photonic3D #155: Can't create printers

kloknibor, 2016-03-01:
Hi,

I tried to run raspbian on an business network. It has some kind of security so it could be that it disables things needed for cwh. It does install fine on raspbian wheezy but there are no printer present  and clicking create doesn't do a thing... I couldn't find anything in the logs yet maybe you do?

the first one is from my own fork and the second one from wes's Dev fork.

Thanks in advance!

[LogBundle)own.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/152992/LogBundle.own.zip.pdf)
[LogBundlewes.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/152993/LogBundlewes.zip.pdf)

jmkao, 2016-03-01:
When you say business network, can you describe the network conditions at your site? Is a proxy server required to connect to the Internet? Is your browser able to connect to the internet?

The logs are pretty empty, they show a startup with no user activity, so it's possible that there is something related to the settings of the browser that is preventing you from accessing the UI properly. Those kinds of problems wouldn't be logged on the CWH side, but would need to be diagnosed from the browser.

kloknibor, 2016-03-01:
I was able to connect to the internet find cwh and download the files on raspberry by internet. But after your hint I tried a different browser than internet explorer and it worked, this can be closed ;)!

WesGilster, 2016-03-01:
There will certainly be some functions that won't work on some networks. For example many corp networks disable multicast, which we depend on to find all cwh instances operating on the network. Also many corp networks have a netmask of 255.255.0.0 and our cwh install  assumes a netmask of 255.255.255.0. This means that unless you happen to share the same third byte of your ip quartet with the Raspberry Pi, you won't find it.

kloknibor, 2016-03-01:
I understand. I saw it uses lots of ports too and some networks won't be to happy with it in sence of security I guess. But well it was just something in internet explorer and I don't think we need to investigate further ;)! I'll close this one

closed 2016-03-01 by kloknibor