area515/Photonic3D #157: Generate Full OS Image Installs

jmkao, 2016-03-02:
Create mechanism to automatically create installable Raspberry PI boot images that can be burned to SD to create appliances.

Reference ideas in Issues #147 and #152

WesGilster, 2016-03-02:
Awesome. I have a question on #147. How deeply is this going to be integrated into the image? Is it just going to ensure that hostap is already installed as an available package or is the AP captive portal going to be scripted in full?

jmkao, 2016-03-02:
That's more for the future, phase one will be based on a file in /boot the way OctoPi does it.

In the future, I think doing it the AstroPrint way would be the best. There probably does not need to be a captive portal, just that the default Apache is configured to mod_proxy from port 80 to port 9091 so people can access the UI w/o a port number.

Then on the AP network interface, put a name server that will make all names resolve to the router's IP, so any host you try to visit when attached to the AP will take you do the printer UI.

jmkao, 2016-03-02:
But the net is the image needs everything necessary for the whole setup workflow to work, from first boot to network configuration to printing.

WesGilster, 2016-03-03:
Yeah I asked specifically due to the port question. Pushing all of this through Apache/modproxy seems unnecessary. We could just set the port to 80 in config.properties since we are running as root.

I realize that Apache gives us a bit of flex, but I'm not sure why it's necessary. I suppose we could put together a redirect rule towards our settings page, but a jetty rule could be setup for the same...

Thanks,
Wes G.

jmkao, 2016-03-03:
Either way is pretty easy, and it would be easy to switch between the options, so I think the first priority is to pump out some form of image creation, and then we can hammer at the details once I have something basic working.

WesGilster, 2016-03-03:
True enough.

WesGilster, 2016-03-03:
By the way, what does the ready tag mean. Ready to work?

jmkao, 2016-03-03:
Yes, ready to work. Although more importantly, it is the basis for this dashboard here:

https://waffle.io/area515/Creation-Workshop-Host

Which helps me organize the things I'm ready to work on, things I am working on, and the things that are on the backlog that need to get into a ready-to-work state.

WesGilster, 2016-03-03:
Cool, a little kanban then?

jmkao, 2016-03-03:
I suppose so although I hesitate to use terminology that starts making this feel like my day job. :)

WesGilster, 2016-03-03:
LOL, I hear ya.

jmkao, 2016-03-05:
Launch image in qemu:

```
qemu-system-arm -kernel kernel-qemu -cpu arm1176 -m 256 -M versatilepb -no-reboot -serial stdio -append "root=/dev/sda2 panic=1 rootfstype=ext4 rw" -hda 2016-02-26-raspbian-jessie-lite.img -net nic -net user,hostfwd=tcp::9091-:9091
```

jmkao, 2016-03-05:
Image generation is working, with Wifi setup done the OctoPi way.

Download either a .zip of .7z compressed version at:

https://www.dropbox.com/sh/zlcn2ywya7h8uiq/AAAIlTyP3NgiQLWqqAbUHA2za?dl=0

Uncompress it and write it to an SD card the typical way (e.g. https://www.raspberrypi.org/documentation/installation/installing-images/windows.md)

Once done, mount the /boot partition of the SD card, which is a VFAT partition and should mount in any OS by  removing and reinserting the SD card.

Edit the `octopi-network.txt` file and set your wifi network settings by uncommenting the section appropriate to your network (WPA, WEP, or no encryption). Typically, most setups should have WPA encryption and thus uncomment and set the parameters in the section that looks like:

```
## WPA/WPA2 secured
iface wlan0-octopi inet manual
    wpa-ssid "MY NETWORK NAME"
    wpa-psk "MY NETWORK PASSWORD"
```

jmkao, 2016-03-05:
@kloknibor Can you have a go at this image and see if it works for you? I've tried it on an RPi 1B and it seems to work okay. The first time boot will take some time since it will resize the image to fill the SD card and then reboot, but it should simply work after that.

I'm going to continue on with the next round of enhancements to incorporate HostAP.

jmkao, 2016-03-05:
And, because I'm kind of OCD that way, the "slimming" optimizations are already done in this image, which is less than half the size of the OctoPi image.

kloknibor, 2016-03-05:
Sounds awesome James! But it's 1 AM here and I'm half a sleep... Gonna test it on the pi 2 in the morning :)! pi B+ to if I got time ;)!

Will report back thirst thing when I wake up! Goodnight :)!

WesGilster, 2016-03-06:
Ok... Now your just showing off. :) Nice job...

jmkao, 2016-03-06:
@WesGilster I have transferred the working repo for image generation to Area515.  However, I need to be made an Admin on that repo (https://github.com/area515/OctoPi) in order to integrate it into Travis-CI so I can trigger builds.

kloknibor, 2016-03-07:
... I LOVE IT Great job James... seriously... that booting time... it's just... wow! under 30 seconds will do an better timing... but it was way aboe 1 minute maybe 2...! jessie light with photonic3D rocks!

kloknibor, 2016-03-07:
Well it is 30-31 seconds till we can connect to to it... But still great!

WesGilster, 2016-03-07:
Let me know if you are admin now, I did this a while back and forgot to update this thread.

jmkao, 2016-03-08:
Yes, I am an admin now and have used those powers to begin integrating this with CI, which is a bit trickier because generating this image requires being able to become root, run qemu, and install some kernel modules. This can't be done inside of a Docker based cloud CI container. Travis can do this, but the delay time for starting non-Docker containers is long. SemaphoreCI has a VM based infrastructure that does work, and it starts builds pretty much instantaneously.

You can see the output of a run at:

https://semaphoreci.com/jkao/octopi/branches/devel/builds/12

With some compressibility enhancements (zeroing out unused filesystem blocks), the size is down from 420 to 350MB.

jmkao, 2016-03-09:
Successful builds will now autodeploy images to S3:

http://s3-us-west-2.amazonaws.com/photonic3d/photonic-image.zip

This download should be blazingly fast. URL has been added to the area515/OctoPi repo reade

WesGilster, 2016-03-09:
This is great. You've got a bunch of stuff going on here. You'll have to give me a bit of a workflow breakdown when you have stuff stable. No rush, by the way...

This is what I do now:

For a New Dev Build:
1. Make code changes to my WesGilster Repo
2. Run tests (or at least I should be running tests)  :)
3. Run Ant script.
      a. Ant script increments build number
      b. Ant scripts creates deployables
4. Check in everything with the first line of the commit being the build number in the zip file
5. Run sudo ./startdev.sh on my Raspberry Pi
6. Test my changes on the Raspberry Pi

Promotion To Prod:
1. Merge WesGilster Repo to area515 Repo

Force install from specific repo:
1. Run sudo ./downgrade.sh

jmkao, 2016-03-09:
I gave some more thought to this, I think that running CI in dev forks may not really be necessary so long as Github Releases in the forks are made easy in the build automation. So I would imagine something like:

User running CWH:

1. Run `start.sh`
1. start.sh goes to the latest release of area515/CWH, compares the version, downloads if newer

With CI on the area515 Repo, promotion to prod would look like:

1. PR or merge to area515/CWH master
1. `git tag <release number> ; git push --tags`
1. Wait for CI to complete
1. New Release appears and is ready to download (this can also trigger a new version of the RPi image)

The manual dev process would be:

1. Make code changes
1. `ant test` or `ant run` for local development
1. Commit local changes
1. `ant release <dev release tag>` - this magically builds, tags, and pushes to your repo, and uploads a GitHub release to your repo based on the tag
1. Run `start.sh <reponame>` like `start.sh WesGilster/Creation-Workshop-Host`
1. start.sh goes to the reponame in the argument and pulls the latest release, same logic as in the regular user case, but goes to the releases of the named repo rather than the default repo

There is one hitch to the manual dev process under our current build, which is making the `ant release` work the way I would like would require ant to spawn  OS binaries to complete work. In particular, it needs to call out to the git command line client, and aktau/github-release (https://github.com/aktau/github-release/releases/tag/v0.6.2) to be able to do the tagging, pushing, and release uploading. The means developers will need to make sure to download those tools and add them to the $PATH or %PATH% to release.

Unfortunately, there are no Github Releases automation plugins for ant, and the git plugins are pretty old and primitive. There are plugins for Gradle that do both of those things, but that would require a deliberate change in build system.

kloknibor, 2016-03-10:
I think it would be nice to make special roms for special printers. Like we just offer roms for specific printers and users don't need to touch anything for it :) And just an rom where they can set everything like we do now. What do you guys think?

jmkao, 2016-03-10:
I think customized images for printers is too heavy weight. Hosting lots of different 300-400MB images will be onerous for whoever has to do it, especially for only 1-2MB of differences. The online update capability for start.sh is crucial for keeping systems up to date, and I think that printer-specific configurations would be more easily and efficiently handled from an in-application online update capability.

I think centralizing that capability within the core product probably makes more sense rather than having people distribute a lot of externally assembled versions.

kloknibor, 2016-03-10:
Hmm allright I get what your saying but than an config pop-up at first boot would be an needed thing, so maybe that would be a good priority ;)! 

By the way, hosting large files wouldn't be a problem, we got 1TB. But still I get your point and totally agree with you ;)!

WesGilster, 2016-03-10:
Awesome, I'm much more attached to a speedy dev build process than I am attached to some technology.

jmkao, 2016-03-11:
Since I think 3 people have tested the image on a few different kinds of RPi's, I'm going to call this one done. We can track the build process enhancements on #143 and open new issues for any bugs.

closed 2016-03-11 by jmkao

kloknibor, 2016-03-11:
I got myself and one other man testing ;) I'm testing on 3 rpi as we speak ;)! (Not active use)

kloknibor, 2016-03-11:
butt all run on your image