area515/Photonic3D #159: Document How To Get Stable Wifi Setup

jmkao, 2016-03-04:


jmkao, 2016-03-04:
With a RPi 2B, stable setup via:

* Powered hub
* Plug EDIMAX, USB Camera, and RPi power source into hub
* Plug Projector & Firmware into RPi

Eliminate CH-341 UARTs, replace with PL2303 based devices (or on-chip at90usb1286 for firmware)

Need to test with RPi 1B

kloknibor, 2016-03-04:
I've got an pi B|B+|2B and I got an : Lb-link 150mbps Mini USB 802.11n/g/b Wireless Adapter with Antenna with an RTL8188CU chipset and I got 2A chargers. 

I'll run tests too. first all without raspberry pi cam. later on with because I suspect it makes it more unstable.

kloknibor, 2016-03-04:
I get an raspberry pi 3 next week :)! gonna test that too for stability than I think we got at all fine

kloknibor, 2016-03-11:
I haven't got the pi 3 so far. I did run 2 raspberri pi 2's for quite some days, one with the Lb-link 150mbps Mini USB 802.11n/g/b Wireless Adapter with Antenna with an RTL8188CU chipset and one raspberry pi 2 with Edimax EW-7811Un. both still run absolutly fine now ;)! Still have to test with raspberry pi cam tough

kloknibor, 2016-03-11:
The one with LB link died today after 7 days. But there isn't any power on the usb dongle at this point (no green led). And I didn't feel like taking the printer apart tonight.. Maybe some wire came loose internal... Will report back later

kloknibor, 2016-04-16:
Regarding the topic of wifi dongles I would advice the Edimax EW-7811Un even without that large antenna it has en better reception and is 100% stable for me ;)! The Lb-link one works... But can crash and signal strength is bad... (Guess that's what you get with chinese cheap dongles ;))

WesGilster, 2016-09-18:
I assume this is long fixed. Reopen if not.

closed 2016-09-18 by WesGilster