area515/Photonic3D #161: Unable to upload files to CWH

XiGua89, 2016-03-04:
I not sure whether this is the right place to post my issue.
I successfully installed CWH on the RPi 2. I am able to connect to my printer and move the build-platform. I also ran some test-prints without resin some days ago and everything seemed to work just fine.

Now that my resin arrived I am unable to upload files anymore! 2 hours ago it still worked, now: nothing. I tried several files (.cws maily, but also .stl, .jpg and URLs) and also tried those with different names. Files that I was able to upload before just do not anymore.
I am uploading the files from a windows-computer (tried different computers with different browsers as well)

Any help/advice would be greatly appreciated
XiGua

kloknibor, 2016-03-04:
Hi,

your at the right place! Did you try to reboot? maybe something as lagging... when you upload the files it won't show an progress bar but will wait till it's uploaded and show it in the interface! So be patient. If that also doesn't work please provide us with the log files, in settings you can add your mail adress (only tested with gmail and you need to enable use on weak secured devices) so we can figure out what's going on. for support mail just fill in your own mail and upload the .zip files containing the logs you get sended on here (just drag it into the comment field)

XiGua89, 2016-03-04:
Thanks for your fast response!
I rebooted several times and also waited for the uploads

I am having some trouble with the right settings for the support, so i will try it over the weekend and post it asap
XiGua

kloknibor, 2016-03-04:
Did you try it with an gmail account? go first here : http://www.google.com/settings/security/lesssecureapps : enable acces to apps with lower security. Than enter for username your mail where you want the logs, for support mail your gmail mail. Password is your password gmail. you can leave the outgoing server as is. and your mail your gmail mail also :)!

XiGua89, 2016-03-07:
i can not get get it working (using gmail). this error appears:
_Failure emailing log bundle:Could not connect to SMTP host: Smtp.gmail.com, port: 0 _

do i need to configure the wifi setup on the lefthand side? Because i can not select one here

kloknibor, 2016-03-07:
Did you fill in as follows?:
[image]

the password you need to fill in is from Yourgmailadres@gmail.com
I think you filled out your smtp adress wrong, it should be : smtp.googlemail.com

Did you also already allow acces to less secure apps like I suggested? And you do not need to set wifi ;)!

XiGua89, 2016-03-07:
first of all, thanks a lot for all your help! i really appreciate it

yes, i turned on access for less secure apps.

and i tried it again with _smpt.googlemail.com_ (everything else like you posted, except for _Username_, here i tried using an outlook-mail or the same gmail like _E-Mail Address_ and _Support Email Adress_ )

now i am getting: _Failure emailing log bundle:Unknown SMTP host: smpt.googlemail.com _

XiGua89, 2016-03-07:
I just noticed you are working with CWH v0.240, mine is CWH v.0.227

Could this be related with my problem?
Also, i believed the CWH onthe raspberry auto-updates?

kloknibor, 2016-03-07:
e-mail adress and username both should be an gmail adress and the password should be from this e-mail adress ;)! the supportmail can be an outlook adress no problem. I just retried it with this smtp host but I can send it without problems... And yes I use a lot of versions (dev from Wes, Stable from area 515 and dev from own repo but all version can send mail without problems ;)! All previous versions worked for me too... So it should really be the settings... And yes it auto updates on reboot!)

kloknibor, 2016-03-07:
(Username and e-mailadress should both be THE SAME adress)

ergobot, 2016-03-07:
@XiGua89 - I'm not sure if I read this right or a typo but I think it's
smtp instead of smpt.

"now i am getting: _Failure emailing log bundle:Unknown SMTP host:
smpt.googlemail.com"

kloknibor, 2016-03-07:
good catch @ergobot

XiGua89, 2016-03-07:
@ergobot - indeed a typo :)

i am using my gmail for all 3 mails now, smtp without typo, correct password and get the same error:
_Failure emailing log bundle:Could not connect to SMTP host: smtp.googlemail.com, port: 0_

i really do not know what to do anymore :/

kloknibor, 2016-03-07:
would you mind to show us an screenshot of how you filled everything in? If you don't want to post it here mail it too : robinkolk@msn.com

kloknibor, 2016-03-07:
I just tried to use it with all 3 the same gmail adress and it worked without a problem... please copy this to smtp : smtp.googlemail.com 

the others you can use the same gmailadress for... Make sure your password is right and you did type in your gmail mail correctly

kloknibor, 2016-03-07:
also tried it with 0.227... also no problems at all so it isn't the software...

XiGua89, 2016-03-07:
i just sent you a screenshot, hope i did sth wrong :)

kloknibor, 2016-03-07:
I wish I could said that too you too... it all seems perfectly fine as long as you are sure your password is correct... It isn't that you enabled acces to less secure apps wrong because it will give an error like this:
Failure emailing log bundle: Username or password incorrect

You pressed save after you set this settings? maybe refresh after clicking save... (this was needed when you enable acces to less secure apps)

XiGua89, 2016-03-07:
yes, i saved and refreshed

anymore ideas? :/

kloknibor, 2016-03-07:
yes... complete reinstall... sorry...! I can reach out a bit to you... download this image : https://www.dropbox.com/sh/zlcn2ywya7h8uiq/AAAIlTyP3NgiQLWqqAbUHA2za?dl=0 and it should work out of the box, it's raspbian jessie light with CWH preinstalled made possible by jmkao! However... It is also an unstable(r) dev branch... Yesterday it worked for me but haven't got an chance to test it properly myself

XiGua89, 2016-03-07:
ok, so i just installed it, printer is running again, i can control the motor etc.

uploading cws files is still not working, but sending the support-mail did! ( same settings as used before...)
[LogBundle.zip](https://github.com/area515/Creation-Workshop-Host/files/161175/LogBundle.zip)

kloknibor, 2016-03-07:
Ok I guess I know your last problem :)! gonna look for the clip but I suppose you didn't anbale export to cws ;)!

kloknibor, 2016-03-07:
do .stl files or .txt do upload?

kloknibor, 2016-03-07:
follow this instruction video : https://www.youtube.com/watch?v=J3HTCkxlKcw

XiGua89, 2016-03-07:
hm, i am quite sure i did. i used compound path (but i am not sure what this one does or is good for)

no, .stl, .cws, .txt all dont work

kloknibor, 2016-03-07:
Oh never mind... when .stl and .txt don;t work too my instruction video won't help... What error do you get while uploading? I haven't found anything strange yet in the logs... The gcode works fine with your printer and I see you had an error when trying to mail again :P Not seen anything strange other than that

kloknibor, 2016-03-07:
Ok so the error is in here somewhere : 
`
(HttpChannel.java:311) [jetty-all-9.2.3.v20140905.jar:9.2.3.v20140905]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:248) [jetty-all-9.2.3.v20140905.jar:9.2.3.v20140905]
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540) [jetty-all-9.2.3.v20140905.jar:9.2.3.v20140905]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:610) [jetty-all-9.2.3.v20140905.jar:9.2.3.v20140905]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:539) [jetty-all-9.2.3.v20140905.jar:9.2.3.v20140905]
	at java.lang.Thread.run(Thread.java:745) [?:1.8.0_65]
**2016-03-07 11:03:11,530 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp18312385-18] Accepting new printable testtest.cws as a Creation Workshop Scene
2016-03-07 11:03:11,608 INFO o.a.r.s.PrintableService [qtp18312385-18] File saved to:/tmp/uploaddir/testtest.cws**
2016-03-07 11:03:11,620 ERROR o.a.u.ExceptionMarshaller [qtp18312385-15] Error caught by exception marshaller and relayed to browser
java.lang.IllegalArgumentException: MALFORMED
	at java.util.zip.ZipCoder.toString(ZipCoder.java:58) ~[?:1.8.0_65]
	at java.util.zip.ZipFile.getZipEntry(ZipFile.java:566) ~[?:1.8.0_65]
	at java.util.zip.ZipFile.access$900(ZipFile.java:60) ~[?:1.8.0_65]
	at java.util.zip.ZipFile$ZipEntryIterator.next(ZipFile.java:524) ~[?:1.8.0_65]
	at java.util.zip.ZipFile$ZipEntryIterator.next(ZipFile.java:480) ~[?:1.8.0_65]
	at java.util.Spliterators$IteratorSpliterator.tryAdvance(Spliterators.java:1812) ~[?:1.8.0_65]
	at java.util.stream.ReferencePipeline.forEachWithCancel(ReferencePipeline.java:126) ~[?:1.8.0_65]
	at java.util.stream.AbstractPipeline.copyIntoWithCancel(AbstractPipeline.java:498) ~[?:1.8.0_65]
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:485) ~[?:1.8.0_65]
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471) ~[?:1.8.0_65]
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:230) ~[?:1.8.0_65]
	at java.util.stream.MatchOps$MatchOp.evaluateSequential(MatchOps.java:196) ~[?:1.8.0_65]
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234) ~[?:1.8.0_65]
	at java.util.stream.ReferencePipeline.anyMatch(ReferencePipeline.java:449) ~[?:1.8.0_65]
	at org.area515.resinprinter.job.CreationWorkshopSceneFileProcessor.zipHasGCode(CreationWorkshopSceneFileProcessor.java:275) ~[cwh/:?]
	at org.area515.resinprinter.job.CreationWorkshopSceneFileProcessor.acceptsFile(CreationWorkshopSceneFileProcessor.java:46) ~[cwh/:?]
	at org.area515.util.PrintFileFilter.findAssociatedPrintProcessor(PrintFileFilter.java:24) ~[cwh/:?]
	at org.area515.resinprinter.services.PrintableService.getPrintables(PrintableService.java:207) ~[cwh/:?]
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:1.8.0_65]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:1.8.0_65]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_65]
	at java.lang.reflect.Method.invoke(Method.java:497) ~[?:1.8.0_65]
	at org.jboss.resteasy.core.MethodInjectorImpl.invoke`

I think Wes or James can help you better on this...

XiGua89, 2016-03-07:
i do not get any error. its just that nothing happens, even after waiting for a while.

could you upload an easy .cws file so i can test it? (maybe i did do sth wrong while slicing!)
or do you think even if .txt wont work, there has to be another issue?

kloknibor, 2016-03-07:
If .txt doesn't work than there's something else wrong

kloknibor, 2016-03-07:
[test.cws.pdf](https://github.com/area515/Creation-Workshop-Host/files/161201/test.cws.pdf) delete the .pdf from the file name and your good to go! This one works fine here...

XiGua89, 2016-03-07:
ok, i will be patient and hope they can help:)

your file also does not work, or at least it doesnt appear...

kloknibor, 2016-03-07:
can you supply more infortmation about the pi, wifi dongle (if you use one), power supply, camera (if any) and printer you use?

XiGua89, 2016-03-07:
it is a selfmade DLP-Printer like the Chimera on insructables (by mastermind): 
- Raspberry Pi 2 Model B
- bipolar stepper motor of a disc drive
- Arduino Uno with GRBL + EasyDriver V4.4
- Arduino is powered by USB of RPi; Easydriver by Arduino (since the stepper and my buildplatform are really small, it has enough torque for now)
- RPI connected via ethernet cable
- no camera
- DLP-Projector is connected to RPi via HDMI-cable (Question: how can i turn on/off the projector using CWH? Do i need further USB-Connection?)

kloknibor, 2016-03-07:
Hmm sounds allright... A little concerned about the power drainage of the stepper and arduino and from the pi... might introduce unstabelity problems but well you could move your motor now... Well Let's wait for the others :)! And which DLP projector you got? You need special hexcodes send over serial connection to your DLP most of the time, these hexcodes can be set in the config files ;)! But you also need the serial connection to your beamer

XiGua89, 2016-03-07:
i am thinking about powering at least the motorshield externally in future steps, but for now i would already be happy to perform my first print :)
It is a Optoma HD141X (i updated the list above)

which config files do you mean?:)

kloknibor, 2016-03-07:
I think it was config.properties you can find this there : hexCodeBasedProjectors=[{"name"\:"Acer","onHex"\:"4F4B4F4B4F4B4F4B4F4B0D", "offHex"\:"2A2030204952203030320D", "detectionHex"\:"2A2030204C616D70203F0D",

these are the hexcodes I described you must find them for your projector yourselves ;)! And connect it to the pi with an USB serial converter. the file is by default available in /opt/cwh/

XiGua89, 2016-03-07:
alright i will try at some point

i was just able to upload your test.cws. other files did somehow not work
now it seems cwh lost the file again and i can not upload it anymore

kloknibor, 2016-03-07:
very strange... files do disappear after an reboot of the pi... but it should be possible to reupload the file...

kloknibor, 2016-03-07:
try it in another browser... try chrome! I had some issues with that as well

XiGua89, 2016-03-07:
it doesnt work

jmkao, 2016-03-07:
@XiGua89 Thanks for working with us on this!

It looks like projector autodetection is not working, so for now you should disable it and configure your printer like this:

[image]

Are you by any chance located in China? If so, then accessing Google for e-mailing LogBundle.zip will only work intermittently and it would probably be easier to harvest and upload the logs directly. You can also use SCP (e.g. FileZilla) to get the log off the Raspberry Pi at /opt/cwh/cwh.log. Then attach that file directly to the issue. If you're not in China, than once you have the LogBundle.zip email working once, it should keep on working and we can use that. LogBundle.zip contains additional information besides the logs, and you should continue using that if possible.

I would like to see the .cws file you are using, because according to the log the zip on that is corrupted. I have seen that happen before, where CWS saves a corrupted .cws and there is another working file called .cws###### instead. I would check the location where you are saving .cws files to see if there are those numbered/lettered temp files also in that directory.

The easiest way to attach a .cws file is to zip it (e.g. right-click --> Send To --> Compressed (Zipped) Folder on Windows) and then upload the zip file.

Also, what version of Creation Workshop are you using to generate the .cws file?

 I'm looking forward to finding a way to get this working for you!

XiGua89, 2016-03-07:
Hello jmkao,
thanks a lot for helping here, this community is great!

i am not located in china and so far the support function seems to work.

I did not find anything else in the location of my cws-files are saved in (i also tried uploading from the desktop).
I am using Creation workshop 1.0.0.75
This is my calibration Cube i am planning for my first print:
[wuerfel.zip](https://github.com/area515/Creation-Workshop-Host/files/161649/wuerfel.zip)

XiGua89, 2016-03-07:
i forgot to mention: i can not configure the printer like you said, following message appears:
_No serial ports found for:Autodetect 3d printer firmware_

WesGilster, 2016-03-07:
I don't want to interrupt the stream of troubleshooting going on because everyone is doing great. I just want to take the time to answer questions and put together an assessment here:

> and after the reinstall i am working with CWH v0.227 again, is that normal?  
Yes this is normal because v0.227 is the latest production release. v0.240 is a development stream release.

> very strange... files do disappear after an reboot of the pi... but it should be possible to reupload the file...
I've never experienced this issue and used to think Kloknibor was a bit crazy. Most likely the reason is 1 of 2 different possibilities, the pi is deleting due being in the tmp directory(I thought we were saving in a safer place), or the file wasn't closed properly on upload and JVM kills all open files on improper exit of the JVM. (Rebooting the pi is considered an 'improper' exit of the JVM) My money is on the second one. I can take a look and try to get a fix in the dev stream.

> No serial ports found for:Autodetect 3d printer firmware
Generally this is because your com port is already in use. We'll be able to tell for sure if you send us a log bundle.  Always try to associate a logbundle with each issue to comment on github if it's a new issue. Basically that bundle describes everything going on inside of CWH.

Is this a full assessment of open questions you have?

XiGua89, 2016-03-07:
Hello Wes, thanks for joining, too

yes, i guess so. besides mentioning the original problem that i can not upload anything.

here is a new log bundle concerning the autodetection of the printer firmware:
[LogBundle_autodetect.zip](https://github.com/area515/Creation-Workshop-Host/files/161799/LogBundle_autodetect.zip) (i also tried uploading _wuerfel.zip_ on this one)
I would not know how my com port could already be in use?

also i want to apologize in general for any unspecific or insufficient information. It must obviously be due to my lack of knowledge in this matter

WesGilster, 2016-03-07:
No problem at all, remember this is beta software so it's not all 'user error'.

Actually I missed one more...

> Projector autodetection

Once we've got your other issues resolved, we'll address this one. As the others have already alluded, you'll need a serial cable to connect your projector, and we'll also need to know where to find your projector serial codes if you have them handy.

> I can't upload anything

I thought this was resolved, is this not true.

XiGua89, 2016-03-07:
> we'll also need to know where to find your projector serial codes if you have them handy.

ok, i did not look for the control commands for my projector yet (i hope this is what you ment)

> I thought this was resolved, is this not true.

no, unfortunately it is not true. i am still not able to upload files (no matter what file). like i posted, it seems to work sometimes (very very rare), but i could not detect any "pattern"

jmkao, 2016-03-07:
So far, there are two problems:

1. wuerfel.cws unzips fine with the Windows built-in zip and with 7zip, but Java does not like something about this file. I get the same exception against my local version of 0.227

1. The presence of the "corrupt" file in /tmp/uploaddir prevents any future uploads from working. This is related to #119. Since acceptsFile() is run constantly on all of the files in /tmp/uploaddir, the "corrupt" file will cause exceptions to repeatedly occur whenever loading the Printables page or trying to upload anything. This will effectively render all interaction with the Printables tab inoperable under the user manually removes the offending file from /tmp/uploaddir. The user cannot use the UI to remove the file because the UI list of printables depends on acceptsFile(), thus once a problematic file reaches /tmp/uploaddir, the list of printables will be empty.

One possible hint is that ZipFile seems to behave differently depending on the Charset that is set (http://stackoverflow.com/questions/20013091/java-unzip-error-malformed), so perhaps using US-ASCII instead of UTF-8 will help.

The problem with the zipfile seems to have something to do with the names of the files in the zip, and not with the zip encoding itself. Even if I unpack and repack the zip file on Windows, the file either continues to fail to upload, or gets the same MALFORMED exception later when I try to print it.

@XiGua89 Most likely, you can get this file to work with a  workaround, which would be to save file file in CW as a different name. rather than calling it "Wuerfel 10mm", just name it simply "wuerfel" or something like that, and verify that the contents of the cws zip (rename the .cws to .zip and then open it in the Windows explore) has simple file names like "wuerfel0001.png".

jmkao, 2016-03-07:
If we instantiate ZipFile objects with a charset for ISO-8859-1 or CP437, like:

```
new ZipFile(jobFile, Charset.forName("ISO-8859-1"));
```

then this works.

Reading this article: https://blogs.oracle.com/xuemingshen/entry/non_utf_8_encoding_in, I think that CP437 would be the right value to match the zip spec.

WesGilster, 2016-03-08:
@jmkao nice job. Did you replicate the disappearing file issue as well, or does this seem to be a different issue?

I think I have a pretty good handle on the:

> No serial ports found for:Autodetect 3d printer firmware

2016-03-07 17:18:14,220 DEBUG o.a.r.s.SerialManager [qtp18312385-26] Chitchat was:
Grbl 0.9i ['$' for help]
 from:{Port:/dev/ttyACM0 speed:115200 databits:8 parity:None stopbits:One handshake:None}
2016-03-07 17:18:16,439 DEBUG o.a.r.s.SerialManager [qtp18312385-26] Unknown data:[ok
, ok
] received on:{Port:/dev/ttyACM0 speed:115200 databits:8 parity:None stopbits:One handshake:None}
2016-03-07 17:18:16,443 ERROR o.a.r.p.PrinterManager [qtp18312385-26] Error starting printer:MST
org.area515.resinprinter.display.InappropriateDeviceException: No serial ports found for:Autodetect 3d printer firmware

It's odd because it says "unknown data" but clearly the data is known and "ok" what I'm guessing happened is that Autodetect isn't liking the \r at the end of the ok. This is a quick fix to make this process more robust, but the current workaround is to stop using Autodetect 3d printer firmware and go back to using /dev/ttyACM0 direct without projector autodetection.

jmkao, 2016-03-08:
I'm fairly certain the disappearing file issue has to do with some kind of /tmp management. It's not as ephemeral as restart because Raspbian does not seem to use tmpfs for /tmp, but there must be something somewhere that erases out of there on some rule or schedule.

I'm not sure it's worth tracking down exactly how Raspbian does this, because regardless of how that happens, the solution for us is probably to move persistent storage out of /tmp.

The bigger and harder issue I think is the problem with corrupt files rendering the Printables tab inoperable because of how acceptsFile() constantly iterates all the files in the directory. Forcing people to SSH into the box to do cleanup feels like it should be a last resort.

WesGilster, 2016-03-08:
I agree, if it's tmp management, then it's pretty simple.

>Forcing people to SSH into the box to do cleanup feels like it should be a last resort.

Oh my gosh no, that would be awful.

I'm not at all convinced that acceptsFile() is somehow corrupting the file. Your zipHasGCode() method is pretty solid(closes nicely) and it's a read only method that doesn't perform any file extraction or write procedures(we perform all of that in the prepare environment). Inefficient, yes because it opens the zip file and loops through the zip meta data on every accept call. However there is no directory extraction process at work there.

I'd like to take a closer look over the area that was rendering the Printables tab inoperable, where did you find that?

jmkao, 2016-03-08:
Oh no, acceptsFile() is definitely **not** corrupting the file! The issue is that because acceptsFile() is being invoked for all the files in the directory constantly, if there is a single corrupt file in the directory, the whole page will fail.

I think the issue is in this call sequence starting with the REST endpoing for listing printables in PrintableService.java:

```
	@GET
	@Path("list")
	public List<Printable> getPrintables() {
    	File dir = HostProperties.Instance().getUploadDir();
		File[] acceptedFiles = dir.listFiles();
		ArrayList<Printable> printables = new ArrayList<Printable>();
		for(File file : acceptedFiles) {
			PrintFileProcessor<?> processor = PrintFileFilter.INSTANCE.findAssociatedPrintProcessor(file);
			printables.add(new Printable(file, processor));
		}
		
		return printables;
	}
```

This calls findAssociatedPrintProcessor() in PrintFileFilter.java for each file in the /tmp/uploaddir directory:

```
    public PrintFileProcessor<?> findAssociatedPrintProcessor(File pathname) {
 		for (PrintFileProcessor<?> currentProcessor : HostProperties.Instance().getPrintFileProcessors()) {
 			if (currentProcessor.acceptsFile(pathname)) {
 				return currentProcessor;
 			}
 		}
```

When then shops out the file to every processor.

If the acceptsFile() for a processor throws an exception (e.g. in the case of a corrupted zip), then that exception will percolate all the way back to the REST endpoint and cause the listing to fail.

That means that any bad files in /tmp/uploaddir will cause the printables list to be blank.

We can probably reduce the impact of this by adding some exception handling so that bad files will be skipped, but that would still leave the bad file invisibly there and un-removable.

Perhaps there should be some kind of last-resort fallback Processor that gets assigned to the file in a catch block, like "UnprintableFileProcessor" that has enough information to make the file show up in the list (so that it can be removed), but otherwise doesn't do anything.

WesGilster, 2016-03-08:
Let's take another recap of issues:
>very strange... files do disappear after an reboot of the pi... but it should be possible to reupload the file...

If this is an OS based tmp file cleanup issue, this can be resolved by adding:
`uploaddir=somedirectory`

>No serial ports found for:Autodetect 3d printer firmware

**Temporary solution**
In the "Basic Printer Setup" choose /dev/ttyACM0 instead of using "Autodetect 3d printer firmware". 
**Permanent solution**
Code fix is needed to make autodetection more robust for GBRL.

>I can't upload anything

**Temporary solution**
You can get this file to work with a workaround, which would be to save file file in CW as a different name. rather than calling it "Wuerfel 10mm", just name it simply "wuerfel" or something like that, and verify that the contents of the cws zip (rename the .cws to .zip and then open it in the Windows explore) has simple file names like "wuerfel0001.png". Then, SSH into your Raspberry Pi and erase all the files in /tmp/uploaddir. Finally, upload the cws file and let us knows what happens. If there is still a problem, please attach the new cws file to your update.
**Permanent solution**
jmkao has a solution he's going to be building a permanent fix into the area515 stream.

>File corruption issue

@jmkao Ahh, I see what you are saying, corruption is due to "bad" zip file. Printables page disablement is due to corruption. Gotcha. Your solution isn't bad, but I'm wondering if we shouldn't just disallow the file to be uploaded if it doesn't match any processor(I thought we were doing that anyway) I like preventative measures if possible. I realize things could still "sneak" through, but our acceptsFile() should be more robust anyway. This would have also given XiGua89 some feedback so that he knew that his file was "corrupted" and couldn't be printed.

>ok, i did not look for the control commands for my projector yet (i hope this is what you ment)

Yep, that's what I meant, we aren't ready for this yet, but will be when you're comfortable with the other issues, you'll need a serial cable, those codes and we'll give you a program that you can run to help us test your hex codes.

jmkao, 2016-03-08:
Yes, I guess a higher up line of defense would be to proactively delete the file at the time of upload if there are any exceptions in the processing chain.

WesGilster, 2016-03-08:
Yeah that's what I'm thinking.

WesGilster, 2016-03-08:
Actually, furthermore it should delete the file at the end of the upload process and return a failure on the upload process itself if there isn't a printFileProcessor that is willing to take ownership of the file.

XiGua89, 2016-03-08:
wow, you guys are amazing!
unfortunately i can not try anything today, but tomorrow i will for sure

> **Temporary solution**
You can get this file to work with a workaround, which would be to save file file in CW as a different name. rather than calling it "Wuerfel 10mm", just name it simply "wuerfel" or something like that,

I did that before and also tried kloknibor's _test.cws_

> and verify that the contents of the cws zip (rename the .cws to .zip and then open it in the Windows explore) has simple file names like "wuerfel0001.png". 

ok!

>Then, SSH into your Raspberry Pi and erase all the files in /tmp/uploaddir. Finally, upload the cws file and let us knows what happens. If there is still a problem, please attach the new cws file to your update.

ok, i will definitely try that asap (tomorrow)!

jmkao, 2016-03-08:
@XiGua89 Actually, you do not need to do any of the workarounds. I put a fix into stable build 0.228, so just restart your CWH and it should update itself automatically.

XiGua89, 2016-03-09:
wow, this would be great.
actually i am not sure how to restart CWH. I can only disconnect the RPi from power and connect it again (when i disconnect a printer in CWH, i only see black screen with a strange symbol of my mouse. therefore i can not shut it down/reboot)

I disconnected the RPi from power supply twice, but still get v0.227

kloknibor, 2016-03-09:
It's easiest to disconnect the power yes :P (there are more gentle ways but involve SSH or keyboard acces) The black background is normal ;) even when diconnecting the printer. Jmkao introduced it so that you won't cure too much when no printer is connected is my guess ;)! The simble you see, could it be an trash icon? Becouse the mouse should be disabled by the software...

[46 more replies]

closed 2016-03-17 by WesGilster