area515/Photonic3D #165: Disable gradient bulb center

kloknibor, 2016-03-06:
Hi,

I'm trying to disable the gradient transparancy thing... It's unneeded for LCD and makes my print fails... I use CWS files but can't get this off... Tried to set the transparancy to 0.0 instead of 0.2 under gradient but didn't work.

Robin

kloknibor, 2016-03-06:
[image]

this is how it looks...

kloknibor, 2016-03-06:
I guess I need to change something here : 

var bulbCenter = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var bulbFocus = new Packages.java.awt.geom.Point2D.Double($buildPlatformXPixels / 2, $buildPlatformYPixels / 2);
var totalSizeOfGradient = $buildPlatformXPixels > $buildPlatformYPixels?$buildPlatformXPixels:$buildPlatformYPixels;
var fractions = [0.0, 1.0];
//Let's start with 20% opaque in the center of the projector bulb
var colors = [new Packages.java.awt.Color(0.0, 0.0, 0.0, 0.2), new Packages.java.awt.Color(0.0, 0.0, 0.0, 0.0)];
new Packages.java.awt.RadialGradientPaint(
	bulbCenter,
	totalSizeOfGradient,
	bulbFocus,
	fractions,
	colors,
	java.awt.MultipleGradientPaint.CycleMethod.NO_CYCLE)

I already tried to change the 0.2 to 0.0 doesn't work for me so far :(

kloknibor, 2016-03-06:
fixed it... Had to press save when printer is disabled... Sorry for bothering!

closed 2016-03-06 by kloknibor

WesGilster, 2016-03-06:
No problem. I believe I've tested this with no formula at all. That's most likely what you want. If that doesn't work it's a bug.