area515/Photonic3D #168: Long waiting before prints starts.

kloknibor, 2016-03-07:
Hi,

With the photocentric printer CWH waits quite a long time and I think I know the cause... Here is the error piece : 
``
2016-03-04 19:09:12,109 INFO o.a.r.s.PrintableService [qtp4990863-17] File saved to:/tmp/uploaddir/gehause.stl
2016-03-04 19:09:51,473 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-83] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:09:51,478 INFO o.a.r.s.PrintableService [qtp4990863-83] File saved to:/tmp/uploaddir/kaas.cws
2016-03-04 19:09:51,642 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-84] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:12:00,638 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-85] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:12:05,636 DEBUG o.a.r.p.PrinterManager [qtp4990863-164] Attempting to assign job:kaas.cws (No Printer) to printer:LC 10" 2(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-04 19:12:05,639 INFO o.a.r.p.PrinterManager [qtp4990863-164] Assigned job:kaas.cws assigned to printer:LC 10" 2(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0) to printer:LC 10" 2(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0)
2016-03-04 19:12:05,679 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-164] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:12:05,683 INFO o.a.r.j.PrintJobProcessingThread [PrintJobProcessorThread-3] Starting:kaas.cws assigned to printer:LC 10" 2(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0) on Printer:LC 10" 2(printerFirmwareSerialPort:/dev/ttyACM0, projectorSerialPort:null Display::0.0) executing on Thread:PrintJobProcessorThread-3
2016-03-04 19:12:05,945 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-59] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:12:07,817 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] BaseName: kaas
2016-03-04 19:12:12,044 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Parsing file:/tmp/printdir/kaas.cwsextract/kaas.gcode
2016-03-04 19:12:12,052 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(****Build and Slicing Parameters****)
2016-03-04 19:12:12,054 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Pix per mm X            = 7.54310 px/mm )
2016-03-04 19:12:12,055 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Pix per mm Y            = 8.38123 px/mm )
2016-03-04 19:12:12,056 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(X Resolution            = 1680 )
2016-03-04 19:12:12,058 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Y Resolution            = 1050 )
2016-03-04 19:12:12,059 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Layer Thickness         = 0.10000 mm )
2016-03-04 19:12:12,060 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Layer Time              = 26300 ms )
2016-03-04 19:12:12,061 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Render Outlines         = False
2016-03-04 19:12:12,062 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Outline Width Inset     = 2
2016-03-04 19:12:12,063 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Outline Width Outset    = 0
2016-03-04 19:12:12,065 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Bottom Layers Time      = 150000 ms )
2016-03-04 19:12:12,066 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Number of Bottom Layers = 4 )
2016-03-04 19:12:12,067 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Blanking Layer Time     = 8000 ms )
2016-03-04 19:12:12,068 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Build Direction         = Bottom_Up)
2016-03-04 19:12:12,070 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Found: LiftDistance of:4.000
2016-03-04 19:12:12,071 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Slide/Tilt Value        = 0)
2016-03-04 19:12:12,072 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Anti Aliasing           = True)
2016-03-04 19:12:12,073 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Use Mainlift GCode Tab  = False)
2016-03-04 19:12:12,074 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Anti Aliasing Value     = 1.5 )
2016-03-04 19:12:12,076 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Found: LiftSpeed of:60.000
2016-03-04 19:12:12,077 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Z Bottom Lift Feed Rate = 50.00000 mm/s )
2016-03-04 19:12:12,078 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Z Lift Retract Rate     = 100.00000 mm/s )
2016-03-04 19:12:12,078 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Flip X                  = False)
2016-03-04 19:12:12,079 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Flip Y                  = False)
2016-03-04 19:12:12,080 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Found:800 slices
2016-03-04 19:12:12,081 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(****Machine Configuration ******)
2016-03-04 19:12:12,082 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Platform X Size         = 222.72mm )
2016-03-04 19:12:12,083 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Platform Y Size         = 125.28mm )
2016-03-04 19:12:12,084 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Platform Z Size         = 180mm )
2016-03-04 19:12:12,085 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Max X Feedrate          = 100mm/s )
2016-03-04 19:12:12,086 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Max Y Feedrate          = 100mm/s )
2016-03-04 19:12:12,087 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Max Z Feedrate          = 100mm/s )
2016-03-04 19:12:12,088 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;(Machine Type            = UV_DLP)
2016-03-04 19:12:12,089 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;********** Header Start ********
2016-03-04 19:12:12,090 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;Here you can set any G or M-Code which should be executed BEFORE the build process
2016-03-04 19:12:12,091 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Send GCode:G21
2016-03-04 19:12:12,093 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-3] Write: G21

2016-03-04 19:12:12,094 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-3] lineRead: rror: movement was interrupted because of door opening

2016-03-04 19:12:12,101 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-3] lineRead: error: parse failure: G21 is not supported at line 'G21'

2016-03-04 19:13:32,028 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [qtp4990863-164] Accepting new printable kaas.cws as a Creation Workshop Scene
2016-03-04 19:13:33,921 INFO o.a.r.s.ProgressiveDownloadServlet [qtp4990863-17] HTTPGet:/video/camera.mp4
2016-03-04 19:13:33,924 WARN o.a.r.s.ProgressiveDownloadServlet [qtp4990863-17] File doesn't exist:/tmp/temp.mp4 for resource:http://192.168.1.88:9091/video/camera.mp4
2016-03-04 19:14:12,159 INFO o.a.r.g.GCodeControl [PrintJobProcessorThread-3] lineRead: null
[... cut ...]
2016-03-04 19:19:24,465 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Show Blank
2016-03-04 19:19:24,467 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Ignored line:;********** Lift Sequence ********
2016-03-04 19:19:24,468 INFO o.a.r.j.CreationWorkshopSceneFileProcessor [PrintJobProcessorThread-3] Send GCode:G1 Z4.0 F50.0
``

It gets an error after sending G21 (To set the measurements to millimeter), this causes an 2 minutes  time out and could be the waiting time I get correct? will try without tonight...

kloknibor, 2016-03-07:
I found out G21 isn't present at all in the firmware (the USA costumers will be happy :P #sarcasm) it only uses millimeters. So I guess deleting this will fix my problem if it's true it introduces a time out of 2 minutes like I tought, is that correct about the time out after an error?

Thanks!

WesGilster, 2016-03-07:
Yep, that's exactly what's going on. I generally expect an ok after every gcode response(even errors). This is the first time I've seen an error message and no ok. I can probably change that to wait for a "\n" instead, but I'll need to write a few tests around this.

So two issues:
1. You've got the wrong gcode for your printer.
2. I need to parser errors faster than 2 minutes.

kloknibor, 2016-03-07:
well the G21 should normally set the the distances to mm and they were using it in CW. But I found out this Gcode isn't defined in the firmware and will give the error as shown... I just need to remove the gcode and that's it than :)! But this printer returns a lot of errors... It also has an safety switch, when the cover is removed it will give errors and won't move while printing... But did an request to pause that already somewhere :P

WesGilster, 2016-03-12:
Hows the testing on this printer going? 

I'm in the middle of making the auto-detection gcode parsing more robust for GRBL, but was just wondering about a new "line ending" theory. Up until now, I've always used "ok" but I'd like to move to "\r?\n" to fix the timeout issues you are experiencing on parsing errors. By the way, I've added your photocentric gcode control to the dev branch build.

kloknibor, 2016-03-12:
Printing went fine ;)! I didn't have time to test my own code yet to be honest... Only tested the full gcode change and I'm quite busy with all kinds of stuff... Sorry!

Oh by the way... I gave a quick glance at changing the eGeneric driver. But I couldn't find the value anymore in root/machines nor root/3dprinters did I just look past ot or did it change place?

kloknibor, 2016-03-12:
What is kind of annoying that is that it returns an error when the hood is off... And cwh just skips the movement... I think it would be better to put the printer in pause and let the user start printing from the GUI again. Maybe a pop up but I guess that's hard to do in een multi user environment, right? So maybe only just pause it ;)

WesGilster, 2016-03-12:
> did I just look past ot or did it change place?

Yeah you looked past it. Answered here #158.

Yeah you are actually asking for a "pause and replay last gcode" action. Could you paste the "Exact" error that you would like the pause for in this issue: #92

WesGilster, 2016-03-15:
I'm using this as a guide for parsing the gcode return values:
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#streaming-a-g-code-program-to-grbl

It should make the gcode response parsing quite a bit more robust.

WesGilster, 2016-03-22:
This is in the dev testing branch as well. You should see that the printer pauses on error. You should also see that there is an error message displayed on the print job as well. I don't have a great way of checking to see if this is working. Eventually it would be cool to start building a full suite of print simulation. I just haven't gotten around to that yet.

closed 2016-04-29 by kloknibor