area515/Photonic3D #172: Simulated RPi Image, Firmware, & Projector Testing

jmkao, 2016-03-09:
Test basic functionality of a fresh install in SemaphoreCI as part of the RPi image building process via:

 * Launch generated RPi image in qemu
 * Set up virtual serial ports (http://nairobi-embedded.org/qemu_character_devices.html)
 * Run Marlin and GRBL on simavr (https://github.com/buserror/simavr/tree/master/examples/board_simduino)
 * Create some kind of simulated projector
 * Call REST API calls to validate that CWH launched and that the correct serial ports are detected

WesGilster, 2016-03-18:
I'd like to add Repetier firmware sim (if possible). This will really harden up the server side.

jmkao, 2016-03-19:
Repetier should be the same process for SimAVR. I'll start looking at first integrating this into the Travis based check-in tests.

WesGilster, 2016-03-19:
Great. No hurry. I've got all of the tests running now including the vision tests.

jmkao, 2016-03-21:
Hmm... while the SimAVR library can simulate the atmega2560, the simduino program that simulates the whole arduino board only works up to a atmega328p, which doesn't have enough memory to load marlin.  There seems to be some kind of known issue with no clear solution. I'll have to prod the project and wait for them to make some progress.

WesGilster, 2016-03-22:
Darn, that will be helpful.

jmkao, 2017-05-20:
Thinking about this again. GRBL will run on an atmega328p and has some interesting real-time machine info features that might feed some useful features like live Z height info

WesGilster, 2017-05-20:
So what I'm hearing is that you'd like a way to latch return data from gcodes into variables that can be affected by both the templating engine and the scripting engine? 

Sounds like an awesome feature. At the moment the return codes are pretty restrictive. They only use return codes(parsed with regex) for the purpose of pausing the printer. Instead I could fold that functionality into a more generic mechanism.

For example, maybe you register regexs with a corresponding JS function. When a response comes back from a gcode that matches that regex, the JS function executes?

So in the example of "Error: Hood was opened" -> JavaScript: { printJob.pause() }
So in the example of "Info: Model Height= \\d{0, 3}" -> JavaScript: { var modelHeight = gCodeReturn[1] }

What are you thinking?

jmkao, 2017-05-20:
Right now, something super simple like a realtime display of the current Z height of the build platform.

WesGilster, 2017-05-21:
Ok, simple it is. I thought you wanted something that integrated with the firmware. If you want simple, then the gui already has all of the information it needs to display the current height of the build platform. I checked in the changes and I'll make a dev build.

WesGilster, 2017-05-21:
Sorry, forgot to upload the example:
[image]

WesGilster, 2018-01-21:
This has been complete for quite some time.

closed 2018-01-21 by WesGilster