area515/Photonic3D #175: Support for raspberry pi 3 wifi

kloknibor, 2016-03-11:
Hi Guys,

Just got the pi 3. Super fast booting times and I'm tested on the latest photonic3d image... But I can't get wifi to work, not with the sd card method nor the ethernet wifi way. It just doesn't show any wifi acces points. Can it be we use an older version of raspbian jessie? I'll dig around some more ;)!

Robin

kloknibor, 2016-03-11:
[LogBundle.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/169989/LogBundle.zip.pdf)

kloknibor, 2016-03-11:
I doubt the log bundle will be useable for getting the wifi running but well I hope I'm wrong :)!

jmkao, 2016-03-11:
The image generation always downloads the latest.

Send me the output from `ifconfig -a`, `lsmod`,`lsusb`, `lspci`, and `dmesg`.

kloknibor, 2016-03-11:
pi@photonic3d:~ $ ifconfig -a
eth0      Link encap:Ethernet  HWaddr b8:27:eb:a5:cc:d0
          inet addr:192.168.1.154  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::64fd:1d3f:8322:ba73/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:34515 errors:0 dropped:4645 overruns:0 frame:0
          TX packets:9220 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:6116487 (5.8 MiB)  TX bytes:3452410 (3.2 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:127 errors:0 dropped:0 overruns:0 frame:0
          TX packets:127 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:35228 (34.4 KiB)  TX bytes:35228 (34.4 KiB)

wlan0     Link encap:Ethernet  HWaddr b8:27:eb:f0:99:85
          inet addr:192.168.1.83  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::ba27:ebff:fef0:9985/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1171 errors:0 dropped:46 overruns:0 frame:0
          TX packets:523 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:247013 (241.2 KiB)  TX bytes:158354 (154.6 KiB)

kloknibor, 2016-03-11:
pi@photonic3d:~ $ lsmod
Module                  Size  Used by
brcmfmac              177546  0
brcmutil                5501  1 brcmfmac
cfg80211              389253  1 brcmfmac
rfkill                 16036  2 cfg80211
snd_bcm2835            19802  0
snd_pcm                73442  1 snd_bcm2835
bcm2835_gpiomem         2860  0
snd_timer              18792  1 snd_pcm
bcm2835_rng             1763  0
snd                    50779  3 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         2944  0
uio                     7753  1 uio_pdrv_genirq
ipv6                  338574  58

kloknibor, 2016-03-11:
lsusb
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

kloknibor, 2016-03-11:
pi@photonic3d:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.18-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #846 SMP Thu Feb 25 14:22:53 GMT 2016
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80876f80, node_mem_map b9bb4000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 12 pages/cpu @bafb0000 s20416 r8192 d20544 u49152
[    0.000000] pcpu-alloc: s20416 r8192 d20544 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=1 dma.dmachans=0x7f35 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2709.boardrev=0xa02082 bcm2709.serial=0x31a5ccd0 smsc95xx.macaddr=B8:27:EB:A5:CC:D0 bcm2708_fb.fbdepth=24 bcm2708_fb.fbswap=1 bcm2709.uart_clock=48000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939416K/966656K available (6024K kernel code, 534K rwdata, 1660K rodata, 448K init, 757K bss, 19048K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807895a0   (7686 kB)
      .init : 0x8078a000 - 0x807fa000   ( 448 kB)
      .data : 0x807fa000 - 0x8087fac0   ( 535 kB)
       .bss : 0x80882000 - 0x8093f79c   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000025] Switching to timer-based delay loop, resolution 52ns
[    0.000240] Console: colour dummy device 80x30
[    0.001288] console [tty1] enabled
[    0.001341] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001409] pid_max: default: 32768 minimum: 301
[    0.001725] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001768] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002761] Initializing cgroup subsys blkio
[    0.002810] Initializing cgroup subsys memory
[    0.002858] Initializing cgroup subsys devices
[    0.002900] Initializing cgroup subsys freezer
[    0.002949] Initializing cgroup subsys net_cls
[    0.003020] CPU: Testing write buffer coherency: ok
[    0.003106] ftrace: allocating 20309 entries in 60 pages
[    0.048500] CPU0: update cpu_capacity 1024
[    0.048563] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.048596] [bcm2709_smp_prepare_cpus] enter
[    0.048748] Setting up static identity map for 0x8240 - 0x8274
[    0.050534] [bcm2709_boot_secondary] cpu:1 started (0) 18
[    0.050871] [bcm2709_secondary_init] enter cpu:1
[    0.050915] CPU1: update cpu_capacity 1024
[    0.050921] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.051294] [bcm2709_boot_secondary] cpu:2 started (0) 18
[    0.051549] [bcm2709_secondary_init] enter cpu:2
[    0.051571] CPU2: update cpu_capacity 1024
[    0.051577] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[... 234 lines cut ...]
[    7.159078] brcmfmac: brcmf_sdio_drivestrengthinit: No SDIO Drive strength init done for chip 43430 rev 1 pmurev 24
[    7.160116] usbcore: registered new interface driver brcmfmac
[    7.297403] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Dec 15 2015 18:10:45 version 7.45.41.23 (r606571) FWID 01-cc4eda9c
[    7.317032] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    7.616505] brcmfmac: brcmf_cfg80211_reg_notifier: not a ISO3166 code
[    7.616534] cfg80211: World regulatory domain updated:
[    7.616544] cfg80211:  DFS Master region: unset
[    7.616552] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[... cut ...]
[    8.176780] brcmfmac: brcmf_add_if: ignore IF event
[    8.595904] Adding 102396k swap on /var/swap.  Priority:-1 extents:4 across:249852k SSFS
[    9.148646] random: nonblocking pool is initialized
[   10.349794] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   10.350960] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.864025] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.866358] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1

kloknibor, 2016-03-11:
lspci was unkown to the pi... By the way I should add I'm trying to get it working with the INTERNAL wifi. So lsusb wouldn't be needed but you can already see the wifi is there with ifconfig ;)! If you need anythingmore let me know! I'll be awake for another hour or so

kloknibor, 2016-03-11:
And the current set-up is ethernet and powerplug, latest photonic3D image ;)!

jmkao, 2016-03-11:
Your wlan0 seems to be up and connected to a wireless network on 192.168.1.83. What problem are you having connecting to it?

```
wlan0 Link encap:Ethernet HWaddr b8:27:eb:f0:99:85
inet addr:192.168.1.83 Bcast:192.168.1.255 Mask:255.255.255.0
inet6 addr: fe80::ba27:ebff:fef0:9985/64 Scope:Link
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
```

kloknibor, 2016-03-11:
You are right.... Really I waited a while and couldn't find with IP scan tools... But it shows up now... Sorry to bother...! Can it be the first boot it takes a while till wifi connects?

jmkao, 2016-03-11:
The first boot takes extra long because it will resize the filesystem and then reboot, but that isn't specific to the RPi 3, it will always do this.

kloknibor, 2016-03-11:
I got it off the ethernet cable and it is offline again... The green led is blinking at an constant blinking speed and it won't come online... (before blinking constant it blinked at "random")

kloknibor, 2016-03-11:
I know first boot takes longer because of a reboot (because the resizing etc) but I waited like 10 minutes... Normal booting time was 30 seconds on rpi 2... 

Now the green led is totally of so no activity and it isn't online (using advance ip scanner)

kloknibor, 2016-03-11:
and sometimes still blinking at constant again don't know if that says anything... How can I debug this the best... Finding an display and send you the commands like we did before?

jmkao, 2016-03-11:
Well, now we know that at least the OS has all of the drivers necessary to get it on the network. The question is one of stability.

/var/log/messages, /var/log/syslog, and /var/log/kern.log should contain information about state changes in the OS. Send those over.

kloknibor, 2016-03-11:
[jmkao.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/170063/jmkao.zip.pdf)

just remove the .pdf ;)!

jmkao, 2016-03-11:
In the future you do not need to rename the extension to PDF if you upload it to github. If you leave it as zip, I can open and extract it on my iPhone. As PDF, it's much harder.

kloknibor, 2016-03-11:
@jmkao for some reason github refuses to let me do that... It returns an error that this file format isn't supported while it is present in the supported file types... very strange... You can use ifile if your on iphone ;) New jailbreak for ios 9.1 came out today. It has an weird boot loop bug but eventually it works haha... Was kind of worried... But in the end it worked :)

kloknibor, 2016-03-11:
Hmm in old internet explorer this seems to work... Chrome refuses (while github says this IE version is unsupported)

jmkao, 2016-03-12:
Hmm... strange... I cannot find any indication that the OS had any problems with the wifi.

I am seeing that the system was restarted at 21:17:04 and at 22:17:04. This could just be a coincidence, but it seem strange to me that the starts occurred exactly one hour apart on the same second.

On a side note, I am seeing the log line:
```
Mar 11 21:17:04 photonic3d systemd[1]: Starting Daily Cleanup of Temporary Directories.
Mar 11 21:17:04 photonic3d systemd[1]: Started Daily Cleanup of Temporary Directories.
```

which I guess could explain some of the upload deletion mysteries. I looked for the tmp cleanup scripts in /etc/cron.daily, where I've seen them on other systems, but apparently this is under systemd control in raspbian.

If you left the RPi3 on overnight, when you wake up, it would be good to send me the /var/log/syslog file to see if there is some kind of reboot happening. If you turned it off, maybe let it run for a 6 to 10 hours and then send me the syslog.

kloknibor, 2016-03-12:
Weird about the rebooting indeed... I did unplug it twice but my timing couldn't that perfect could it? And I turned it off and I can't play too much with it this weekend... However if it may help we could plan an moment on monday or later in which I can set port forwarding for the pi so you can ssh into it and do whatever you need to do ;)! Or maybe even better use teamvieuwer because of security reasons.

jmkao, 2016-03-12:
Easiest thing for now is just leave it on, and send me the log on Monday.

kloknibor, 2016-03-12:
With or without ethernet? Side by side with wifi

Verstuurd vanaf mijn iPhone

jmkao, 2016-03-12:
I think no Ethernet.

kloknibor, 2016-03-14:
I totally forgot to plug in the pi this weekend... I ran it for like 4-5 hours now and I hope that still helps ;)! Here are the logs just after connecting ethernet : 

[JMKAO2.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/172940/JMKAO2.zip.pdf)

(Sorry for the pdf again... I tried different browsers... On work I had luck with firefox... At home not... WIll look into this... sorry!)

jmkao, 2016-03-15:
Did you experience any problem with the wifi over this time?

kloknibor, 2016-03-15:
yes, no wifi at all the times I scanned for it with my ip scanner...

jmkao, 2016-03-15:
Setting the IP scanner aside, are you able to reach the the print UI or ssh directly at 192.168.1.83?

kloknibor, 2016-03-15:
No I can't... I got it on for the whole day now... Shall the logs help?

jmkao, 2016-03-16:
Yeah, send the logs.

kloknibor, 2016-03-16:
[Dropbox (1).zip](https://github.com/area515/Creation-Workshop-Host/files/175478/Dropbox.1.zip)
yayy firefox from work still works to upload zips  :) So here are the logs along with some more I tought might be helpfull ;)! I also did ifconfig -a and found that wifi and ethernet is online (83 and 154) but when ethernet is out it isn't reachable on neither...

jmkao, 2016-03-17:
The only thing I can see from the logs is a large number of group rekeying attempts. This type of behavior only matches the problem described at:

http://ubuntuforums.org/showthread.php?t=2261225

But doesn't explain why other wifi devices would not be affected. Otherwise, I can't find any reason why the problem you are encountering would be specific to anything that is in the image.

The only further thing I can think of trying would be to install a regular Raspbian Jessie and configure wifi the standard way, and then see if the problem exists. If the problem goes away, then the problem may be related to some package that I removed, and if the problem persists, then there is something wrong with Raspbian itself or your specific RPi.

kloknibor, 2016-03-17:
Thanks for looking into this James, All other wifi devices are totally fine. So it shouldn't be that... And also I tested with 2 raspberry pi 3's on 2 different networks. So I'm afraid it might be some removed package... But still it's weird since the driver works... I'll test tonight with normal raspbian and let you know ;)!

jmkao, 2016-03-17:
The various threads about the group rekeying thing more commonly occur with Broadcom (BCM####) devices than other types, so it is possible that the RPi 3 may be the only device of this type on your network.

kloknibor, 2016-03-17:
The wifi works like a charm on raspbian jessie full version, good signal too ;)! Shall I try the light version too?

jmkao, 2016-03-17:
Try lite. And then try:

apt-get remove -y --purge aptitude aptitude-common g++ g++-4.9 gcc gcc-4.9 gdb g
dbserver luajit perl perl-modules python-rpi.gpio bluez bluez-firmware geoip-dat
abase libc-dev-bin libfreetype6-dev samba-common libdrm-radeon1 libdrm-amdgpu1 l
ibdrm-freedreno1 libdrm-nouveau2 xserver-xorg-input-synaptics xserver-xorg-input
-wacom

kloknibor, 2016-03-17:
on it ;)! Should I run it at once?

kloknibor, 2016-03-17:
Hmm it seems a raspbian jessie light issue... It worked fine on complete version but with the light it doesn't... I found this somewhere : http://blog.hypriot.com/post/run-docker-rpi3-with-wifi/ They do something with an overlay... Is that needed for us too?

kloknibor, 2016-03-17:
I found an not so good for us fix... :

Pretty sure this is a "simple" issue of device tree not beeing enabled.

In a shell do
$sudo raspi-config

Then select item 9, Advanced Options
Select item A5, Device Tree
Choose Yes
Reboot

Good luck!

PS If any of you (like me) have issues with the serial port /dev/ttyAMA0 not working, then _disable_ device tree. Seems we can't have both WiFi and serial working right now....

this worked out for me... 

From: James Kao 
Sent: Thursday, March 17, 2016 8:15 PM
To: area515/Creation-Workshop-Host 
Cc: Robin 
Subject: Re: [Creation-Workshop-Host] Support for raspberry pi 3 wifi (#175)

Try lite. And then try:

apt-get remove -y --purge aptitude aptitude-common g++ g++-4.9 gcc gcc-4.9 gdb g
dbserver luajit perl perl-modules python-rpi.gpio bluez bluez-firmware geoip-dat
abase libc-dev-bin libfreetype6-dev samba-common libdrm-radeon1 libdrm-amdgpu1 l
ibdrm-freedreno1 libdrm-nouveau2 xserver-xorg-input-synaptics xserver-xorg-input
-wacom

kloknibor, 2016-03-17:
never mind... It died pretty soon after boot...

From: James Kao 
Sent: Thursday, March 17, 2016 8:15 PM
To: area515/Creation-Workshop-Host 
Cc: Robin 
Subject: Re: [Creation-Workshop-Host] Support for raspberry pi 3 wifi (#175)

Try lite. And then try:

apt-get remove -y --purge aptitude aptitude-common g++ g++-4.9 gcc gcc-4.9 gdb g
dbserver luajit perl perl-modules python-rpi.gpio bluez bluez-firmware geoip-dat
abase libc-dev-bin libfreetype6-dev samba-common libdrm-radeon1 libdrm-amdgpu1 l
ibdrm-freedreno1 libdrm-nouveau2 xserver-xorg-input-synaptics xserver-xorg-input
-wacom

kloknibor, 2016-03-17:
I tested it... Wifi works great with device tree enabled... It works very fast and smooth... But.... Indeed serial ports are broken... This seems more off an raspbian jessie light issue... Do any of you want to take a look into it? Or are we just not gonna support the raspberry pi 3 untill this jessie light is being fixed? Anyways I uploaded my .log files here : 
[LogBundle.zip (2).pdf](https://github.com/area515/Creation-Workshop-Host/files/178736/LogBundle.zip.2.pdf)

jmkao, 2016-03-17:
This is one of those things that would probably be fixable if I had a RPi3 and spent enough time on it, but might not be worth it compared to other work since there are probably other people working on this in the community.

kloknibor, 2016-03-17:
I can't seem to find much about other people complaining about it...

WesGilster, 2016-03-17:
Yeah. I agree, supporting hardware through a forum is not ideal and not interesting.

jmkao, 2016-03-18:
@kloknibor Since you have the relevant hardware, it would probably be easiest for you to perhaps examine the difference between Jessie Lite and full Jessie and ask for help on a purely OS level basis in the Raspbian forums. That might elicit some help from someone more familiar with Jessie and the RPi 3 as a standalone separate issue from Photonic3D. Then if you can report back what you find based on plain Jessie, I can integrate your findings into the image.

kloknibor, 2016-03-18:
On it ;)!

kloknibor, 2016-03-19:
Hmm weird... I tried once to set the serial ports root acces to true in raspi config and everything started working again... However now when I reflash the photonic3d image and only enable device tree in raspi-config it starts working without an problem. Only the wifi shows up as serial port (Or the ethernet) But the search for 3dprint firmware helps. I replicated it 3 times now with only setting device tree to true... It works ;)! Only the first time it failed for unkown reasons.... So if we can enable device tree by default in the next release that would be great :)! Thanks!

jmkao, 2016-03-20:
Device Tree is a fairly advanced option that significantly changes the way that modules are loaded, and is very specific to the Raspberry Pi (as opposed to linux in general). This will make the system diverge significantly from most peoples'standard Raspbian installs and exposes us to future  issues that arise from being in a non-mainstream configuration.

In your testing between Jessie and Jessie Lite, did you find that full Jessie had DT enabled? Or did it work with DT disabled? Have you seen a lot of people with RPi3's having to enable DT?

kloknibor, 2016-03-21:
Sorry for the late response! Was a busy weekend! I understood that raspbian enabled Device tree by default (Source : https://www.raspberrypi.org/documentation/configuration/device-tree.md). In my normal raspbian version DT was enabled if I remember it right ;)!

jmkao, 2016-03-21:
I guess what I want to confirm is whether DT is enabled in Raspbian Jessie, but disabled in Jessie Lite.

If so, if you can do some asking as to why the images are built this way, that would give me a better idea as to what kind of longer-term lifecycle effects altering the default Jessie Lite config would have.

WGAndrew, 2016-09-15:
From what I can tell, DT is already enabled in Raspbian Jessie Lite/the photonic image.

Running ifconfig shows that the wlan0 interface is there, and Wpa_cli and iwlist show local SSIDs.

I'm not sure if it's a bug in Photonic3d that's preventing that Wifi setup being populated now. If you'd like logs or CLI output, more than happy to help, but certainly in Photonic3d 1.0.0-beta2 (aka build 298 iirc?) WiFi's not showing on rpi3.

WesGilster, 2016-09-15:
I'm kindof flying blind with the RPi3 as I don't own one. WIFI management is a simple class org.area515.resinprinter.network.LinuxNetworkManager which is nothing more than a commandline parser that enumerates your networks with the following command:
ifconfig | grep Link | awk '{ print $1 }'

Then checks each interface to determine if it's a wifi interface with:
wpa_cli -i [interface] ping
...to receive a pong

Next it tries to bring the interface up with:
ifup [interface]
but I don't believe that is the problem since it builds wireless info before that.

One of those commands isn't finding your networks on the RPi3. You can probably try this out yourself to determine what the issue is. Or send the logs and I'll take a look, but this area has very little logging in it. :( If you'd like to send me the commands that work for you, I can turn out a dev release that has your fix.

WGAndrew, 2016-09-15:
So from the top:
**ifconfig | grep Link |awk '{print $1}'**
```
eth0
inet6
lo
wlan0
inet6
```
...which would lead to:
**wpa_cli -i wlan0 ping**
`PONG`
now ifup on the pi seems to require SU, so here's both:
**ifup wlan0**
```ifup: failed to open lockfile /run/network/.ifstate.lock: Permission denied'''
**sudo ifup wlan0**
```ifup: interface wlan0 already configured```

It's been about 13 years since I did any serious Java, so I'm not really great at going through the code and picking apart what's being done and where it might be hitting a wall :(

What I can do though is help with more info:
**wpa_cli -i wlan0 scan** and **wpa_cli -i wlan0 scan_results** both work and give local SSIDs - a few censored samples:
'''
bssid / frequency / signal level / flags / ssid
04:18:d6:19:79:21       2462    -67     [WPA-PSK-CCMP][WPA2-PSK-CCMP][ESS]      Wildcard | Super-fast Broadband
92:63:fe:43:9a:1d       2437    -78     [WPA-EAP-CCMP+TKIP][WPA2-EAP-CCMP+TKIP][ESS]    BTWifi-X
90:4d:4a:26:50:ee       2412    -80     [WPA2-PSK-CCMP][WPS][ESS]       BTHub6-2Q92
90:4d:4a:12:b5:ac       2412    -81     [WPA2-PSK-CCMP][WPS][ESS]       BTHub6-Z63C
d4:63:fe:43:9a:1b       2437    -81     [WPA2-PSK-CCMP][WPS][ESS]       BTHub5-NP7H
92:4d:4a:26:52:ec       2412    -82     [WPA2-EAP-CCMP-preauth][ESS]    BTWifi-X
3c:83:75:bd:09:5a       2462    -86     [WPA2-PSK-CCMP][WPS][ESS][P2P]  Windows Phone2909
50:17:ff:34:39:02       2437    -88     [WPA-EAP-TKIP][WPA2-EAP-CCMP][ESS]      BTWifi-X
92:4d:4a:26:52:ef       2412    -79     [ESS]   BTWifi-with-FON
92:63:fe:43:9a:1c       2437    -79     [ESS]   BTWifi-with-FON
92:4d:4a:27:37:37       2437    -80     [ESS]   BTWifi-with-FON
50:17:ff:34:39:01       2437    -87     [ESS]   BTOpenzone
02:33:bb:00:1b:6a       2422    -89     [ESS]   BTWiFi
ac:d1:b8:3e:96:10       2437    -91     [ESS][P2P]      HP-Print-10-LaserJet 1102
'''
I'm guessing that'lll be the next set of info to pull, regex out the SSID info, and then present in the photonic settings page?

Also: Where are you in the world Wes? Just thinking I might be able to donate a pi3 if that'd help?

jmkao, 2016-09-15:
Actually, I should have updated this, but I have an RPi 3 and have been doing all my recent image development on it. I have not encountered any issues with it.

Using the Photonic3D image that is downloadable from photonic3d.com, Wifi works out of the box without any modification, outside of modifying the octopi-network.txt file as described in the documentation.

As the problem described earlier in the thread from @kloknibor is intermittent, my best guess is that the root cause in that case may be the rekeying issues described in an earlier posting that is a specific interaction between the RPi's wifi and certain wifi access points.

For this new issue, can you describe the exact symptoms and the sequence you went through when installing the RPi system image?

WGAndrew, 2016-09-15:
James: what're the mods to octopi-network.txt? I haven't encountered those docs yet, but it might be what I'm after potentially if there's something in there preventing browsing Wifi SSIDs.
[I had a quick google to find the doc you mentioned, but the only thing I could find was an empty wiki page here: https://wiki.photonic3d.com/doku.php?id=set_up_wifi ]

Sequence was pretty much grabbing the autogenerated vendor-specific Pi image here:
https://photonic3d.com/downloads/Photocentric_photonic3d.zip
extract, setup onto SD, boot in the pi, update the pi with apt-get and similar, etc.

jmkao, 2016-09-15:
Raspberry Pi image installation instructions: https://wiki.photonic3d.com/doku.php?id=raspberry_pi

WGAndrew, 2016-09-15:
Thanks James - I'd actually went through and looked into the config file directly.
Obviously it's great for a static config to a single wifi network.

Sadly, I'm looking more for it to work like a wifi detection + setup util, like the photonic settings page sorta suggests is possible...

WesGilster, 2016-09-15:
I live in the US(Iowa), but we might be able to fix this without a pi3. It's possible we aren't parsing your output properly with this regex:
\\s*([A-Fa-f0-9:]+)\\s+(\\d+)\\s+(\\d+)\\s+([\\[\\]\\+\\-\\w]+)\\t(.+)

Immediately it looks like signal level is now negative? I'm not sure what that means, but it's an easy fix, and I can add a test for this, if this is the precise output of your command.

WesGilster, 2016-09-15:
There was already an existing test for this parsing routine so I modified it with a negative signal value and all networks failed to come up which pretty much reproduces your exact situation. Fixed/checked in and a build started for: WesGilster-bd82963.

[3 more replies]

closed 2016-09-16 by WesGilster