area515/Photonic3D #177: Branding Changes: CWH --> Photonic3D

jmkao, 2016-03-15:
Need to update:
 - Name of repo
 - Text in README.md
 - Text in HTML
 - Name of zips

anything else?

kloknibor, 2016-03-15:
Maybe also the instal folder (easy to adjust) but maybe even the filenames with cwh in it? Like cwh.log but I suppose there are more ;)!

WesGilster, 2016-03-15:
Exciting... What about all of the code?

WesGilster, 2016-03-16:
and the entire directory structure where CWH is stored.

jmkao, 2016-03-16:
At this stage, I think we probably do not need to go below UI and public site-facing strings, so code or log file name changes are probably not necessary. Although Eclipse can probably do many of those renaming changes fairly easily.

WesGilster, 2016-03-16:
Makes sense to me, it all depends on how far you want to cut...

jmkao, 2016-03-18:
Github automatically creates redirects from the old repo name, so here's one potential plan:

1. Update HTML and README in WesGilster/CWH
1. Push to area515/CWH
1. Rename area515/CWH to area515/Photonic3D
1. We each update our forks and then start updating the build scripts (which will still work with the old names too, because of the automatic redirects, so step 3 will not break anything)
1. We do a second pass update with the build files and URLs updated

WesGilster, 2016-03-18:
Yep, I was thinking something similar. No migration needed, and you keep backwards compatibility.

jmkao, 2016-04-08:
To reduce the amount of work, renaming the zips may not even be necessary. Split this into 3 phases: #203, #204, #205

closed 2016-04-08 by jmkao