area515/Photonic3D #178: Pluggable 2d platforms

WesGilster, 2016-03-17:
As I start to add support for more file types I think it would be better to allow pluggable 2d platforms to print on rather than a simple cirtangle. That would add quite a bit of variety when printing 2d files.

kloknibor, 2016-03-17:
Sorry to bother but I'm unsure what you're trying to say about that 2d platforms... Would you mind to explain?

WesGilster, 2016-03-17:
2d files(like text files) have to be printed on some sort of a platform. Currently I force the user to print on a rectangle with curved edges(cirtangle). That seems pretty restrictive. By adding 1 additional shape you can make a keychain, Christmas ornament, door hanger or whatever... Imagine what you could make if your platform could change dynamically. Artists would have a hay day on a Mandelbrot alone...

WesGilster, 2016-03-17:
Actually I might implement this weekend I kind of want to try that out...

kloknibor, 2016-03-17:
Sounds very cool Wes!

jmkao, 2016-03-18:
Interesting, I have been experimenting with 3D printed letter press plates with a friend of mine who has a letter press (https://www.youtube.com/watch?v=kyFzZK_Ftas). This presents a simplified way to quickly layout some text and print a plate.

WesGilster, 2016-03-18:
That's a really cool idea! I built this for making a stencil and stamp kit for my daughter with different fonts. Did you have to make a 3d file out of all of that before you printed it? That must have been a bit painful...

Maybe you weren't aware, but you can also upload gifs/jpgs/pngs as well. That's probably more up your alley since you don't want to print plain text. I had envisioned that people would want to use it to print photos, so unfortunately I put those files through a edge detection algorithm. I certainly could make that optional though.

jmkao, 2016-03-18:
The image was supplied to me as an Illustrator file, which I converted to SVG, important into Autodesk Fusion 360, and the extruded along with a base. It was somewhat annoying at first because there were a bunch of non-watertight vectors (even in 2D) which had to be fixed to make it form a correct solid shape in 3D. PNG input is something that I'll definitely try once I get back to printing. The two main projects taking up my time right now are the Photonic3D build and test system, and integrating the Autodesk Print manager with the Titan 1 via Photonic3D. :)

WesGilster, 2016-03-18:
My that would be annoying. Let me know if you need anything done to better integrate with Autodesk print manager.

WesGilster, 2016-03-22:
I'm testing this now in development. I've exposed the underlying offline screen buffer so that you can manipulate it with Javascript. The default implementation is just a cirtangle as I described above:
platformGraphics.fillRoundRect(centerX - (extrusionX / 2), centerY - (extrusionY / 2), extrusionX, extrusionY, 50, 50);

[image]

I'm pretty excited to start trying out some fractals in 3d. A large part of this change was to compute the platform in parallel to the exposure + gcode. I'm hoping that makes allot of this computation time invisible during a print.

kloknibor, 2016-04-29:
Maybe it will be nice to fill in a standard platform calcutlator, not all our users want to make the code them selves I suppose ;)! But we can put this in the printer profile right?

WesGilster, 2016-04-30:
This already fills in the platform calculator a simple round rect(cirtangle) when creating a new printer:

Code is in here:
`org.area515.resinprinter.services.PrinterService.createTemplatePrinter(String, String, String, double, double, double)`

Feel free to reopen if there was something else you were looking for.

closed 2016-04-30 by WesGilster