area515/Photonic3D #180: Auto upgrade from 228 to 277 didn't work

kloknibor, 2016-03-19:
At least on my pi3 an upgrade from 228 to 277 didn't work...! Maybe because the wifi comes online too late? But it comes online and is stable. Gonna run it for a while now ;)!

kloknibor, 2016-03-19:
Manual upgrade with :+1: 
sudo wget https://github.com/area515/Creation-Workshop-Host/raw/master/host/bin/start.sh
sudo chmod 777 start.sh
sudo ./start.sh

worked fine (I know I could just use the old start.sh... but this was easier :P copy and paste ;))

WesGilster, 2016-03-19:
This is a little disturbing, you're the second one now to say the upgrade didn't work... I just tried it and it worked fine for me. I should clarify that I used start.sh not newstart.sh.

jmkao, 2016-03-19:
When you say that it didn't work, what were you left with? Did it error out at a particular spot, or seem to go okay but somehow leave an inoperable set of binaries?

kloknibor, 2016-03-19:
Sorry didn't check... Blamed it too soon on the wifi... But the manual worked like a charm... So I think it can't be the start.sh... Can I still get it from logs? Even after reboot and some hours?

closed 2016-04-25 by kloknibor