area515/Photonic3D #189: Disect GUI

WesGilster, 2016-03-27:
In order to keep CWH fast, it's about time we start breaking the bootcards into chunks that will allow fragmented loads. For example, we are now loading all of the MachineConfigs and SlicingProfiles from the server just because we entered the printers screen.  

I don't think I'm going to be messing with the bootcards since they are already organized well enough. I think it would be nicer if we asked the user which cards they want to see...

kloknibor, 2016-03-27:
Maybe like in settings somewhere an option to disable and enable things. Like call it advanced settings. Here you can disable settings  like disable open close shutter etc. Some people wont need this setting. I think it isn't exactly what you meant but it's in the same range ;) nicest would be if we could set this from the printer profiles that we pull from github

WesGilster, 2016-03-27:
Yeah, those settings will come from the printer, I think that jmkao already opened a feature request for that. Same would go for x/y controls and other stuff available on the printer controls bootcards. Basically the way that it would work is by checking if your printer supported a function, and the controls would become available. But your right, it's in the same range. 

The strong difference between these are that I'd like to make these user settings(based on cookie settings) rather than printer settings or HostProperties.

jmkao, 2016-03-27:
Good reminder, my request is buried in the comments for #158, I'll open a separate issue for it and summarize the request.