area515/Photonic3D #190: Parameterize Basic Machine Control Functionality

jmkao, 2016-03-27:
Perhaps as a future enhancement we could parameterize the machine control G-code and basic UI into the machine configuration Originally discussed in #158.

For instance, even on my one machine, depending on which stepper driver I have installed (A4988 or DRV8825) and what the current limit potentiometer is set to, there can be a particularly nasty Z-axis resonance at certain speeds. It would be nice to change the feed rates to avoid the resonating speed w/o having to compile a custom G-code driver...

Thinking about this in terms of screen controls, one way that might work would be to have a section in the machine dedicated to properties about the machine controls, which might even allow config-based addition of custom buttons, so that not every small machine-specific UI change would require one to fork a whole set of resources. Maybe something like an XML section like:

```
<controls>
  <showX>true</showX>
  <showY>true</showY>
  <showZ>true</showZ>
  <defaultGCode>
    <moveX>G1 X%1.3f F900</moveX>
    <moveY>G1 Y%1.3f F900</moveX>
    <moveZ>G1 Z%1.3f F100</moveZ>
    <homeX>G28 X</homeX>
    <homeY>G28 Y</homeY>
    <homeZ>G28 Z</homeZ>
    <homeAll>G28</homeAll>
    <modeAbsolute>G90</modeAbsolute>
    <modeRelative>G91</modeRelative>
    <motorsOn>M17</motorsOn>
    <motorsOff>M18</motorsOff>
  </defaultGCode>
  <customButtons>
    <!-- Some kind of default, unconfigurable, grid layout for easy addition of small stuff -->
    <button>
      <label>Show Position</label>
      <gcode>M114</gcode>
    </button>
  </customButtons>
</controls>
```

The idea is purely based on the control screen, to try to bridge the gap between the most common machine specific requirements and the more exotic ones that would requiring forking all of the resources.

WesGilster, 2016-07-22:
I was in the middle of fiddling around in this code, any requests for features here? The "defaultGCode" section of this XML aren't appropriate. Those functions need to be implemented each as a separate GCodeTemplates exactly like the other templates in SlicingProfile:
gCodeHeader;
gCodeFooter;
gCodePreslice;
gCodeLift;
zLiftSpeedGCode;
zLiftDistanceGCode;
gCodeShutter;

Here are the settings for GUI setup XML for CW:
<DisplayedControls>XYZTEHBPG</DisplayedControls>
X-X  Axis
Y-Y Axis
Z-Z Axis
T-Tilt control
E-Extruder motor
H-Extruder Heater
B-Heated Platform
P-Projector control
G-Manual GCode

I'll put nicer names on all of the  JSON of course, but this should give us a start on all of the GUI functionality.

jmkao, 2016-08-06:
Hmm.. looking back at the original feature request, a lot of what seems to be needed is a modular way to customize machine controls, which tend to vary from machine to machine, without having to make a fork of the whole UI.

Most of this is already separated within printerControls.html and printerControls.js, the only need then would be away to allow just these two files to be separated out into some machine-specific component without substituting the whole resources directory.