area515/Photonic3D #191: Pretty unstable wifi suddenly

kloknibor, 2016-03-28:
Hi,

I installed the latest image. Now  all of the sudden my wifi is very unstable... It comes up very quick but than disappears. Some times it comes back but most of the time it needs rebooting... I tested it for like an hour but the logs are very empty... Geussing it won't start cwh properly even... But since it rebooted important logs are gone now right...??

[LogBundle.zip (3).pdf](https://github.com/area515/Creation-Workshop-Host/files/192435/LogBundle.zip.3.pdf)

jmkao, 2016-03-28:
What hardware is this? I haven't updated the image in a couple of weeks, there should be no change at the OS level.

kloknibor, 2016-03-28:
I tried 2 dongles, the lb link and the tp-link Wn725n. Both worked fine before... And weird thin is that right after boot everything is reachable...! I'll try a freah download and install tomorrow ;)! Sleepwell and happy easter :)

WesGilster, 2016-03-29:
Happy Easter to you as well. As you mentioned, wifi isn't operational on boot:
I saw this:
`wget: unable to resolve host address â€˜github.comâ€™`
in the startup logs. 

CWH logs have nothing interesting in them.

kloknibor, 2016-03-30:
I did change start.sh in the image to my personal repo. Also I changed index.html | printerscontrols.html | printers.js and config.properties. But this shouldn't break my wifi right?

WesGilster, 2016-03-30:
html, js, config.properties will have no effect.

start.sh could have a huge effect, depending on what you did.

kloknibor, 2016-03-30:
I only changed the repo to pull the sources from so from area515 to kloknibor, furthermore I didn't touch a thing...!

WesGilster, 2016-03-30:
That wouldn't cause any issues either. At this point that pretty much rules out Photonic3D as being the culprit though. What image are you using?

WesGilster, 2016-03-30:
Also what hardware...

WesGilster, 2016-03-30:
Actually jmkao already asked that. Sorry... You aren't even resolving DNS, are you sure you haven't put yourself in AP mode?

kloknibor, 2016-03-30:
I downloaded the newest image from photonic3D, Mounted it with OSFMOUNT and addes the new GUI folder, and changed both start.sh and the config.properties. Hardware is pi 2 with wifi dongles : lb link and the tp-link Wn725n

jmkao, 2016-03-30:
I need the OS level logs, same as what you sent for the RPi 3.

kloknibor, 2016-03-30:
Do I also need to let it run for some hours?

jmkao, 2016-03-30:
You should already have historical logs from what you experienced previously.

!DSPAM:56fbf166101191110174325!

kloknibor, 2016-03-30:
Here is the dropbox link so you can open it on your phone ;)! https://www.dropbox.com/s/lgwew0smhbq6tu7/wifi.zip?dl=0 I should mention that when I booted it to get the logs everything worked fine... It's still up after some minutes but the day before yesterday I did a lot of tries... And it didn't work...

kloknibor, 2016-03-30:
it went fine for 3 (hard)reboots it fails to start wifi now again...

kloknibor, 2016-03-30:
It seems it does not work only on some boots... Rebooting helps in most cases... Will it help to wait for network on boot? Or wont this work because we set the wifi with octopinetwork?

kloknibor, 2016-03-31:
So far it seems to work on another pi in another location... same .iso no problems so far... I'll close it for now when I can reproduce it on other pi's I'll reopen it ;)!

closed 2016-03-31 by kloknibor

kloknibor, 2016-03-31:
I can confirm this is an hardware issue... Sory to bother... Borrowed another pi 2 with anothe tp link (should be the same but blue instead of green led) works like a charm... ! thanks for looking into it anyway... Thinking the pi 2 itself is broken ;(!