area515/Photonic3D #193: [Feature request] : photonic3d.local

kloknibor, 2016-03-31:
Hi,

I know I talked about this some time back, but since we use the octopi image which supports that already I was hoping implementing this would be easy.

I would like to see the raspberry pi reachable from photonic3d.local just like octopi is reachable from octopi.local . 

I think this is easier than the UPNP script for most users. You don't need java sdk installed and find that icon everytime. I'm aware we need bonjour for the .local thing but much people got that already.

let me know what you think!

Robin

WesGilster, 2016-04-01:
So I'm just curious what happens when you have three Raspberry Pis on your network?

kloknibor, 2016-04-01:
Uhhhhhh... Chaos :p? At the moment I use only 1 pi at a time but this can become 2 or more so that might be troubling... Is it possible to add something like a prefix? Or like photonic3d.local photonic3d2.local photonic3d3.local the pi can see if there are already others... But well this can make it more complicated for us yes... Maybe make it an advanced option where people can enable it and give their own names to the pi's this would be great ;)! 

This way you can easily get the right pi :)! But I have no idea how hard this is :$

WesGilster, 2016-04-01:
I can add another Advertiser for Bonjour and we can enable/disable that similar to the UPNP advertiser.

kloknibor, 2016-04-01:
Sounds fine, Is the option I mentioned about choosing the name possible? (within settings)

WesGilster, 2016-04-01:
Yep. Just adding this to keep for later reference:
https://github.com/jmdns/jmdns

kloknibor, 2016-04-01:
great! That makes me really excited thanks Wes! I still haven't got enough time to check the mobile bug... And I want to prioritize the website first... Sorry just very busy lately!

jmkao, 2016-04-01:
At the photonic image level. I can just add the avahi daemon package and it will make its host name available without any configuration, that's how octopi  does it, and it maintains consistency between dhcp/dns based naming and bonjour based naming.

The daemon doesn't even need to be configured, it will do this by default.

!DSPAM:56fe701590331169921191!

WesGilster, 2016-04-01:
@jmkao That's not a bad idea. I'm just not a huge fan of OS dependent solutions. It also doesn't really give much in the way of configuring the name of your host through the GUI. Also, maybe you can answer my original question. what happens when you have three pis advertising themselves as photonic3d.local or do they listen to each other and advertise on different hostnames based on some random number?

jmkao, 2016-04-01:
part of the way that name resolution works will likely always be OS dependent.

There are two common ways for names to be dynamically registered onto the network:

 - DHCP w/ local DNS
 - mDNS (e.g. Bonjour, Avahi, zeroconf)

The DHCP way is common in home routers that do DNS proxying. When a computer gets its IP address via DHCP, it also broadcasts its host name to the DHCP server. When the DHCP server and local DNS server are on the same device, the DHCP server will pass that hostname and IP address to the DNS server and create an entry for it. My home router does this. Since this is done based on information at the DHCP layer, it's dependent on the OS and the mapping is fixed, since a computer can only be assigned one hostname. Conflicts would be resolved in a manner specific to the DNS proxy, which could be first-in-wins or last-in-wins. In any case, this behavior would be common to the whole newtork, based on the router, and not on the DHCP implementation on a particular device on the network.

The mDNS way allows individual processes to broadcast names for their IPs that may or may not align with the computer's hostname because it's not tied to an OS level. mDNS peers are supposed to be both listening and writing to the broadcast network fabric and are expected to not create conflicting entries. So, for instance, avahi provides both mDNS registration and mDNS resolution, thus it keeps track of what other names have been registered in the subnet. If a registration it has would conflict with an existing one, it will append a number to the end of the name. In this example, if I have 2 devices named `photonic3d` on the network, when the second one comes online, the avahi on it will see that there is already a `photonic3d` being broadcast and will then broadcast itself as `photonic3d-0`. If a third one comes online, it will notice that `photonic3d-0` is already taken and broadcast itself as `photonic3d-1`. In avahi, this separator character is controlled by a config parameter, but in general it's controlled by each individual mDNS peer.

Thus, if we were to implement mDNS at the process level, we would need to end up using whatever conflict resolution mechanism was the default in the library we chose, or perhaps have to implement it ourselves.

That also brings into question whether having the device get a name other than the hostname of the device would create confusion on the network. The internet seems rife with reports of mDNS/DNS conflicts where mDNS broadcasts a name that is already in use by DNS (since mDNS can see other mDNS names, but generally does not do any conflict resolution with DNS names). There appears to then be client specific behavior when it gets different IP addresses from DNS and mDNS that result in issues like "I see in my router web UI that it mapped name X to IP Y, but when I actually enter that name, I get some totally different IP!".

WesGilster, 2016-04-02:
Ah cool so it works as I had hoped, but using a sequential number to increment the host name, not a random number. If you'd like to absorb that into the OS, I think that would be fine. I'd like to see this work on Mac and Windows, but honestly it's much less of a priority than Linux.

kloknibor, 2016-04-25:
I saw @jmkao made a new image on 21 march does this also include this new feature? Than I'll test it later this week :) or has it only an updated cwh?

jmkao, 2016-04-25:
Only an updated CWH for the last prod build. I'll add avahi-daemon for the next image version whenever Wes promotes the next stable release.

kloknibor, 2016-04-25:
allright thanks !

jmkao, 2016-05-14:
This was implemented in area515/OctoPi@e88bed8, but not yet tested.

jmkao, 2016-06-13:
This is tested and working, I can see it in the bonjour browser. Saved my bacon when I took my printer to a new location to train the interns where the internet was flaky and I ran into #225.

closed 2016-06-13 by jmkao

WesGilster, 2016-06-14:
Cool. Did Bonjour mDNS make the feature list along with UPNP?

jmkao, 2016-06-14:
Just Bonjour mDNS, I didn't do anything with the UPNP, which already works fine given that cwh actually starts.

jmkao, 2016-06-14:
I implemented it at the OS level in the generated image by including the Avahi daemon in the install. No changes to photonic3d itself.

WesGilster, 2016-06-14:
I understand. I was just hoping it made our "Advertised" feature list for bragging purposes.

jmkao, 2016-06-14:
Oh, I think it definitely does, since the code that generates the image is certainly part of our project even if not in the server proper :)