area515/Photonic3D #194: [bug] : STL slicer broken in 0277

kloknibor, 2016-03-31:
Hi,

It seems the slicer is broken in 0.277. I immidiatly get an failed message. I tried updating with startdev.sh but since the zip files are gone in this repo it won't .... makes it hard to test will open a new issue for that.

Here are the logs : 

[LogBundle.zip (2).pdf](https://github.com/area515/Creation-Workshop-Host/files/198433/LogBundle.zip.2.pdf)

I got 2 more logs from other pi with other errors. But the support function didn;t work so my password wasn't masked. I can send it by mail if you want or you can tell me what files I need to edit to delete all passwords ;)! (or I';; look into it later)

The errors there was first something with heapspace java thingy (I tried a very smal smal object) and afterwards other models which seem to slice, showed slice height but no picture ever came and printer didn't do or show a thing... always stayed at layer 0. Printing from .cws worked fine. latest problem on my image and are515 one too.

kloknibor, 2016-03-31:
Oh by the way I can give you the files I used, but it happened to very much files...

WesGilster, 2016-04-01:
I think this is a duplicate of this:
https://github.com/area515/Creation-Workshop-Host/issues/108

Could you upload an stl file that you were having trouble with to be sure?

kloknibor, 2016-04-01:
The heap space probably. I can get you the file if you want, but it returned that only on 1 file. At home I got an failed message directly with all stl files.

At the other location it seemed starting (this was after heapspace but still when installing stock image without using that heapspace thing) but stayed always on slice 0 and didn't show a thing... Can it be the syntax in myslice profile since it looks like i'm the only one experiencing it :$ will try ;)!

WesGilster, 2016-04-01:
Yes, I need a file. 

The problem is that there is an STL compatibility issue. The file has the total number of triangles in a place where it isn't expected, it reads those bytes and believes it needs to allocate space for billions of triangles. Of course there isn't space for space for that amount of triangles and we run out of heap space. I haven't been able to reproduce it, which is why I need the file...

kloknibor, 2016-04-01:
Is it ok if I mail it? or prefer here?

WesGilster, 2016-04-01:
Here would be great for everyone, but either way would be fine.

kloknibor, 2016-04-04:
See your mail ;)! Just noticed it is non-manifold too... This might have caused the issue maybe? Did anyone test the stable version latelly in combination with the stl slicer? It doesn't slice any stl for me still...

WesGilster, 2016-04-04:
After you opened this bug, I tested the STL slicer. Unfortunately it worked fine for the gear that I had added to my original bug, it even properly filled the gear for a few slices as well.

I'm still looking for you to provide a file that I can use to test this bug.

kloknibor, 2016-04-04:
I emailed you this morning... Did that also work?

kloknibor, 2016-04-04:
(PS sorry for the delay I had the file on another pc I couldn't acces during the weekend)

WesGilster, 2016-04-05:
Yes I did, taking a look at that now.

WesGilster, 2016-05-27:
Confirmed Duplicate of #111

closed 2016-05-27 by WesGilster