area515/Photonic3D #196: [bug] : dev repo wes error at start print

kloknibor, 2016-03-31:
Hi,

I'm having an error when I try to start a printer with everything simulated. I got a fresh copy of raspbian jessie light, ran newstart.sh (after wget and chmod) installed photocentric printer. Changed all ports to simulated except projector left that choose serial port. And got an long http 500 popup with in reason a jetty erro, logs here : 

[LogBundle.zip.pdf](https://github.com/area515/Creation-Workshop-Host/files/198561/LogBundle.zip.pdf)

jmkao, 2016-03-31:
It looks like the X server didn't start properly on your box. Do you have the screen output from running newstart.sh that shows what happened when the start script was run?

kloknibor, 2016-04-01:
no I forgot to copy it :( will try to reproduce later

WesGilster, 2016-04-05:
Is this possible to reproduce?

kloknibor, 2016-04-06:
Gonna try it tonigh

kloknibor, 2016-04-06:
Sorry tomorrow I mean

closed 2016-04-29 by kloknibor