area515/Photonic3D #197: [bug] : Windows dev not working

kloknibor, 2016-04-01:
Hi,

I downloaded the latest CWH version from releases at the wesgilster repo. I unzipped it and ran the start.bat When I choose to install one of the 2 template printers I get the following error : 

Unrecognized field "Font" (class org.area515.resinprinter.printer.SlicingProfile), not marked as ignorable (41 known properties: "LiftDistance", "DotsPermmX", "TestExposureStep", "DotsPermmY", "PlatformTemp", "TwoDimensionalSettings", "UseMainLiftGCode", "name", "MinTestExposure", "XOffset", "YOffset", "ProjectorGradientCalculator" [truncated]]) at [Source: HttpInputOverHTTP@3fcfa433; line: 1, column: 3180] (through reference chain: org.area515.resinprinter.printer.Printer["configuration"]->org.area515.resinprinter.printer.PrinterConfiguration["slicingProfile"]->org.area515.resinprinter.printer.SlicingProfile["Font"])

error log here : https://www.dropbox.com/s/g5soy1hbs61jhq1/LogBundle%20%281%29.zip?dl=0

Ps sweet to add the download button :)!

WesGilster, 2016-04-01:
Darn, I was afraid of that one. I thought I tested that. I moved Font and FontSize to a different location in the JSON file(under 2d file properties) and it's not expected in the same place as what you have checked into the Repo. Removing them out of your printer configuration is the best route...

kloknibor, 2016-04-01:
This also happens with the MuVe from your repo :$ So that should be changed too :)

WesGilster, 2016-04-05:
This should be fixed in the latest dev version.

kloknibor, 2016-04-06:
Works now!

closed 2016-04-06 by kloknibor