area515/Photonic3D #201: newstart.sh erasing cwh directory and not installing

WesGilster, 2016-04-04:
I had some questions and testing out a few scenarios in the newstart.sh:

1. Offline manual upgrade (Is this tested?)
2. Downgrade (Is this supported?)
3. Online upgrade with "newstart.sh area515" = Seems to work when CWH is not installed
4. Online upgrade with "newstart.sh WesGilster" = wipes out the cwh install when area515 is currently installed as shown below:

sudo ./newstart.sh WesGilster
mv: cannot stat `cwh-.*.zip': No such file or directory
Installing latest version of cwh-: DEV020
--2016-04-04 04:16:11--  https://github.com/WesGilster/Creation-Workshop-Host/releases/download/DEV020/C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip
Resolving github.com (github.com)... 192.30.252.129
Connecting to github.com (github.com)|192.30.252.129|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/28702982/18a6a2c4-f98a-11e5-9ca8-fc2e3010440a.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160404%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160404T041614Z&X-Amz-Expires=300&X-Amz-Signature=5a9e02e4e2609ac54f6724c29fc9c294aa9f77cfa5350aa211aedb4b61d1963c&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DC.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip&response-content-type=application%2Foctet-stream [following]
--2016-04-04 04:16:14--  https://github-cloud.s3.amazonaws.com/releases/28702982/18a6a2c4-f98a-11e5-9ca8-fc2e3010440a.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160404%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160404T041614Z&X-Amz-Expires=300&X-Amz-Signature=5a9e02e4e2609ac54f6724c29fc9c294aa9f77cfa5350aa211aedb4b61d1963c&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DC.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip&response-content-type=application%2Foctet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.82.203
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.82.203|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 31758566 (30M) [application/octet-stream]
Saving to: `C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip'

100%[=======================================================>] 31,758,566   685K/s   in 47s

2016-04-04 04:17:05 (658 KB/s) - `C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip' saved [31758566/31758566]

unzip:  cannot find or open cwh-*.zip, cwh-*.zip.zip or cwh-*.zip.ZIP.

No zipfiles found.
chmod: cannot access `*.sh': No such file or directory
rm: cannot remove `cwh-*.zip': No such file or directory
Turning off screen saver and power saving
Determinging if one time install has occurred
Starting printer host server

pi@raspberrypi /opt/cwh $ ls -al
total 0
pi@raspberrypi /opt/cwh $ cd ..
pi@raspberrypi /opt $ cd cwh
pi@raspberrypi /opt/cwh $ ls
C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.291.zip  log.err  log.out

jmkao, 2016-04-04:
The newstart.sh in area515 is an older version that has bugs that can't handle the Windows named uploads and wipe out the cwh directory regardless of whether the download succeeds or not.

The one in WesGilster has logic that downloads the zip to /tmp first and only clears the cwh directory when the correct zip can be definitively identified.

newstart.sh doesn't so much support the notion of upgrade/downgrade is it does switching between forks. Although if you switch to area515, you get screwed because it has the older newstart.sh with bugs.

I did test offline upgrade, so it should work, although I have not yet added it to the automated CI test.

!DSPAM:5701ecb2278198279493443!

WesGilster, 2016-04-04:
Ahh, I see what happened. I'll test that as well, and then you think we are ready to replace the start.sh?

WesGilster, 2016-04-04:
By test that as well, I meant the offline upgrade.

jmkao, 2016-04-04:
Yes, I tested offline upgrade with the cwh-###.zip files created by CI or Linux. It probably does not work with the names with the Windows paths in them generated by the manual `fullRelease` target.

I've been using newstart.sh quite a bit to switch between my fork and your fork, and I haven't found any issues. So I think we're ready to make the switch.

kloknibor, 2016-04-16:
I do still have some problems with newstart.sh , Here is the log : 

pi@photonic3d:~ $ sudo wget https://github.com/WesGilster/Creation-Workshop-Host/raw/master/host/bin/newstart.sh
--2016-04-16 18:41:49--  https://github.com/WesGilster/Creation-Workshop-Host/raw/master/host/bin/newstart.sh
Resolving github.com (github.com)... 192.30.252.121
Connecting to github.com (github.com)|192.30.252.121|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/master/host/bin/newstart.sh [following]
--2016-04-16 18:41:49--  https://raw.githubusercontent.com/WesGilster/Creation-Workshop-Host/master/host/bin/newstart.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 23.235.43.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|23.235.43.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5794 (5.7K) [text/plain]
Saving to: ‘newstart.sh.2’

newstart.sh.2       100%[=====================>]   5.66K  --.-KB/s   in 0s

2016-04-16 18:41:50 (23.2 MB/s) - ‘newstart.sh.2’ saved [5794/5794]

pi@photonic3d:~ $ sudo chmod 777 newstart.sh
pi@photonic3d:~ $ sudo ./newstart.sh WesGilster
mv: cannot stat ‘cwh-.*.zip’: No such file or directory
Checking for new version from Github Repo: WesGilster/Creation-Workshop-Host
Local Tag: unknown
Network Tag: DEV025
Installing latest version of cwh-: DEV025
--2016-04-16 18:41:52--  https://github.com/WesGilster/Creation-Workshop-Host/releases/download/DEV025/C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.296.zip
Resolving github.com (github.com)... 192.30.252.121
Connecting to github.com (github.com)|192.30.252.121|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-cloud.s3.amazonaws.com/releases/28702982/6228fc76-00fb-11e6-8469-7dd8596b86ce.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160416%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160416T184153Z&X-Amz-Expires=300&X-Amz-Signature=c54f3084414660710822c95e2c5c2fb778fa8753599e7cf30e98f7dfc0d9202b&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DC.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.296.zip&response-content-type=application%2Foctet-stream [following]
--2016-04-16 18:41:54--  https://github-cloud.s3.amazonaws.com/releases/28702982/6228fc76-00fb-11e6-8469-7dd8596b86ce.zip?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAISTNZFOVBIJMK3TQ%2F20160416%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20160416T184153Z&X-Amz-Expires=300&X-Amz-Signature=c54f3084414660710822c95e2c5c2fb778fa8753599e7cf30e98f7dfc0d9202b&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DC.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.296.zip&response-content-type=application%2Foctet-stream
Resolving github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)... 54.231.112.112
Connecting to github-cloud.s3.amazonaws.com (github-cloud.s3.amazonaws.com)|54.231.112.112|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 31766107 (30M) [application/octet-stream]
Saving to: ‘/tmp/C.Users.wgilster.git.Creation-Workshop-Host.host.cwh-0.296.zip.3’

ost.host.cwh-0.296.  12%[=>                    ]   3.81M   349KB/s   eta 67s

It keeps hanging on downloaden the zip file.. Tried it 5-6 times... Photonic3d is still up... New putty connections accepted and I can escape the script with CTRL + C ... I will do a fresh install an report back

kloknibor, 2016-04-16:
I downloaded a fresh image and swapped hardware... Works like a charm now... So it won't be something wrong with the code I guess ;)!

kloknibor, 2016-04-25:
I tried this numberous times to swap different repo's and it current so I think it is ready to migrate :)! (not the area515 one but wes and mine repo)

closed 2016-05-01 by jmkao