area515/Photonic3D #203: Rebrand to Photonic3D Phase 1: HTML & Logo

jmkao, 2016-04-08:
Update README.md and HTML pages to use the Photonic3D logo and name

jmkao, 2016-04-08:
@ergobot @kloknibor Do either of you have the full resolution Photonic3D logo images? Is the only asset  the `photonic3d-icon-sign.png` at https://photonic3d.com/wp-content/themes/g5_hydrogen/custom/images/photonic3d-icon-sign.png, or is there a master asset that this was derived from (especially for the icon)?

kloknibor, 2016-04-08:
@ergobot orginally uploaded the images, I do found some iterations on the webhost. But I think it is best to wait for ergobots answer ;)

ergobot, 2016-04-13:
@jmkao - Where do you want the photonic3d vector (svg)?

jmkao, 2016-04-13:
Check it into host/images. Thanks!

ergobot, 2016-04-13:
Added vector assets in e92ad486595b547e79f16adaf7c4ba172c21da75

closed 2016-04-13 by ergobot