area515/Photonic3D #204: Rebrand to Photonic3D Phase 2: Rename Repo

jmkao, 2016-04-08:
Rename the area515 repo from Creation-Workshop-Host to Photonic3D.

Old links should continue to work due to automatic redirects.

kloknibor, 2016-04-25:
Who has got rights for this? Because this this is a very small task right?

jmkao, 2016-04-25:
Any of Wes, ergo, or I can do it, but it only makes sense after Wes next promotes his fork to area515 since that contains phase 1.

kloknibor, 2016-06-20:
Since we pushed wes fork to area515 I suppose it is time to do this step right? I think currently photonic3D scares people by it's complexity so I was thinking about making a bunch of clips. But it would be nice to use the right links :)!

jmkao, 2016-06-20:
Given the recent issue with the templates, I do not think we're ready to rename. I have fixed most of the bigger items like releasing builds and having start.sh work, but the next step would be for Wes's repo to be renamed, run for a few weeks, and then do area515.

The current URL will continue to work forever, so there shouldn't be any problems using it. I suggest you proceed with whatever you want to do with the current setup.

WesGilster, 2016-06-20:
I can rename my repo any time, but we need to get things straightened a bit. That will need to be timed with a change to the production code base to read printer profiles from that newly renamed repo and/or change the default repo to area515. Having my repo be the default repo to pull printer configurations was only a temporary development change until area515 had all of the printer profiles available. Now that area515 has all the printer profiles available, we can change the default repo to area515.

jmkao, 2016-06-20:
That should be doable since the repo for printer profiles is controlled by the `printerProfileRepo` parameter in `config.properties` since da2ce0f53a7bd3c107748f15e236d8ada56e077f.

I don't remember finding any other issues when I renamed my repo, but they might still pop up as we go up the chain.

kloknibor, 2016-06-21:
the printerProfileRepo setting works just fine. At first I overruled all hardcoded adresses but currently I'm using that setting for a while which seems to work just fine for me. 

PS : I use it to show only photocentric printers and slice settings, it works fine.

WesGilster, 2016-09-18:
Closed in cwh-1.0.0-beta3.

closed 2016-09-18 by WesGilster