area515/Photonic3D #207: Stupid question

kloknibor, 2016-04-15:
Hi,

I'm trying to build with the new gradle way instead of the ant build... But to be honest... I don't understand how I start the build with gradle :$? anyway you guys can give me a quick/kik start?

Thanks!

Robin

kloknibor, 2016-04-15:
Oh I use eclipse on windows ;)!

ergobot, 2016-04-15:
First you'll need a way to run gradle scripts in eclipse.  (Optionally, you
can run gradle commands/tasks from the command line without an ide)

If you are looking for a plugin:
http <http://projects.eclipse.org/projects/tools.buildship/downloads>://
<http://projects.eclipse.org/projects/tools.buildship/downloads>
projects.eclipse.org
<http://projects.eclipse.org/projects/tools.buildship/downloads>/projects/
<http://projects.eclipse.org/projects/tools.buildship/downloads>
tools.buildship
<http://projects.eclipse.org/projects/tools.buildship/downloads>/downloads
<http://projects.eclipse.org/projects/tools.buildship/downloads>

If you want to download eclipse, and have gradle tools built in, try the
spring tool suite:
https <https://spring.io/tools/sts/all>:// <https://spring.io/tools/sts/all>
spring.io <https://spring.io/tools/sts/all>/tools/
<https://spring.io/tools/sts/all>sts <https://spring.io/tools/sts/all>/all
<https://spring.io/tools/sts/all>

Next you'll want a simple tutorial to start using gradle in eclipse:
https <https://spring.io/guides/gs/gradle/>://
<https://spring.io/guides/gs/gradle/>spring.io
<https://spring.io/guides/gs/gradle/>/guides/
<https://spring.io/guides/gs/gradle/>gs
<https://spring.io/guides/gs/gradle/>/gradle/
<https://spring.io/guides/gs/gradle/>

After that, you can use the build.gradle file available at:
/Creation-Workshop-Host/host/build.gradle

I haven't run this gradle file yet as it is newer.  I can't comment on how
to run/debug this specific build.

Oh I use eclipse on windows ;)!

kloknibor, 2016-04-15:
Thanks @ergobot the buildship plugin is already activated in my eclipse install ;)! I found an tutorail of @jmkao somewhere else on the forum but I can't seem to find where I shoudl run  gradlew eclipse but I'm not sure how :/ But I'll try the preinstalled one and the tutorials ! Thanks ! (will try it tomorrow or sunday ;))

kloknibor, 2016-04-16:
With some help from #199 I managed to complete almost every step and can run the launchers now :)!  But I get this error : 

org.gradle.tooling.GradleConnectionException: Could not execute build using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.11-bin.zip'.
	at org.gradle.tooling.internal.consumer.ResultHandlerAdapter.onFailure(ResultHandlerAdapter.java:63)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:57)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
	at org.gradle.tooling.internal.consumer.BlockingResultHandler.getResult(BlockingResultHandler.java:46)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher.run(DefaultBuildLauncher.java:72)
	at com.gradleware.tooling.toolingclient.internal.DefaultToolingClient.executeAndWait(DefaultToolingClient.java:123)
	at com.gradleware.tooling.toolingclient.internal.DefaultBuildLaunchRequest.executeAndWait(DefaultBuildLaunchRequest.java:49)
	at com.gradleware.tooling.toolingclient.internal.DefaultBuildLaunchRequest.executeAndWait(DefaultBuildLaunchRequest.java:28)
	at org.eclipse.buildship.core.launch.BaseLaunchRequestJob.runToolingApiJob(BaseLaunchRequestJob.java:94)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob$1.run(ToolingApiJob.java:67)
	at org.eclipse.buildship.core.util.progress.ToolingApiInvoker.invoke(ToolingApiInvoker.java:54)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob.run(ToolingApiJob.java:64)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: org.gradle.launcher.daemon.client.DaemonConnectionException: The newly created daemon process has a different context than expected.
It won't be possible to reconnect to this daemon. Context mismatch: 
Java home is different.
Wanted: DefaultDaemonContext[uid=null,javaHome=C:\Program Files\Java\jre1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=null,idleTimeout=null,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]
Actual: DefaultDaemonContext[uid=416ef4d2-ab22-4fe7-b06e-a77c6bd6a403,javaHome=C:\Program Files\Java\jdk1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=13196,idleTimeout=10800000,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]

	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.connectToDaemonWithId(DefaultDaemonConnector.java:134)
	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.startDaemon(DefaultDaemonConnector.java:114)
	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.connect(DefaultDaemonConnector.java:88)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:123)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:80)
	at org.gradle.tooling.internal.provider.DaemonBuildActionExecuter.execute(DaemonBuildActionExecuter.java:58)
	at org.gradle.tooling.internal.provider.DaemonBuildActionExecuter.execute(DaemonBuildActionExecuter.java:40)
	at org.gradle.tooling.internal.provider.LoggingBridgingBuildActionExecuter.execute(LoggingBridgingBuildActionExecuter.java:60)
	at org.gradle.tooling.internal.provider.LoggingBridgingBuildActionExecuter.execute(LoggingBridgingBuildActionExecuter.java:34)
	at org.gradle.tooling.internal.provider.ProviderConnection.run(ProviderConnection.java:132)
	at org.gradle.tooling.internal.provider.ProviderConnection.run(ProviderConnection.java:108)
	at org.gradle.tooling.internal.provider.DefaultConnection.getModel(DefaultConnection.java:168)
	at org.gradle.tooling.internal.consumer.connection.CancellableModelBuilderBackedModelProducer.produceModel(CancellableModelBuilderBackedModelProducer.java:58)
	at org.gradle.tooling.internal.consumer.connection.PluginClasspathInjectionSupportedCheckModelProducer.produceModel(PluginClasspathInjectionSupportedCheckModelProducer.java:41)
	at org.gradle.tooling.internal.consumer.connection.AbstractConsumerConnection.run(AbstractConsumerConnection.java:58)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher$1.run(DefaultBuildLauncher.java:84)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher$1.run(DefaultBuildLauncher.java:78)
	at org.gradle.tooling.internal.consumer.connection.LazyConsumerActionExecutor.run(LazyConsumerActionExecutor.java:83)
	at org.gradle.tooling.internal.consumer.connection.ProgressLoggingConsumerActionExecutor.run(ProgressLoggingConsumerActionExecutor.java:58)
	at org.gradle.tooling.internal.consumer.connection.RethrowingErrorsConsumerActionExecutor.run(RethrowingErrorsConsumerActionExecutor.java:38)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:55)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

Strange is that the locations are correct... The files are located there... The step I didn't know how to execute was the ./gradlew eclipse which was needed to rebuild the dependencies... But I'm not quite sure where to put this? CMD won't work... I did some small changes to the code but I can't try it out without gradle now :( Ant doesn't work too and I want to start using gradle anyway ;)!

jmkao, 2016-04-16:
Before you start using eclipse, you should make that your system Java development environment is correct for running builds from the command line from CMD.

You should be able to launch cmd and then successfully run commands like:

`gradlew.bat run`
`gradlew.bat build`

Before generating the eclipse project with `gradlew.bat eclipse` and then opening the project in eclipse.

If running from cmd doesn't work, make sure that you have the latest JDK 8 installed on your computer and that JAVA_HOME is set in your system properties globally (set it in the System control panel on Windows). Then re-launch any cmd prompts you have.

kloknibor, 2016-04-17:
I can run the gradle.bat with build and run but it still will give me errors because the tests fail... 

[image]

is this an dependencie problem? And I can run gradlew now ;)!

kloknibor, 2016-04-17:
I guess it is set up right now. I did an pull from wes DEV repo yesterday but I think there might be an error in it... I'll try the area515 one ;)!

kloknibor, 2016-04-17:
Ah the error I show previously the : 

Launching Gradle tasks failed due to an error connecting to the Gradle build.
Could not execute build using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.11-bin.zip'.

The newly created daemon process has a different context than expected.
It won't be possible to reconnect to this daemon. Context mismatch: 
Java home is different.
Wanted: DefaultDaemonContext[uid=null,javaHome=C:\Program Files\Java\jre1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=9736,idleTimeout=null,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]
Actual: DefaultDaemonContext[uid=49a0080c-965c-4187-98c4-6263545a4b95,javaHome=C:\Program Files\Java\jdk1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=8000,idleTimeout=10800000,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]
org.gradle.tooling.GradleConnectionException: Could not execute build using Gradle distribution 'https://services.gradle.org/distributions/gradle-2.11-bin.zip'.
	at org.gradle.tooling.internal.consumer.ResultHandlerAdapter.onFailure(ResultHandlerAdapter.java:63)
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:57)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
	at org.gradle.tooling.internal.consumer.BlockingResultHandler.getResult(BlockingResultHandler.java:46)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher.run(DefaultBuildLauncher.java:72)
	at com.gradleware.tooling.toolingclient.internal.DefaultToolingClient.executeAndWait(DefaultToolingClient.java:123)
	at com.gradleware.tooling.toolingclient.internal.DefaultBuildLaunchRequest.executeAndWait(DefaultBuildLaunchRequest.java:49)
	at com.gradleware.tooling.toolingclient.internal.DefaultBuildLaunchRequest.executeAndWait(DefaultBuildLaunchRequest.java:28)
	at org.eclipse.buildship.core.launch.BaseLaunchRequestJob.runToolingApiJob(BaseLaunchRequestJob.java:94)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob$1.run(ToolingApiJob.java:67)
	at org.eclipse.buildship.core.util.progress.ToolingApiInvoker.invoke(ToolingApiInvoker.java:54)
	at org.eclipse.buildship.core.util.progress.ToolingApiJob.run(ToolingApiJob.java:64)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: org.gradle.launcher.daemon.client.DaemonConnectionException: The newly created daemon process has a different context than expected.
It won't be possible to reconnect to this daemon. Context mismatch: 
Java home is different.
Wanted: DefaultDaemonContext[uid=null,javaHome=C:\Program Files\Java\jre1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=9736,idleTimeout=null,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]
Actual: DefaultDaemonContext[uid=49a0080c-965c-4187-98c4-6263545a4b95,javaHome=C:\Program Files\Java\jdk1.8.0_73,daemonRegistryDir=C:\Users\Robin\.gradle\daemon,pid=8000,idleTimeout=10800000,daemonOpts=-Dfile.encoding=windows-1252,-Duser.country=US,-Duser.language=en,-Duser.variant]

	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.connectToDaemonWithId(DefaultDaemonConnector.java:134)
	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.startDaemon(DefaultDaemonConnector.java:114)
	at org.gradle.launcher.daemon.client.DefaultDaemonConnector.connect(DefaultDaemonConnector.java:88)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:123)
	at org.gradle.launcher.daemon.client.DaemonClient.execute(DaemonClient.java:80)
	at org.gradle.tooling.internal.provider.DaemonBuildActionExecuter.execute(DaemonBuildActionExecuter.java:58)
	at org.gradle.tooling.internal.provider.DaemonBuildActionExecuter.execute(DaemonBuildActionExecuter.java:40)
	at org.gradle.tooling.internal.provider.LoggingBridgingBuildActionExecuter.execute(LoggingBridgingBuildActionExecuter.java:60)
	at org.gradle.tooling.internal.provider.LoggingBridgingBuildActionExecuter.execute(LoggingBridgingBuildActionExecuter.java:34)
	at org.gradle.tooling.internal.provider.ProviderConnection.run(ProviderConnection.java:132)
	at org.gradle.tooling.internal.provider.ProviderConnection.run(ProviderConnection.java:108)
	at org.gradle.tooling.internal.provider.DefaultConnection.getModel(DefaultConnection.java:168)
	at org.gradle.tooling.internal.consumer.connection.CancellableModelBuilderBackedModelProducer.produceModel(CancellableModelBuilderBackedModelProducer.java:58)
	at org.gradle.tooling.internal.consumer.connection.PluginClasspathInjectionSupportedCheckModelProducer.produceModel(PluginClasspathInjectionSupportedCheckModelProducer.java:41)
	at org.gradle.tooling.internal.consumer.connection.AbstractConsumerConnection.run(AbstractConsumerConnection.java:58)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher$1.run(DefaultBuildLauncher.java:84)
	at org.gradle.tooling.internal.consumer.DefaultBuildLauncher$1.run(DefaultBuildLauncher.java:78)
[... cut ...]
	at org.gradle.tooling.internal.consumer.async.DefaultAsyncConsumerActionExecutor$1$1.run(DefaultAsyncConsumerActionExecutor.java:55)
	at org.gradle.internal.concurrent.ExecutorPolicy$CatchAndRecordFailures.onExecute(ExecutorPolicy.java:54)
	at org.gradle.internal.concurrent.StoppableExecutorImpl$1.run(StoppableExecutorImpl.java:40)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

Is only when using the launcher full release! That's what went wrong...

WesGilster, 2016-04-17:
Yeah, I saw that. I haven't been able to run the tests with the JUnit plugin so they are slowly degrading. The gradle plugin does run them but feedback is fairly useless in Eclipse.

WesGilster, 2016-04-17:
Fixed tests.

jmkao, 2016-04-17:
I'm thinking that probably, the BuildShip plugin isn't necessary in order to setup the gradle/eclipse integration with dependency management. Just doing `gradlew eclipse` should be sufficient.

Is something with BuildShip interfering with the JUnit plugin? Or is something about the classpath entries made by `gradelw eclipse` not agreeing with JUnit?

WesGilster, 2016-04-17:
I'm not using BuildShip so that's not it. Every few minutes I'm seeing a process kicked off. I'm betting it's the Gradle Project Builder since it acts very much like the Ant builder(if you enable it) in that it leaves traces of it's destruction in the console after it decides it needs to run. Anyway, often times this leaves my project in a state where Eclipse can't find any of the classes for JUnit or for regular class execution. I can generally get it fixed, but I haven't really determined what is causing it, or a standard procedure for fixing it. If it wasn't such a popular plugin I'd write it off as unstable.

kloknibor, 2016-04-20:
Allright so I had to set CLASSPATH and edit Path too in environment variables in the system properties, these are set now and javac runs. I also updated the newest JDK to be sure (got 2 versions installed now...) So now we got improvement but still errors from command line when I use gradlew(.bat) run : 

D:\git1\Creation-Workshop-Host\host>gradlew.bat run
:compileJava UP-TO-DATE
:processResources UP-TO-DATE
:classes UP-TO-DATE
:run
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further detail
s.
Apr 20, 2016 10:28:39 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: >>> Starting UPnP service...
Apr 20, 2016 10:28:39 PM org.fourthline.cling.UpnpServiceImpl <init>
INFO: Using configuration: org.fourthline.cling.DefaultUpnpServiceConfiguration
Apr 20, 2016 10:28:39 PM org.fourthline.cling.transport.RouterImpl <init>
INFO: Creating Router: org.fourthline.cling.transport.RouterImpl
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Intel
(R) Centrino(R) Wireless-N 2230
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Creating wildcard socket (for receiving multicast datagrams) on port: 1900

Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.impl.MulticastReceiverIm
pl init
INFO: Joining multicast group: /239.255.255.250:1900 on network interface: Virtu
alBox Host-Only Ethernet Adapter
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.RouterImpl startAddressB
asedTransports
WARNING: Failed to init StreamServer: java.net.BindException: Address already in
 use: bind
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.RouterImpl startAddressB
asedTransports
WARNING: Removing unusable address: /192.168.1.195
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.RouterImpl startAddressB
asedTransports
WARNING: Failed to init StreamServer: java.net.BindException: Address already in
 use: bind
Apr 20, 2016 10:28:40 PM org.fourthline.cling.transport.RouterImpl startAddressB
asedTransports
WARNING: Removing unusable address: /192.168.56.1
> Building 75% > :run

Gradlew build works fine... 

it seems I miss an class from the logger and the software doesn't like my hardware too much... I do seem to have the exact same errors as before when I enter eclipse but CMD looks better now... Any hints/tips or tricks?

jmkao, 2016-04-20:
When the command line shows `Building 75% > :run` and hangs, the server is running and will keep running until you hit Control-C. At that point, you should be able to go to localhost:9091 as usual.

kloknibor, 2016-04-21:
Yeah sorry you are right :$ It works fine now! I expected it to hit 100% thatś why I tough it didn work.

Even releases to my github from cmd work fine now so that great! 

If someone else has problems try to run javac in cmd if this returns an error you need to set you java home class path and edit your path. If this works gradle should work too!

I'll close this issue for now, only some eclipse problem but that will be solved simpeler and otherwise i will use the cmd to release ;)! Thanks everyone!

closed 2016-04-21 by kloknibor